darktable-org/rawspeed #19: getCamera() with prefix-search issue

LebedevRI, 2017-01-06:
https://github.com/klauspost/rawspeed/issues/166

I added that api in https://github.com/klauspost/rawspeed/commit/a45724ec3c201b67f5ff732317921c0feca878c6 and https://github.com/klauspost/rawspeed/commit/e1b5ccae3515773755621cee305170fa5ea63808.
And now i have just discovered how it backfires.
With raw file from https://redmine.darktable.org/issues/11354 from `Canon EOS Rebel T6`, which currently has broken alias, it matches to `Canon EOS Rebel T6i`.
That happens because internally "make model mode" is a string, https://github.com/klauspost/rawspeed/blob/a45724ec3c201b67f5ff732317921c0feca878c6/RawSpeed/CameraMetaData.h#L37

tredlie, 2017-02-03:
This issue still seems to be around. What are the chances of getting this fixed as it brings darktable usage for a Leica M9 to a grinding halt, reporting this camera top to be a Mamiya.
Or, maybe, this does not do any harm other than reporting the wrong Maker in the Image Information window in DT. The exported JPG looks OK.

LebedevRI, 2017-02-03:
I believe i have just fixed it locally. At least the unittest i just wrote no longer fails.

commit cfceec42 (2017-02-04): Merge remote-tracking branch 'upstream/pr/68' into develop

* upstream/pr/68:
  compiler-warnings: on apple, raise frame/stack usage warn to 5K
  ColorFilterArray: while there, add some ugly tests to get 100% coverage
  CameraMetaData: make id a tuple, fix search. Fixes #19
  CameraMetaDataTest: cover search with kind-of test
commit fc1b8e33 (2017-02-04): CameraMetaData: make id a tuple, fix search. Fixes #19
closed 2017-02-04 by LebedevRI