darktable-org/rawspeed #30: CFA tag is relative to the Crop XY

LebedevRI, 2017-01-11:
CFA2 is not, thankfully.

So e.g. if we have:
```xml
	<Camera make="Panasonic" model = "DMC-FZ28">
		<ID make="Panasonic" model="DMC-FZ28">Panasonic DMC-FZ28</ID>
		<CFA width="2" height="2">
			<Color x="0" y="0">BLUE</Color>
			<Color x="1" y="0">GREEN</Color>
			<Color x="0" y="1">GREEN</Color>
			<Color x="1" y="1">RED</Color>
		</CFA>
		<Crop x="0" y="0" width="-56" height="-4"/>
		<Sensor black="15" white="3986"/>
	</Camera>
```
If we change crop x/y by a multiple of two, everything is fine.
But if we change it by just one, we also need to adjust `<CFA>`

axxel, 2017-01-11:
The code says otherwise: https://github.com/darktable-org/rawspeed/blob/develop/RawSpeed/RawDecoder.cpp#L579-L583

I was just about to suggest to implement just that when I remembered having seen it before. If there is some bug it should be fixed but the meaning of the origin of the CFA pattern should stay as is (relative to the uncropped sensor).

LebedevRI, 2017-01-11:
> The code says otherwise: https://github.com/darktable-org/rawspeed/blob/develop/RawSpeed/RawDecoder.cpp#L579-L583

Do note that it is in `if (applyCrop) {`

axxel, 2017-01-11:
Yes? Please have a closer look at what the code does: if `applyCrop` is false, then the 'cropped' image is simply the 'complete'/'uncropped' image. And since the CFA data is originally parsed with respect to this uncropped image, it stays as is.

LebedevRI, 2017-01-11:
Very interesting, i remember the opposite thing.
But even https://github.com/darktable-org/rawspeed/blob/develop/data/README.md agrees with you
```
This is the position at of the colors on the uncropped image, so it will be the same no matter what crop you specify.
```