darktable-org/rawspeed #34: GREEN2 color is most likely bogus

LebedevRI, 2017-01-13:


axxel, 2017-01-27:
While I was thinking about my last comment regarding the dcraw filter cropping, I thought, I might as well hack something. Here it is, in case anyone wants to replace the implementation in https://github.com/darktable-org/darktable/blob/ef45705b5ab6f47eb317a473b33dfe38f2f03b09/src/common/imageio_rawspeed.cc#L104-L114.

Note: this code actually allows to perform a y-shift on filters like the ones used in the old Canon PowerShot A50, which is not supported by the current rawspeed version of this.

```
#include <algorithm>
#include <cassert>
#include <inttypes.h>

uint32_t dcraw_filter_shift(uint32_t filter, int x, int y)
{
  // filter is a series of 4 bytes that describe a 2x8 matrix. 2 is the width,
  // 8 is the height. each byte describes a 2x2 pixel block. so each pixel has
  // 2 bits of information. This allows to distinguish 4 different colors.

  if (std::abs(x) & 1) {
    // A shift in x direction means swapping the first and second half of each nibble.
    // see http://graphics.stanford.edu/~seander/bithacks.html#SwappingBitsXOR
    for (int n = 0; n < 8; ++n) {
      int i = n * 4;
      int j = i + 2;
      uint32_t x = ((filter >> i) ^ (filter >> j)) & ((1U << 2) - 1);
      filter = filter ^ ((x << i) | (x << j));
    }
  }

  // A shift in y direction means rotating the whole int by 4 bits.
  y *= 4;
  y = y >= 0 ? y % 32 : -((-y) % 32);
  filter = (filter >> y) | (filter << (32-y));

  return filter;
}

int main()
{
  // see dcraw.c near line 240
  uint32_t
      bggr = 0x16161616,
      grbg = 0x61616161,
      gbrg = 0x49494949,
      rggb = 0x94949494;
  assert(dcraw_filter_shift(rggb, 0, 0) == rggb);
  assert(dcraw_filter_shift(rggb, 1, 0) == grbg);
  assert(dcraw_filter_shift(rggb, 0, 1) == gbrg);
  assert(dcraw_filter_shift(rggb, 1, 1) == bggr);
  assert(dcraw_filter_shift(rggb, 0, 2) == rggb);
}
```

axxel, 2017-01-29:
@LebedevRI If this code (or what ever version you end up choosing) got into dt and removed the requirement for the dcraw_filter constructor, please let me know. I can then continue to simplify the CFA code.

LebedevRI, 2017-01-29:
I think it should be in rawspeed. Why should it be in dt?

axxel, 2017-01-29:
> I think it should be in rawspeed. Why should it be in dt?

Oh, well. I thought this could pretty much be a replacement of the above linked dt code. Only missing would be the one line `if(!filters || filters == 9u) return filters;`.

If you prefer to have it in rawspeed, why not. Maybe other projects would have a use for it as well, I don't know. If you would put it in dt, then the rs code can be changed later (by me) and your dt build never breaks.

LebedevRI, 2017-01-29:
I don't think this cfa shift is too niche thing, that could only be used by dt, thus i think it should be in rawspeed.

The rawspeed changes never break dt anyway, the rawspeed [submodule] repo commit id is kept in git history, and git is smart-enough to check-out the appropriate submodule revision.

axxel, 2017-01-29:
> I don't think this cfa shift is too niche thing, that could only be used by dt, thus i think it should be in rawspeed.

Ok, I'll prepare something then.

> The rawspeed changes never break dt anyway, the rawspeed [submodule] repo commit id is kept in git history, and git is smart-enough to check-out the appropriate submodule revision.

I have no experience with sub modules: so you basically manually gatekeep any updates of the submodule by checking in a new commit id of the sub module at the time you'd like to update the submodule? Would make sense to me that way.

LebedevRI, 2017-01-29:
> so you basically manually gatekeep any updates of the submodule by checking in a new commit id of the sub module at the time you'd like to update the submodule?

yep

PR #54 (2017-02-01): Remove the usage of invalid CFA_GREEN2
This fixes #34. The actual number '3' is still kept as a dummy tag to
keep the ColorFilterArrayTest.ToDcrawAndBack unit test happy.

The string output of this dummy tag is now "UNKNOWN" therefore the
string length check unit test had to be slightly adjusted.

This change 'breaks' 3 files in rstest by actually fixing them: the old
"GREEN2" color is now reported as "GREEN".
commit 5204d0d9 (2017-02-01): Remove the usage of invalid CFA_GREEN2

This fixes #34. The actual number '3' is still kept as a dummy tag to
keep the ColorFilterArrayTest.ToDcrawAndBack unit test happy.

The string output of this dummy tag is now "UNKNOWN" therefore the
string length check unit test had to be slightly adjusted.

This change 'breaks' 3 files in rstest by actually fixing them: the old
"GREEN2" color is now reported as "GREEN".
closed 2017-02-01 by LebedevRI