darktable-org/rawspeed #40: GoPro HERO5 Black raw "format"

LebedevRI, 2017-01-19:
Just so it does not get lost
Copied from https://redmine.darktable.org/issues/11466

```
The .GPR format is an extension Adobe’s .DNG format, adding enhanced compression (VC5 Standard)

A sample file:
https://raw.pixls.us/getfile.php/936/raw/GOPR0020.GPR

FAQ from GoPro about the file format:
https://gopro.com/help/articles/Question_Answer/What-is-RAW-Format

An interesting blogpost about the "VC5 Standard" 
https://medium.com/@kierank_/reverse-engineering-the-gopro-cineform-codec-7411312bfe1c#.9f7ztpjo8
```

supertobi, 2017-01-19:
Here is the jpeg version of the photo:

[image]

supertobi, 2017-01-19:
GoPro HERO5 Black RawTherapee issue:
https://github.com/Beep6581/RawTherapee/issues/3569

shoffmeister, 2017-02-01:
https://trac.ffmpeg.org/ticket/1087 

https://git.ffmpeg.org/gitweb/ffmpeg.git/blob_plain/HEAD:/libavcodec/cfhd.c <-- that is supposed to decode the video, so the compression might serve as a template? 

This codec goes by VC-5 (misnamed, it seems), cfhd, CineForm HD.

Also: http://cineform.com/compression-subsystem

supertobi, 2017-10-27:
GoPro Open Sources the CineForm Codec:
https://gopro.com/news/gopro-open-sources-the-cineform-codec

https://github.com/gopro/cineform-sdk

LebedevRI, 2017-10-27:
Yep, seen that yesterday.
I suppose libraw will want to link to that.
As for rawspeed, i suppose a clean, minimal implementation (with less horrible code than that, and no double/triple-licensing shenanigans) could be added.

LibRaw, 2017-10-28:
>> I suppose libraw will want to link to that.
Most likely no in (coming) 0.19, but most likely yes in next version cycle

pzemlyan, 2018-05-05:
General Purpose Raw image format API was released: https://github.com/gopro/gpr
Also a few HERO6 samples was uploaded: https://raw.pixls.us/getfile.php/2556/nice/GoPro%20-%20HERO6%20Black%20-%2016bit%20(4:3).GPR

supertobi, 2018-09-24:
GoPro just released the HERO7, are there any news here ore in @LibRaw?

LebedevRI, 2018-09-24:
Here - no, nothing has changed.

After reading through the existing crumbs of info on the Canon's CRX codec,
they both seem to be based on wavelets, which is, i suppose, not all that surprising.
So _maybe_ after Canon's CRX codec is implemented,
_maybe_ this could be implemented on top of that code.

lclevy, 2018-09-24:
https://github.com/lclevy/canon_cr3/blob/master/readme.md#crx-compression

stloeffler, 2018-09-25:
As I needed this myself, I implemented a basic GPR decompressor in #146 - I hope this helps.

supertobi, 2018-09-25:
Btw. I just discovered this project:
https://github.com/emericg/libcineform

closed 2018-10-20 by LebedevRI