darktable-org/rawspeed #49: Coverity CID 1397763

LebedevRI, 2017-01-25:
@axxel just so that it does not get lost
```c++
*** CID 1397763:    (SIZEOF_MISMATCH)
/RawSpeed/LJpegPlain.cpp: 232 in RawSpeed::LJpegPlain::decodeN_X_Y<(int)2, (int)1, (int)1>()()
226               dest[3 + i*pixel_pitch] = p[0] += ht[0]->decodeNext(bitStream);
227             });
228
229             dest[1] = p[1] += ht[1]->decodeNext(bitStream);
230             dest[2] = p[2] += ht[2]->decodeNext(bitStream);
231
>>>     CID 1397763:    (SIZEOF_MISMATCH)
>>>     Adding "6UL /* 3 * sizeof (RawSpeed::ushort16) */" to pointer "dest" of type "RawSpeed::ushort16 *" is suspicious because adding an integral value to this pointer automatically scales that value by the size, 2 bytes, of the pointed-to type, "RawSpeed::ushort16".  Most likely, the multiplication by "sizeof (RawSpeed::ushort16)" in this expression is extraneous and should be eliminated.
232             dest += 3 * sizeof(ushort16);
233           }
234
235           pixInSlicedLine -= X_S_F;
236         }
237
/RawSpeed/LJpegPlain.cpp: 232 in RawSpeed::LJpegPlain::decodeN_X_Y<(int)3, (int)2, (int)2>()()
226               dest[3 + i*pixel_pitch] = p[0] += ht[0]->decodeNext(bitStream);
227             });
228
229             dest[1] = p[1] += ht[1]->decodeNext(bitStream);
230             dest[2] = p[2] += ht[2]->decodeNext(bitStream);
231
>>>     CID 1397763:    (SIZEOF_MISMATCH)
>>>     Adding "6UL /* 3 * sizeof (RawSpeed::ushort16) */" to pointer "dest" of type "RawSpeed::ushort16 *" is suspicious because adding an integral value to this pointer automatically scales that value by the size, 2 bytes, of the pointed-to type, "RawSpeed::ushort16".  Most likely, the multiplication by "sizeof (RawSpeed::ushort16)" in this expression is extraneous and should be eliminated.
232             dest += 3 * sizeof(ushort16);
233           }
234
235           pixInSlicedLine -= X_S_F;
236         }
237
/RawSpeed/LJpegPlain.cpp: 232 in RawSpeed::LJpegPlain::decodeN_X_Y<(int)3, (int)2, (int)1>()()
226               dest[3 + i*pixel_pitch] = p[0] += ht[0]->decodeNext(bitStream);
227             });
228
229             dest[1] = p[1] += ht[1]->decodeNext(bitStream);
230             dest[2] = p[2] += ht[2]->decodeNext(bitStream);
231
>>>     CID 1397763:    (SIZEOF_MISMATCH)
>>>     Adding "6UL /* 3 * sizeof (RawSpeed::ushort16) */" to pointer "dest" of type "RawSpeed::ushort16 *" is suspicious because adding an integral value to this pointer automatically scales that value by the size, 2 bytes, of the pointed-to type, "RawSpeed::ushort16".  Most likely, the multiplication by "sizeof (RawSpeed::ushort16)" in this expression is extraneous and should be eliminated.
232             dest += 3 * sizeof(ushort16);
233           }
234
235           pixInSlicedLine -= X_S_F;
236         }
237
/RawSpeed/LJpegPlain.cpp: 232 in RawSpeed::LJpegPlain::decodeN_X_Y<(int)4, (int)1, (int)1>()()
226               dest[3 + i*pixel_pitch] = p[0] += ht[0]->decodeNext(bitStream);
227             });
228
229             dest[1] = p[1] += ht[1]->decodeNext(bitStream);
230             dest[2] = p[2] += ht[2]->decodeNext(bitStream);
231
>>>     CID 1397763:    (SIZEOF_MISMATCH)
>>>     Adding "6UL /* 3 * sizeof (RawSpeed::ushort16) */" to pointer "dest" of type "RawSpeed::ushort16 *" is suspicious because adding an integral value to this pointer automatically scales that value by the size, 2 bytes, of the pointed-to type, "RawSpeed::ushort16".  Most likely, the multiplication by "sizeof (RawSpeed::ushort16)" in this expression is extraneous and should be eliminated.
232             dest += 3 * sizeof(ushort16);
233           }
234
235           pixInSlicedLine -= X_S_F;
236         }
237
```

axxel, 2017-01-25:
Yep. Will be gone with my next pr.

LebedevRI, 2017-01-27:
fixed

closed 2017-01-27 by LebedevRI