darktable-org/rawspeed #53: .clang-format tuning

axxel, 2017-01-26:
Here is what I came up with after playing a bit with the options:
```
# strongly requested
ColumnLimit: 100 # at least 100
BreakBeforeBraces: Linux # break before braces on function/namespace/class

# nice to have
KeepEmptyLinesAtTheStartOfBlocks: true # at least if BreakBeforeBraces is set to default
MaxEmptyLinesToKeep: 2

# maybe nice to have
#BreakConstructorInitializersBeforeComma: true
#AllowShortIfStatementsOnASingleLine: false
#AllowShortFunctionsOnASingleLine: false
#AllowShortBlocksOnASingleLine: false
```

LebedevRI, 2017-01-27:
```
BreakBeforeBraces: Linux # break before braces on function/namespace/class
```
i would prefer to avoid cases like:
```c
if(very_complicated_check());
{
  slow_code();
  // FIXME: why is it so slow?
}
```
With `BreakBeforeBraces: Linux`, it can't happen, so it might be ok..

---

```
KeepEmptyLinesAtTheStartOfBlocks: true # at least if BreakBeforeBraces is set to default
```
`BreakBeforeBraces` is not default, so i guess not needed

---

```
MaxEmptyLinesToKeep: 2
```
Any specific details? i'd say a comment should be inserted in such cases

---

```
#BreakConstructorInitializersBeforeComma: true # looks too verbose, but apparently places " : " on new line, which i actually like
#AllowShortIfStatementsOnASingleLine: false
#AllowShortFunctionsOnASingleLine: false
#AllowShortBlocksOnASingleLine: false
```
Uhm, that looks too verbose, but it does help with readability.
But it is practically the direct opposite of `ColumnLimit`.

Regarding `ColumnLimit` - it is not a hard limit, i'm pretty sure that clang-format has some internal logic to not break some long lines in some cases. So in reality it may result in even longer lines. I really don't want to change it :(

So, how about:
```diff
diff --git a/.clang-format b/.clang-format                                                                                                                                                                              
index cea0494..ffc13c6 100644
--- a/.clang-format
+++ b/.clang-format
@@ -5,4 +5,11 @@ Standard:     Cpp11
 # Force pointers to the type for C++.
 DerivePointerAlignment: false
 PointerAlignment: Left
+# break before braces on function/namespace/class
+BreakBeforeBraces: Linux
+# helps with readability
+BreakConstructorInitializersBeforeComma: true
+AllowShortIfStatementsOnASingleLine: false
+AllowShortFunctionsOnASingleLine: false
+AllowShortBlocksOnASingleLine: false
 ...
```

axxel, 2017-01-27:
> Any specific details? i'd say a comment should be inserted in such cases

I just think sometimes having 2 empty lines might help further visually separate unrelated parts. maybe between two function bodies. I have no 'smoking gun' type of example, just thought, that giving the developer the freedom to actually use 2 empty lines is not really _bad_.

**About the `Allow...OnASingleLine`:** I actually would like to be able put stuff on a single line, e.g. here:
```
    switch(sizeof(T)) {
    case 1: break;
    case 2: ret = BSWAP16(ret); break;
    case 4: ret = BSWAP32(ret); break;
    case 8: ret = BSWAP64(ret); break;
    }
```
Which is a lot more readable that way. I know that none of the mentioned `Allow...` settings effect this switch statement, but I found this: `AllowShortCaseLabelsOnASingleLine: true`, so please add that.

And I noticed that `AllowShortFunctionsOnASingleLine` can also be set to `Emtpy`, which I like a lot for empty constructors.

**ColumnLimit**
> Regarding ColumnLimit - it is not a hard limit, i'm pretty sure that clang-format has some internal logic to not break some long lines in some cases. So in reality it may result in even longer lines. I really don't want to change it :(

Just to give an example why I _really_ think that limit makes it worse:
```
    return ((mRaw->metadata.subsampling.y * mRaw->metadata.subsampling.x) - 1) >> 1;
```
becomes
```
    return ((mRaw->metadata.subsampling.y * mRaw->metadata.subsampling.x) -
            1) >>
           1;
```
Which completely ruined the readability in my opinion. so again: is 80 _really_ the absolute maximum you can tolerate with your side-by-side diff?

There is one other option that 'fixes' this particular example: `PenaltyExcessCharacter: 1`. So this generally sticks with the 80 limit but makes it a little more tolerant in cases where you just scratched the limit. Still: anything that is greater than 80 and still acceptable would be preferred.

**BreakConstructorInitializersBeforeComma:**

> looks too verbose, but apparently places " : " on new line, which i actually like

I also would prefer just to have the ':' on a new line and not necessarily all member initializers on a separate line, like this:
```
  Buffer(const Buffer& rhs)
    : data(rhs.data), size(rhs.size) {}
```
I did not find a combination of options that would allow that. Also this option seems to have a bug (in clang-format-3.7 at least) where it breaks the '{}' into separate lines no matter what `AllowShortFunctionsOnASingleLine` is set to. So I get either
```
  Buffer(const Buffer& rhs) : data(rhs.data), size(rhs.size) {}
```
or
```
  Buffer(const Buffer& rhs)
      : data(rhs.data)
      , size(rhs.size)
  {
  }
```

Hence, I don't really care about this option because both choices I have don't really convince me. :-(

It would be nice if the tool would allow a few more "don't care / leave as is" options instead of forcing everything to be either one way or an other. Maybe there is and I simply have not found it?

LebedevRI, 2017-01-27:
> I just think sometimes having 2 empty lines might help further visually separate unrelated parts. maybe between two function bodies.

It should be solved by a small comment, describing something about the code
The code admittedly currently has an awful documentation.
But ok.

>```c++
>switch(sizeof(T)) {
>case 1: break;
>case 2: ret = BSWAP16(ret); break;
>case 4: ret = BSWAP32(ret); break;
>case 8: ret = BSWAP64(ret); break;
>}
>```

Shouldn't it be solved via C++ templates in the first place, might help by being compile-time?

> ```c++
>return ((mRaw->metadata.subsampling.y * mRaw->metadata.subsampling.x) - 1) >> 1;
> ```
So 9 operations on one line. "Readability" :)
Something like this, and the line is readable, and no longer too long
```c++
const auto& ss = mRaw->metadata.subsampling;
return ((ss.y * ss.x) - 1) >> 1;
```

---

So the rest can be summed-up as more longer lines everywhere. :(

It is probably a personal preference, but it is easier to read a lot of shorter lines, than a fewer longer lines. Because in the second case one needs to move eyes horizontally too, not just vertically.
"modern" sites can be seen using only the middle of the screen, even github

I do understand that all this coding style is a personal preference, and that is why i really want to just use one well-defined style.

So i'd be ok with
```
BreakBeforeBraces: Linux
MaxEmptyLinesToKeep: 2
```
but not the line length stuff. :(

> It would be nice if the tool would allow a few more "don't care / leave as is" options instead of forcing everything to be either one way or an other. Maybe there is and I simply have not found it?

`git rm -f .clang-format` :)
Seen http://cf.monofraps.net/ ?

axxel, 2017-01-27:
> Shouldn't it be solved via C++ templates in the first place, might help by being compile-time?

I guess this could indeed be done completely without a run-time switch but in a release build, the whole switch and the `if` will be removed by the optimizer. This is close to the inner most loop of every BitPump and I benchmarked this to death, looking at the resulting assembly, playing with SSE, etc. ;). I'm saying: in terms of speed: don't worry. :) To remove the switch for a debug build would be possible but probably less readable and not worth the effort.

There are other places where `AllowShortCaseLabelsOnASingleLine: true` is useful, e.g. https://github.com/darktable-org/rawspeed/blob/develop/src/librawspeed/metadata/ColorFilterArray.cpp#L234-L240

> but not the line length stuff. :(

Alright (I can say, I tried ;)). So this will then simply lead to more changes like your example and a preference for shorter names on my side to prevent stuff like that in the first place. Although variable names like `ss` might be bit too short in terms of understanding what is going on ;).

Based on your previous suggestion I'd like to propose the following 'compromise':
```
BreakBeforeBraces: Linux
MaxEmptyLinesToKeep: 2
AllowShortFunctionsOnASingleLine: Empty
AllowShortCaseLabelsOnASingleLine: true
AllowShortIfStatementsOnASingleLine: false
AllowShortBlocksOnASingleLine: false
```
And to prevent really ugly stuff like above I'd suggest to actually not go for one huge reformat commit but rather use it where we clean-up anyway. And have an eye on those issues.

> git rm -f .clang-format :)

I found something a little less drastic: `// clang-format off`. Not sure where, but maybe that might come in handy in very special cases. I'll keep it in mind.

LebedevRI, 2017-01-27:
> I found something a little less drastic: // clang-format off. Not sure where, but maybe that might come in handy in very special cases. I'll keep it in mind.

**_very special cases_**
Right now the only place where i *could* accept such usage is [enum TiffTag](https://github.com/darktable-org/rawspeed/blob/5aa22bcf8e57be071a0e0aa5ba9c04e9edcdf97f/src/librawspeed/tiff/TiffTag.h) and such.

axxel, 2017-01-27:
> very special cases

Those were my words, yes. :)

There is something related to the subject of shorter names and since this just became more of an issue, I'll mention it here. You recently introduced 'std::' prefixes in all header files. I agree with the idea that it is bad style to force the inclusion of a whole namespace on someone just because he included your header file. So having `using namespace std;` at the top of `StdAfx.h` (or where ever it was) is bad. But: we could very well add that `using` in, say `Common.h` **inside** the `RawSpeed` namespace. Best of both worlds... Only real difference compared to now: we can not use standard names for our own stuff but that would be bad anyway. Right?

LebedevRI, 2017-01-27:
>  we can not use standard names for our own stuff but that would be bad anyway. Right?

yes

> But: we could very well add that using in, say Common.h inside the RawSpeed namespace. 

Wouldn't it pollute the RawSpeed namespace?
Anyway, i think `using namespace` should only be done in non-headers.

axxel, 2017-01-27:
> Wouldn't it pollute the RawSpeed namespace?

Yes, but exactly the same way it does when you 'import' it in the individual cpp-file. Only that in the latter case this pollution only takes effect in this single translation unit. So could decided to pollute one cpp file but not another one. But why would you want to do that? It would allow the reuse of a std name in that other cpp file, but we just agreed that this would be a bad idea anyway. And you added `using namespace std;` declarations to all our cpp files, anyway (if not all, then most).

> Anyway, i think using namespace should only be done in non-headers.

But why? I tried my best explaining the technical reasoning behind my suggestion. If there is a mistake, please let me know.

There would also be the possibility to limit the 'pollution' by only using a specific set of essential names from the std namespace. I found that with 10 lines we would cover like 95% of our use-cases:
```
using std::string;
using std::vector;
using std::map;
using std::unique_ptr;
using std::exception;
using std::runtime_error;
using std::move;
using std::min;
using std::max;
using std::abs;
```

LebedevRI, 2017-01-27:
> Yes, but exactly the same way it does when you 'import' it in the individual cpp-file. Only that in the latter case this pollution only takes effect in this single translation unit. So could decided to pollute one cpp file but not another one.

That exactly is the point.

```c++
...

#pragma once

namespace RawSpeed {

using std::string;
...
```

```
[ 14%] Building CXX object src/librawspeed/CMakeFiles/rawspeed.dir/common/Common.cpp.o
In file included from /home/lebedevri/rawspeed/src/librawspeed/common/Common.cpp:38:0:
/home/lebedevri/rawspeed/src/librawspeed/common/Common.h:26:12: error: ‘std::string’ has not been declared
 using std::string;
            ^~~~~~
```
So that header would need to include all these headers first. Which also exactly what i *don't* want.

axxel, 2017-01-27:
> That exactly is the point.

As I said a few sentences later, you 'pollute' each and every cpp file of the library anyway. So where is the point here?

> So that header would need to include all these headers first. Which also exactly what i don't want.

That header already includes pretty much everything I was referring to: https://github.com/darktable-org/rawspeed/blob/develop/src/librawspeed/common/Common.h#L23-L29 ?!?

And for other stuff like std::map, which is not included in Common.h, we can just as easily add this `using` line in that header file that actually includes `<map>` if that is what you prefer.

LebedevRI, 2017-01-27:
> That header already includes pretty much everything I was referring to

It includes *only* the stuff that is actually used in that header. Same goes for all other headers.

> pollute each and every cpp file of the library anyway. So where is the point here?

headers are `#include`d into other headers and into sources. sources are not (should not be) `#include`d into anything.
If one header pollutes global namespace, it will be polluted everywhere where that header is included from.
If one source pollutes global namespace, it will be polluted only in that standalone compilation unit.

Edge-case: `<zlib.h>` is currently only used in dngdecoderslices. If we include it in e.g. `common.h`, it will make `compress()` available in all the other headers/sources too. Why, if only dngdecoderslices uses it? Same with jpeg headers, which define crapton of ugly shit like `TRUE`/`FALSE`

Same with **everything** else. Why pollute everything with the stuff you don't need.
So not counting probably replacing `using namespace std;` with more fine-grained `using std::?`, everything is *EXACTLY* how it should be.

LebedevRI, 2017-01-28:
> That header already includes pretty much everything I was referring to: https://github.com/darktable-org/rawspeed/blob/develop/src/librawspeed/common/Common.h#L23-L29 ?!?

I mean, include before `using std::something`
One simple point: 
> You recently introduced 'std::' prefixes in all header files. I agree with the idea that it is bad style to force the inclusion of a whole namespace on someone just because he included your header file.

If the headers do `using namespace` / `using std::something`, then it will no longer be possible to reliably find where these `std::` prefixes need to be added to headers.

axxel, 2017-01-29:
I think, we are actually agreeing about a lot more than it seems. :) Looks like more misunderstanding than actual disagreement to me:

> If one header pollutes global namespace, it will be polluted everywhere where that header is included from.
> If one source pollutes global namespace, it will be polluted only in that standalone compilation unit.

Absolutely. Maybe the key insight here is that I did not (intend to) suggest to put `use std::string` into the global namespace but in the `RawSpeed` namespace. Maybe I was being imprecise or sloppy with my language somewhere. To remove all misunderstandings, here is my suggestion in an example from Common.h:
```
#include <string>
#include <vector>

namespace RawSpeed {

using std::vector;
using std::string;

inline vector<string> split_string(const string &input, char c = ' ') {
  vector<string> result;
  // somehow fill result
  return result;
}

}
```
The only thing this does is introducing the names `string` and `vector` into the `RawSpeed` namespace and thereby blocking the use of those names inside that namespace for anything else than the std meaning. And it does that for every cpp-file that defines stuff from the `RawSpeed` namespace and includes Common.h, which is likely every single cpp file in the lib, but not a single cpp file from a client code including Common.h.

In contrast, the current code would allow to, e.g. do not put a `using namespace std;` at the top of one of our cpps and then declare it's own something called `string`. But neither of us would like this anyway. So 'blocking' the name `string` in `RawSpeed` inside Common.h is a _good thing_ (TM).

Other than that, is simply removes the need to type and read all those `std::` prefixes in the header code, which is, of course, the main motivation behind my suggestion.

There is no need to put more includes into Common.h than required by Common.h. E.g. the `using std::map` can be placed inside whatever header file is including `<map>`. It might be necessary we'll end up with more than one `using std::map` line, in case independent headers include map separately, no problem there.

> If the headers do using namespace / using std::something, then it will no longer be possible to reliably find where these std:: prefixes need to be added to headers.

I'm not sure what you mean here. If you are using a name the compiler does not know about in a particular namespace, then it will unambiguously tell you about it.

LebedevRI, 2017-01-29:
> The only thing this does is introducing the names string and vector into the RawSpeed namespace and thereby blocking the use of those names inside that namespace for anything else than the std meaning.

And then consumer includes that header, does `using namespace RawSpeed;`, and for him all these `std` symbols appear in global namespace.

> Other than that, is simply removes the need to type and read all those std:: prefixes in the header code, which is, of course, the main motivation behind my suggestion.

I intentionally added those prefixes.

Look, i don't know else to say this, i **DO NOT** want `using namespace std;` or  `using std:<something>;` in header files. That is all.

axxel, 2017-01-29:
> And then consumer includes that header, does using namespace RawSpeed;, and for him all these std symbols appear in global namespace.

That is right. Next to all other names that we have in RawSpeed.

> I intentionally added those prefixes.

I know and it made sense, you removed the global `using namespace std;`. I was simply suggesting an alternative way reaching the same effect (well, almost, see above) without introducing the verbosity.

> i DO NOT want

I was trying to have a discussion based on technical facts, with the goal to find the best solution. That is all. But if you don't want to engage it that, so be it. You are the boss ;).

I do all this for fun, trying to give back to the open source community on a playing field I know something about. I have no interest at all in 'fighting' over stuff. I'm still looking forward to keep on working together with you and whoever is also interested on this project. So no hard feelings, please :).

closed 2017-10-28 by LebedevRI