darktable-org/rawspeed #57: Hash mismatch between build types

LebedevRI, 2017-01-29:
the hashes seem to differ between ASAN and release builds:
```
WARNING: the following 7 tests have failed:
./Canon/EOS 5D Mark III/5G4A9395.CR2 failed: hash/metadata mismatch
./Canon/EOS 5D Mark IV/B13A0733.CR2 failed: hash/metadata mismatch
./Canon/EOS 60D/IMG_2015.CR2 failed: hash/metadata mismatch
./Canon/EOS 6D/EOS_6D_mRAW.CR2 failed: hash/metadata mismatch
./Canon/EOS 70D/mRAW_CANON_EOS70D_01.CR2 failed: hash/metadata mismatch
./Canon/EOS 7D/RAW_CANON_EOS_7D-mraw.CR2 failed: hash/metadata mismatch
./Canon/EOS 80D/20160623_0001.CR2 failed: hash/metadata mismatch
See rstest.log for details.

```

axxel, 2017-01-29:
That looks like mRaw/sRaw related. I guess the log file says that the md5sums are different but everything else is the same? Do you know when this problem appeared?

LebedevRI, 2017-01-29:
> I guess the log file says that the md5sums are different but everything else is the same?

yes

> Do you know when this problem appeared?

zero clues, may have been like that since always. this was the first time i did such a comparison

axxel, 2017-01-29:
> may have been like that since always.

Or maybe it is not there even now: I can't reproduce this. I am testing my 'own' release build based on some quick and dirty qmake file and that produces the same as the upstream cmake build when set to release type ASAN.

LebedevRI, 2017-01-29:
1. build with gcc, -DCMAKE_BUILD_TYPE=ASAN
2. delete all hashes, generate them
3. build with gcc, -DCMAKE_BUILD_TYPE=Release
4. verify hashes
5. ???
6. profit

axxel, 2017-01-29:
Done. All good. Now where is my money? ;)

LebedevRI, 2017-01-29:
that is bad :(
this is `gcc version 6.3.0 20170124 (Debian 6.3.0-5)`

axxel, 2017-01-29:
I am using `c++ (Ubuntu 5.2.1-22ubuntu2) 5.2.1 20151010` right now.

here is the md5sum from the one failing 80D file: `data md5sum: 65086dcc586860f0feadf1ac7ace6369` Maybe this fits with one or yours.

LebedevRI, 2017-03-20:
Apparently was fixed by 870daf912d84c8f57323f7bfe18a3bc0f3951875
Canon EOS 5D Mark IV  m/s-raws are just completely broken anyway

closed 2017-03-20 by LebedevRI