darktable-org/rawspeed #59: Sony A850/A900 get wrong meta data from cameras.xml

axxel, 2017-01-29:
Working on a simplification of the `TrimSpaces` function, it seems I fixed a bug by accident. The two mentioned sony models get a new hash. But interestingly the new values are exactly those from the cameras.xml. So they must have been wrong before?!?

@LebedevRI Question is: which are the correct ones?

```
--- test/Sony_DSLR-A850_0519.arw.hash	2017-01-29 16:25:17.249804551 +0100
+++ test/Sony_DSLR-A850_0519.arw.hash.failed	2017-01-29 17:47:09.757193209 +0100
@@ -9,2 +9,2 @@
-blackLevel: 125
-whitePoint: 3750
+blackLevel: 500
+whitePoint: 15000
@@ -21 +21 @@
-dimUncropped: 6080x4048
+dimUncropped: 6080x4056
@@ -29 +29 @@
-data md5sum: 1919026cd90ace97aa66743adfd9c55a
+data md5sum: 5810840e9c542b9b51ffe9222e7ff5da
--- test/Sony_DSLR-A900_0520.arw.hash	2017-01-29 16:25:20.481738480 +0100
+++ test/Sony_DSLR-A900_0520.arw.hash.failed	2017-01-29 17:47:09.725193869 +0100
@@ -9,2 +9,2 @@
-blackLevel: 130
-whitePoint: 4095
+blackLevel: 520
+whitePoint: 16383
@@ -21 +21 @@
-dimUncropped: 6080x4048
+dimUncropped: 6080x4056
@@ -29 +29 @@
-data md5sum: 3c2b50754a57bc57388fecc06d38ef20
+data md5sum: 0b3860cd1ad251c38e80a0cf6c1ab811
```

The filenames are local renames but the make/model are enough to identify them.

LebedevRI, 2017-02-01:
the cameras.xml says that it should be:
```
	<Camera make="SONY" model="DSLR-A850">
		<ID make="Sony" model="DSLR-A850">Sony DSLR-A850</ID>
		<CFA width="2" height="2">
			<Color x="1" y="1">BLUE</Color>
			<Color x="1" y="0">GREEN</Color>
			<Color x="0" y="1">GREEN</Color>
			<Color x="0" y="0">RED</Color>
		</CFA>
		<Crop x="0" y="0" width="6080" height="4048"/>
		<Sensor black="500" white="15000"/>
	</Camera>
	<Camera make="SONY" model="DSLR-A900">
		<ID make="Sony" model="DSLR-A900">Sony DSLR-A900</ID>
		<CFA width="2" height="2">
			<Color x="0" y="0">RED</Color>
			<Color x="1" y="0">GREEN</Color>
			<Color x="0" y="1">GREEN</Color>
			<Color x="1" y="1">BLUE</Color>
		</CFA>
		<Crop x="0" y="0" width="6080" height="4048"/>
		<Sensor black="520" white="16383"/>
	</Camera>
```

LebedevRI, 2017-02-01:
Exif says that they are 12-bit, so the old black/white levels look more correct
gonna check in dt next

LebedevRI, 2017-02-01:
@axxel for RAW_SONY_A850.ARW and RAW_SONY_A900.ARW from rpu, the
```diff
-blackLevel: 125
-whitePoint: 3750
```
and
```diff
-blackLevel: 130
-whitePoint: 4095
```
are the right ones.
with the values you pasted, images look completely crippled.

axxel, 2017-02-01:
OK, so the numbers in the xml are wrong. The code I pushed did not change the behavior. So the old numbers are still used, wherever they are actually from (did not investigate).

LebedevRI, 2017-02-01:
> OK, so the numbers in the xml are wrong. 

No, they are not. Please read the ArwDecoder once more :)
Especially grep for mShiftDownScale.

The metadata in xmp (for arw raws in general it seems) is specified for 14-bit raws.
These raws are 12-bit. So for them the metadata is computed by simply `>> 2`.

axxel, 2017-02-01:
> No, they are not. Please read the ArwDecoder once more :)

I see. Makes sense. As I said: I did not investigate further. So my worries were unfounded.

closed 2017-02-01 by axxel