darktable-org/rawspeed #62: Variable initialization syntax consistency

axxel, 2017-01-30:
I started introducing c++11 default member initializations a while back. I used the 'old' syntax:
```
int x = 0;
```
I noticed @LebedevRI also added some new code but chose the 'new' uniform initialization syntax:
```
int x{0};
```
I personally prefer the first because it is easier for my eye to parse it. I can also type it way faster. It's a matter of style. I'd like to use it consistently across the code base, but then ideally not only for default member initialization inside a class but like it was invented: uniformly. This would also mean using it for any other kind of object initialization:
```
   ByteStream bs{t->getData()};
   TiffIFD* p{parent};
   return new DngDecoder{move(root), mInput};
```
Disclamer: I dislike those 3 lines so much, that I'd still rather use it only in class definitions and not everywhere. My point being: can we just agree on the `type var = value;` version. Please? :)

LebedevRI, 2017-02-01:
> chose the 'new' uniform initialization syntax:
> `int x{0};`

The bulk of these changes used this syntax because that is the syntax that clang-tidy suggested.
i.e. i don't *really* have any real preference to the ether syntax.

> but then ideally not only for default member initialization inside a class but like it was invented: uniformly. This would also mean using it for any other kind of object initialization:

Hmm, sounds too restrictive. It' not like there is any hard-required coding style for rawspeed (and no, i don't think there will be any hard-required coding style.), so it will be pretty much impossible to enforce.

>ByteStream bs{t->getData()};
>TiffIFD* p{parent};
>return new DngDecoder{move(root), mInput};

Ok, now **that** is just plain ugly.
I don't really insist on using `{}` in classes. But here, `()` should be used unless `{}` are really needed.

axxel, 2017-02-01:
> and no, i don't think there will be any hard-required coding style

Fine, I was just worried that if I would have come up with a patch set that used my preferred style  (`=`) that you might have objections based on what you committed in the mean time.

closed 2017-02-01 by axxel

LebedevRI, 2017-02-01:
No, i don't think i will object to using `=` for in-class initialization. :)