darktable-org/rawspeed #63: Lossless compressed DNG from Adobe DNG converter broken

LebedevRI, 2017-01-31:
@axxel 

When working towards minimizing hard-require dependencies for msvs ci build, i noticed that there is no lossy jpeg samples. I used ADC to produce all the variants of the compression it can, and the lossless compression is broken in current git develop branch.

the sample is [5G4A9394-compressed-lossless.DNG](https://raw.pixls.us/getfile.php/1024/raw/5G4A9394-compressed-lossless.DNG)

```
=================================================================
==12936==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x7fee514db780 at pc 0x0000006d1e11 bp 0x7fee4e83d490 sp 0x7fee4e83d488
WRITE of size 2 at 0x7fee514db780 thread T1
    #0 0x6d1e10 in void RawSpeed::LJpegPlain::decodeN_X_Y<2, 1, 1>()::{lambda(int)#2}::operator()(int) const /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegPlain.cpp:208:21
    #1 0x6d1c1f in RawSpeed::unroll_loop_t<void RawSpeed::LJpegPlain::decodeN_X_Y<2, 1, 1>()::{lambda(int)#2}, 1ul>::repeat({lambda(int)#2} const&) /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegPlain.cpp:109:5
    #2 0x6d1be4 in RawSpeed::unroll_loop_t<void RawSpeed::LJpegPlain::decodeN_X_Y<2, 1, 1>()::{lambda(int)#2}, 2ul>::repeat({lambda(int)#2} const&) /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegPlain.cpp:108:5
    #3 0x6d1a04 in void RawSpeed::unroll_loop<2ul, void RawSpeed::LJpegPlain::decodeN_X_Y<2, 1, 1>()::{lambda(int)#2}>(void RawSpeed::LJpegPlain::decodeN_X_Y<2, 1, 1>()::{lambda(int)#2} const&) /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegPlain.cpp:120:3
    #4 0x6cde59 in void RawSpeed::LJpegPlain::decodeN_X_Y<2, 1, 1>() /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegPlain.cpp:207:11
    #5 0x6c90e1 in RawSpeed::LJpegPlain::decodeScan() /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegPlain.cpp:90:7
    #6 0x6b9f10 in RawSpeed::LJpegDecompressor::parseSOS() /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegDecompressor.cpp:131:3
    #7 0x6b8665 in RawSpeed::LJpegDecompressor::decode(unsigned int, unsigned int) /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegDecompressor.cpp:53:18
    #8 0x5f189a in RawSpeed::DngDecoderSlices::decodeSlice(RawSpeed::DngDecoderThread*) /home/lebedevri/rawspeed/src/librawspeed/decoders/DngDecoderSlices.cpp:424:11
    #9 0x5f02e7 in RawSpeed::DecodeThread(void*) /home/lebedevri/rawspeed/src/librawspeed/decoders/DngDecoderSlices.cpp:64:13
    #10 0x7fee5689a423 in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x7423)
    #11 0x7fee559239be in clone (/lib/x86_64-linux-gnu/libc.so.6+0xe89be)

0x7fee514db780 is located 0 bytes to the right of 46768000-byte region [0x7fee4e841800,0x7fee514db780)
allocated by thread T0 here:
    #0 0x542eb0 in posix_memalign (/home/lebedevri/rawspeed/build/src/utilities/identify/darktable-rs-identify+0x542eb0)
    #1 0x65bc69 in _aligned_malloc(unsigned long, unsigned long) /home/lebedevri/rawspeed/src/librawspeed/common/Common.cpp:82:12
    #2 0x58623a in RawSpeed::RawImageData::createData() /home/lebedevri/rawspeed/src/librawspeed/common/RawImage.cpp:92:20
    #3 0x5de5df in RawSpeed::DngDecoder::decodeData(RawSpeed::TiffIFD*) /home/lebedevri/rawspeed/src/librawspeed/decoders/DngDecoder.cpp:149:9
    #4 0x5e0b61 in RawSpeed::DngDecoder::decodeRawInternal() /home/lebedevri/rawspeed/src/librawspeed/decoders/DngDecoder.cpp:300:7
    #5 0x631094 in RawSpeed::RawDecoder::decodeRaw() /home/lebedevri/rawspeed/src/librawspeed/decoders/RawDecoder.cpp:304:20
    #6 0x580add in main /home/lebedevri/rawspeed/src/utilities/identify/rawspeed-identify.cpp:167:8
    #7 0x7fee5585b2b0 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x202b0)

Thread T1 created by T0 here:
    #0 0x4a505d in pthread_create (/home/lebedevri/rawspeed/build/src/utilities/identify/darktable-rs-identify+0x4a505d)
    #1 0x5f3bd3 in RawSpeed::DngDecoderSlices::startDecoding() /home/lebedevri/rawspeed/src/librawspeed/decoders/DngDecoderSlices.cpp:118:5
    #2 0x5df98b in RawSpeed::DngDecoder::decodeData(RawSpeed::TiffIFD*) /home/lebedevri/rawspeed/src/librawspeed/decoders/DngDecoder.cpp:226:10
    #3 0x5e0b61 in RawSpeed::DngDecoder::decodeRawInternal() /home/lebedevri/rawspeed/src/librawspeed/decoders/DngDecoder.cpp:300:7
    #4 0x631094 in RawSpeed::RawDecoder::decodeRaw() /home/lebedevri/rawspeed/src/librawspeed/decoders/RawDecoder.cpp:304:20
    #5 0x580add in main /home/lebedevri/rawspeed/src/utilities/identify/rawspeed-identify.cpp:167:8
    #6 0x7fee5585b2b0 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x202b0)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/lebedevri/rawspeed/src/librawspeed/decompressors/LJpegPlain.cpp:208:21 in void RawSpeed::LJpegPlain::decodeN_X_Y<2, 1, 1>()::{lambda(int)#2}::operator()(int) const
Shadow bytes around the buggy address:
  0x0ffe4a2936a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffe4a2936b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffe4a2936c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffe4a2936d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ffe4a2936e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0ffe4a2936f0:[fa]fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffe4a293700: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffe4a293710: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffe4a293720: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffe4a293730: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0ffe4a293740: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==12936==ABORTING
```

axxel, 2017-01-31:
I'll push a fix. Most likely later today.

axxel, 2017-01-31:
Question: what happens if you convert an mRaw/sRaw to dng? In case that is even possible. Could you provide a sample for that as well?

LebedevRI, 2017-01-31:
https://www.dropbox.com/sh/x1pa1814zojyc86/AADE1deE3OnggYAwH7yorc4na?dl=0

axxel, 2017-02-01:
Fixed. All samples look fine to me now. The uncompressed and compressed-lossless versions produce the same hash file.

LebedevRI, 2017-02-01:
The m/s-raw dng, are they different from the full raw dng's?

axxel, 2017-02-01:
> The m/s-raw dng, are they different from the full raw dng's?

Yes. They contain RGB data for each pixel, no subsampling/CFA. So the ADC basically rendered them to RGB and stored them with the LJpeg compression algorithm. These are the only ones using 3 component non-subsampled data.

LebedevRI, 2017-02-01:
What i am trying to understand is, is there any point in adding those samples i put on dropbox to the rpu?

axxel, 2017-02-01:
> So yes, they are unique, and are wanted for rpu, correct?

If you want to support them (which seems obvious): yes, absolutely.

axxel, 2017-02-01:
Appendix: IIRC the two files (mRaw vs sRaw) are the same in terms of code coverage, so one of them should be enough.

LebedevRI, 2017-02-01:
Okay, uploaded, thanks.

PR #64 (2017-02-01): Fix overlapping tiled ljpeg compressed dngs
Fix #63 and thereby splitting LJpegPlain in two separate classes dealing with almost completely different use cases. Details see individual commits.
commit b766e105 (2017-02-01): Fix overlapping tiled LJpeg compressed DNGs

Border tiles of tiled DNGs may overlap the actual raw image buffer. This
has to be dealt with by decoding but then discarding the pixels that are
overlapping the raw image frame. Fixes #63. It also fixes m/sRaw files
that were recoded into RGB DNGs.

The uncompressed and the compressed-lossless versions of the EOS 5D
files 5G4A9395/5G4A9396 produce the same hash now.

Since this was yet another special casing situation that was affecting
only one of the two use cases (DNG vs CR2), a DNG specific loop has been
added that is a lot easier to read/under [... cut]
closed 2017-02-01 by LebedevRI