darktable-org/rawspeed #73: DNG image is upside down

supertobi, 2017-02-05:
I converted a GoPro .gpr image with Adobe DNG Converter 9.8 to a .dng file. The files are upside down in darktable 2.2.1.

https://filebin.net/7h8d4s8vmw6z7gtn/GOPR0020.dng
As github doesn't support uploading dng files I used filebin. The files will expire automatically 2 months from now. Please ping me if it is expired and you need it again.

LebedevRI, 2017-02-05:
@supertobi so it is decoded correctly, but in darktable, it is upside down?
Then it is not a rawspeed issue, but darktable.
Could you please open a new issue at https://redmine.darktable.org/projects/darktable/issues/new ?

closed 2017-02-05 by LebedevRI

supertobi, 2017-02-05:
done:
https://redmine.darktable.org/issues/11499