darktable-org/rawspeed #74: DNG Converter DNGS: Please consider providing samples on <https://raw.pixls.us/>, thanks!

supertobi, 2017-02-05:
I converted a GoPro .gpr image with Adobe DNG Converter 9.8 to a .dng file. now I get this message from darktable:

> RawSpeed:Unable to find camera in database: 'GoPro' 'HERO5 Black' 'dng'
> Please consider providing samples on <https://raw.pixls.us/>, thanks!

As raw.pixls.us is  not interested in DNG files converted from Adobe DNG Converter, aren't you able to not show this message, if the DNGs are from the Adobe DNG Converter?

axxel, 2017-02-05:
@LebedevRI noticed this contradiction right after he introduced the message: #20 :). Simply has not come around to change it, I guess.

LebedevRI, 2017-02-05:
yes, what @axxel said :)
We should not ask for dng samples from ADC, and some other software (hdrmerge, dt)

LebedevRI, 2017-02-05:
Closing as duplicate of https://github.com/darktable-org/rawspeed/issues/20

closed 2017-02-05 by LebedevRI