darktable-org/rawspeed #79: Bad pixel correction for DNGs is broken

axxel, 2017-02-09:
See https://redmine.darktable.org/issues/10078.

Example image: https://raw.pixls.us/getfile.php/683/raw/RAW_CANON_POWERSHOT_SX130IS_CHDK.DNG

I stumbled on this issues while cleaning up the DngOpcode classes. This has almost certainly never worked (I found 2 separate issues).

I have a complete rewrite of the DngOpcode classes, fixing this issue (among others). Just waiting for #75 to be merged.

LebedevRI, 2017-02-09:
Coverity CID 1025896 and 1025897 are probably related.

axxel, 2017-02-09:
> Coverity CID 1025896 and 1025897 are probably related.

Yes and no:
 * affecting some kind of bad pixel correction (float images): yes
 * completely preventing bad pixel correction of DNGs: no

PR #80 (2017-02-25): Reimplement dng opcodes
Fix #79 and a bunch of other issues (interface bloat, copy'n'paste errors, code duplication, leaks on exceptions). Details see commit messages.
commit 3cfe4da1 (2017-02-25): Reimplement DngOpcode classes

The former DngOpcode classes used yet another version of some byte
swapping code and it manually dealt with keeping track of the current
position in the byte stream. Both is readily available in the ByteStream
class.

Also there has been a lot of duplicated code in the opcode
implementations that differ only slightly from one another. The new code
is less than half the size of the old.

The internal interface were unnecessarily complex and have been
simplified.

Due to some design flaw (all opcodes had some area-of-interest member
which was only used by some) the [... cut]
closed 2017-02-25 by LebedevRI