darktable-org/rawspeed #96: Canon EOS 5D Mark IV m/s-raws broken

LebedevRI, 2017-03-20:


closed 2017-03-28 by LebedevRI

LynxChaus, 2017-03-29:
Still not load raw from Canon 700D.

```
$ darktable-rs-identify Pictures/IMG_3030.CR2
Loading file: "Pictures/IMG_3030.CR2"
make: Canon
model: Canon EOS 700D
canonical_make: Canon
canonical_model: EOS 700D
canonical_alias: EOS 700D
ERROR: [rawspeed] RawSpeed::RawImage RawSpeed::RawDecoder::decodeRaw(), line 305: const uchar8* RawSpeed::Buffer::getData(RawSpeed::Buffer::size_type, RawSpeed::Buffer::size_type) const, line 103: Buffer overflow: image file may be truncated
```

LebedevRI, 2017-03-29:
@LynxChaus 
1. open new issue
2. the sample from raw.pixls.us, [IMG_0972.CR2](https://raw.pixls.us/getfile.php/1283/raw/IMG_0972.CR2) loads just fine, so you need to provide the sample here.

LynxChaus, 2017-03-29:
Ok, open #97

LebedevRI, 2017-03-29:
Are *all* the raw files from that camera are like that?
Have you tried different SD card, newer firmware?
Either the file is slightly truncated, or the camera wrote corrupted file.

commit 71d844e6 (2017-03-28): Cr2Decompressor::decodeN_X_Y(): drop bogus '6D' mRaw w/h swap.

Does not break any hashes. Unbreaks mRAW from 5Dmk4 and 5DS R.

Fixes #96
Refs. #57
Refs. #39