darktable-org/rawspeed #97: Can't load raw from Canon 700D

LynxChaus, 2017-03-29:
```
$ darktable-rs-identify Pictures/IMG_3030.CR2
Loading file: "Pictures/IMG_3030.CR2"
make: Canon
model: Canon EOS 700D
canonical_make: Canon
canonical_model: EOS 700D
canonical_alias: EOS 700D
ERROR: [rawspeed] RawSpeed::RawImage RawSpeed::RawDecoder::decodeRaw(), line 305: const uchar8* RawSpeed::Buffer::getData(RawSpeed::Buffer::size_type, RawSpeed::Buffer::size_type) const, line 103: Buffer overflow: image file may be truncated
```
Raw direct from camera, manual M42 lens with chinese "focus assist" chip on ring.
Grab raw here: [IMG_3030.CR2](https://yadi.sk/i/2qlwNkbG3GTRFZ)

LebedevRI, 2017-03-29:
Are all the raw files from that camera are like that?
Have you tried different SD card, newer firmware?
Either the file is slightly truncated, or the camera wrote corrupted file.

LynxChaus, 2017-03-30:
Files damaged when transferred over wire. 
Sorry for noise.

LebedevRI, 2017-03-30:
> Files damaged when transferred over wire.

I'm sorry to hear that...
However in this case, given that the file is just few bytes short, i guess rawspeed should not completely refuse to load it.
I have an idea how to workaround this particular case, so i'll keep the issue for now.