darktable-org/rawspeed #98: Simplify&parallelize Cr2Decompressor

LebedevRI, 2017-03-29:
Since predictor is reset for each row of the slice to the first pixel of previous slice row, i _believe_ it means only the first pixel of the row depends on the value of the first pixel of the previous row. Thus it should be feasible to parallelize it?

axxel, 2017-03-31:
> i believe it means only the first pixel of the row depends on the value of the first pixel of the previous row

That is correct.

> Thus it should be feasible to parallelize it?

Unfortunately not: you need to decode each and every pixel of the full row, just to know where exactly in the bitstream the next row starts.

LebedevRI, 2017-03-31:
> Unfortunately not: you need to decode each and every pixel of the full row, just to know where exactly in the bitstream the next row starts.

Who said anything about decoding huffmantable in parallel too?