darktable-org/rawspeed #100: Regression: huffmantable is broken.

LebedevRI, 2017-04-18:


LebedevRI, 2017-10-28:
Apparently happens for [Nikon D3400](https://redmine.darktable.org/issues/11791) too. 
Really not cool.

LebedevRI, 2017-10-28:
So basically all nikon raws "lossy after split" are broken.
https://github.com/darktable-org/rawspeed/blob/46e80e0624acf65043893f9e708fd701bdbba46b/src/librawspeed/decompressors/NikonDecompressor.cpp#L39-L41
https://github.com/darktable-org/rawspeed/blob/46e80e0624acf65043893f9e708fd701bdbba46b/src/librawspeed/decompressors/NikonDecompressor.cpp#L48-L50

axxel, 2017-10-28:
Based on your comments above you seem to make the assumption that my ["fucked"](https://github.com/LebedevRI/darktable/commit/f92f3a2577bda0d2faf73f496b9d687e518dac88) HuffmanTable code broke the "lossy after split" files. For that assumption to be true, it would have needed to work before my changes. Have you tried that?

I have the my doubts: The thing is, when working on the rewrite, I was noticing a (presumed) bug in the old code and wanted to investigate further but there was no sample available triggering that bug. In fact, [I asked you personally](https://github.com/klauspost/rawspeed/issues/169) for one, but you did not have one either.

Now that there seems to be sample available, I'm pretty sure you'll manage to "completely unfuck" my HuffmanTable implementation. Maybe my mentioned analysis from last year can be of some help, too, who knows.

LebedevRI, 2017-10-29:
> Based on your comments above you seem to make the assumption that my "fucked" HuffmanTable code broke the "lossy after split" files. For that assumption to be true, it would have needed to work before my changes. Have you tried that?

I absolutely did. darktable-2.2.5, that still had the old rawspeed, loads these files flawlessly.
Feel free to try yourself and catch me on a lie.

> I have the my doubts: The thing is, when working on the rewrite, I was noticing a (presumed) bug in the old code and wanted to investigate further but there was no sample available triggering that bug. In fact, I asked you personally for one, but you did not have one either.

https://github.com/darktable-org/rawspeed/commit/b213eafc6b3d4a565e52aa97b95872adebcc1879#diff-98d884ed1d3efccaf22be005567a6726R106
```diff
+  // 2. This is the actual huffman encoded data, i.e. the 'alphabet'. Each value
+  // is the number of bits following the code that encode the difference to the
+  // last pixel. Valid values are in the range 0..16.
```
Now, what is the maximal code value (the second array) for `12-bit lossy after split`
https://github.com/darktable-org/rawspeed/blob/46e80e0624acf65043893f9e708fd701bdbba46b/src/librawspeed/decompressors/NikonDecompressor.cpp#L39-L41
, or for `14-bit lossy after split`
https://github.com/darktable-org/rawspeed/blob/46e80e0624acf65043893f9e708fd701bdbba46b/src/librawspeed/decompressors/NikonDecompressor.cpp#L48-L50
?

> Now that there seems to be sample available, I'm pretty sure you'll manage to "completely unfuck" my HuffmanTable implementation. Maybe my mentioned analysis from last year can be of some help, too, who knows.

Yes, i do hope so. Hopefully even without sacrificing performance.

LebedevRI, 2017-10-29:
https://discuss.pixls.us/t/nikon-a-specific-raw-sample-wanted/5483

axxel, 2017-10-30:
> Feel free to try yourself and catch me on a lie.

?!? Why would I have reason to question the truthfulness of your statements?

Given that it worked for the old code, I must have made a mistake in my analysis that lead to me suspecting the code was not working for that special case. As you might have inferred from my awareness of the fact that the "lossy after split" stuff seemed fishy, I was deliberately ignoring the existence of those invalid lossy after split numbers. It made no sense to me to try to retain some functionality that could not be tested _and_ I suspected to be not working anyway. Much the same reasoning that presumably led you to drop a whole decoder recently.

I had another quick look at the old code to see if I was able to find the fault in my earlier reasoning. Now I'm thinking, the fault might have been that I thought the "big table" was used for decoding the whole image (which could not have worked) but now it seems to me that simply every lookup in the big table for the image part after the split would simply "fail" (which is supposed to happen if the code is too big for the "big table") and then the special/custom `shl != 0` stuff would kick in.

LebedevRI, 2017-11-30:
And one more broken piece, from https://discuss.pixls.us/t/darktable-for-windows/4966/264
[DSC_0143.dng.gz](https://github.com/darktable-org/rawspeed/files/1518510/DSC_0143.dng.gz)
`HuffmanTable::decode() {
...
bs.fill(32);` is the culprit, but even then something else is broken still.
Probably related to #94

@LibRaw i really do hope you weren't planning on upgrading anytime soon :)

LibRaw, 2017-11-30:
Yes, it looks like we'll skip RawSpeed(v2) support for LibRaw 0.19

We'll try to implement some kind of 'flexible' support. May be via a whitelist of cameras w/o known problems and with sensor megapixel large enough for significant speedup (so, Canon 5DS*, Nikon D8xx, may be Canon 5Dmk4).

LebedevRI, 2017-11-30:
> RawSpeed(v2)

I'm not aware which commit was the v2 release btw.
Current tag is v3.0, and i'm following the same scheme as with darktable - odd tag versions mean development version, even - release.

LibRaw, 2017-12-01:
v1/v2 was used as 'conventional name' for 'Klaus master' and 'current devel' branches.

Is (some) current branch already used in some darktable release?

LebedevRI, 2017-12-01:
> Is (some) current branch already used in some darktable release?

Yeah, darktable git master is using rawspeed git develop since 
```
commit 3b4c49d16b859758ddeccdf8a261719ac3f0c5f9 (upstream/pr/1402)
Author: Roman Lebedev <lebedev.ri@gmail.com>
Date:   Mon Jan 2 17:03:41 2017 +0300

    Track rawspeed as submodule

commit b82d189e00a8741cd1fce5207f60814c1959cc7f
Author: Roman Lebedev <lebedev.ri@gmail.com>
Date:   Mon Jan 2 17:02:04 2017 +0300

    Purge bundled rawspeed
```
So 2.4.0 series will be using (and current 2.4.0rc0 *is* using) this code.

axxel, 2017-12-01:
> And one more broken piece, from https://discuss.pixls.us/t/darktable-for-windows/4966/264
> DSC_0143.dng.gz

I don't use darktable and my rstest has no problem extracting some whitebalance information out of that file. Can you provide some steps to reproduce the issue with using only librawspeed tools?

> HuffmanTable::decode() { ... bs.fill(32); is the culprit, but even then something else is broken still.
> Probably related to #94

I can decode that dng but it contains a few apparently wrongly decoded tiles. Do you see any connection between the nikon lossy-after-split behavior and the issues with that DNG file? Can you elaborate what the problem with the `bs.fill(32);` line is in your opinion?

LebedevRI, 2017-12-01:
@axxel i'm sorry for bothering you, only the original comment in this issue was directed at you, the rest of the comments i'm simply using to record my findings.

axxel, 2017-12-01:
No worries, I'm not bothered. As I said earlier this year, I'd be happy to look into any issues with my code, in case something shows up later.

ccmoralesj, 2018-03-01:
I was trying to process a .NEF RAW photo from my Nikon D5600 and I got this message:

> failed to read camera white balance information from `DSC_0371.NEF

The console error I got was:
```[rawspeed] (DSC_0371.NEF) void rawspeed::HuffmanTable::setCodeValues(const rawspeed::Buffer&), line 176: Corrupt Huffman. Code value 92 is bigger than 16
[temperature] failed to read camera white balance information from `DSC_0371.NEF'!
[temperature] `NIKON CORPORATION NIKON D5600' color matrix not found for image
[temperature] failed to read camera white balance information from `DSC_0371.NEF'!
[rawspeed] (DSC_0372.NEF) void rawspeed::HuffmanTable::setCodeValues(const rawspeed::Buffer&), line 176: Corrupt Huffman. Code value 92 is bigger than 16```

It seems that here is the issue I'm looking for. I'll wait for support or upgrade for this.

PS. I would be glad to share with you (if needed) photos that triggers this error.

LebedevRI, 2018-03-01:
Sorry for the trouble :/

> It seems that here is the issue I'm looking for. 

I think so. Though there is more than one issue here.
Some part of it (not the nikon part you are seeing) is fixed in my [WIP branch](https://github.com/LebedevRI/rawspeed/tree/huffmantable)

> PS. I would be glad to share with you (if needed) photos that triggers this error.

Thanks, but for D5600, all the needed samples are already present on RPU.

exaexa, 2018-03-21:
+1 also waiting for the fix.

Btw @LebedevRI some months ago we've found a workaround to this issue for Nikon J1's (a one-liner patch that I accidentally lost), would you happen to remember what the problem was?

LebedevRI, 2018-03-21:
Uhh, if it's one-liner i think it would be changing https://github.com/darktable-org/rawspeed/blob/d41fd6dadc161d1c0bb56463dc35263779fa02f1/src/librawspeed/decompressors/HuffmanTable.h#L309
```diff
-bs.fill(32);
+bs.fill(maxCodePlusDiffLength());
```

exaexa, 2018-03-21:
@LebedevRI  Fixed the issue. Thanks a lot. :+1: 

(btw -- if that helps someone -- the point where this breaks for me is in BitStream.h, the error I receive is from this IOE:

https://github.com/darktable-org/rawspeed/blob/develop/src/librawspeed/io/BitStream.h#L128

from GDB I could see that pos was larger than size (exactly pos was size+4) when the exception was thrown.)

ralfbrown, 2018-05-14:
I just uploaded a D7000 14-bit "lossy after split" image to RPU (though it's not showing in the repo search yet -- looks like that's updated once a day?).  I discovered that even last night's git master still breaks on them when I went back to revisit some old photos.

shoffmeister, 2018-05-16:
Would there be any chance that the patch from above

-bs.fill(32);
+bs.fill(maxCodePlusDiffLength());

gets dropped into develop / into some darktable release? It's difficult for me to set up a build environment (on Windows)...

LebedevRI, 2018-05-16:
> I just uploaded a D7000 14-bit "lossy after split" image to RPU

Thanks!

> (though it's not showing in the repo search yet -- looks like that's updated once a day?

There is a pre-review.

> Would there be any chance that the patch from above
> -bs.fill(32);
> +bs.fill(maxCodePlusDiffLength());

It's *quite* broken beyond that i'm afraid.
Now that i finally bothered to push the NFC test changes (https://github.com/darktable-org/rawspeed/compare/f5eeb2dee1bfe8e16496c97c91d179d60c884b42...9a520178f355c5bc2a68eb3c82c12e256a9ab5df),
i just need one more push to finish the more major part of the fix
(https://github.com/darktable-org/rawspeed/compare/develop...LebedevRI:huffmantable), that should bring it closer to life..

ccmoralesj, 2018-07-24:
I was wondering... Is this bug fixed, yet? and added to release candidate?

LebedevRI, 2018-07-25:
Fully - no, not yet.
I've fixed the base glaring issues in #130, but as cross-fuzzers (between two different huffman table implementations) tell me, there are still _issues_.
i'm going to re-look into the nikon issue (#129) next.

exaexa, 2018-07-25:
regarding nikon, please ask if you'd need NEFs that fail to decode; I can generate them on demand.

LebedevRI, 2018-07-25:
Which camera? There are ~6 such raws on RPU already, so i think *now* * the samples are not an issue

* If only those samples were in rawsamples.ch back then :/
   transition of rawspeed, and a fork fo rawsamples.ch -> RPU happened at the same time

LebedevRI, 2018-07-25:
Which camera? There are ~6 such raws on RPU already, so i think *now* * the samples are not an issue

EDIT: right, J1

* If only those samples were in rawsamples.ch back then :/
   transition of rawspeed, and a fork fo rawsamples.ch -> RPU happened at the same time

exaexa, 2018-07-25:
Yeah, J1. I've already put a sample from that to RPU. If I find any other camera doing the same I'll post it.

m-mmm, 2018-08-14:
Is there a way to test rawspeed on a raw file using command line?

My idea is to convert affected files to DNG. I would be better than converting all my files.

Also: I have a lot of affected D5500 samples with no people in, if needed I can provide those.

LebedevRI, 2018-08-14:
Sure. `darktable-rs-identify` tool if you are using darktable.

m-mmm, 2018-08-14:
How do I use it? It always shows this message:
WARNING: Couldn't find cameras.xml in 'C:/darktable/darktable-install/share/darktable/rawspeed/cameras.xml'
WARNING: Couldn't find cameras.xml in 'darktable-rs-identify.exe/../share/darktable/rawspeed/cameras.xml'
ERROR: Couldn't find cameras.xml in 'darktable-rs-identify.exe/../share/darktable/rawspeed/cameras.xml'
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid

OK --> all hard coded paths are not valid, creating C:\Program Files\darktable\share\darktable\rawspeed myself and copying cameras.xml to it.

LebedevRI, 2018-08-14:
No idea, i don't use windows, i can't help here.

dsaronin, 2018-10-22:
Is there any type of workaround possible for this error until the bug gets fixed?
Is it possible to use the white balance info from another RAW NEF file which isn't broken?
I see that Darktable is able to generate a thumbnail image for the file.

dsaronin, 2018-10-22:
> It's not about white balance. That dt's message is
> uninformative/misleading i'd say.
> It quite literally fails to decode the whole image.
> No. Those are exif2 thumbnails, which are shown by default unless you opt
> into raw preview.
> *Which* bug are you seeing? I.e. can you show the error console output?
>    - use older darktable (2.2.x should be last good i think?)
>    - convert those specific nef's to dng via adobe dng converter/etc
> Right now I'm dead in the water and unable to edit some files from the

Thanks for the response.
The error message was something about an error with the White Balance data.

I see that POLARR has no problem accessing the NEF raw files, so I can
temporarily use that, even though I prefer DarkTable.

LebedevRI, 2018-10-30:
FYI i have finally (sigh) restored then nikon lossy-after-split support (#129).
So the _main_ remaining problem is #128, the last decode.
Not quite sure what is up with #127, might be a semi-duplicate of #132.

m-mmm, 2018-10-30:
Hi,

 

this sounds wonderful! Thank you.

 

Did you read my mail about this script which can mass-test files with rawspeed? I'd like to share it but I'm not sure how to properly do that.

 

BR, Micha.

 

Gesendet: Dienstag, 30. Oktober 2018 um 10:30 Uhr

LebedevRI, 2018-10-30:
> Did you read my mail

Uh, nothing recent comes to mind..

> about this script which can mass-test files with rawspeed? I'd like to share it but I'm not sure how to properly do that.

In principle, you can just pass `-DRAWSPEED_ENABLE_SAMPLE_BASED_TESTING=ON -DRAWSPEED_REFERENCE_SAMPLE_ARCHIVE=<raw.pixls.us checkout>` when building,
and use `ninja rstest-{create,recreate,test,check,clean}` build targets.
(see [sample-based-testing.cmake](https://github.com/darktable-org/rawspeed/blob/b12be2298333174e62c042af28c10511cdf6b436/cmake/sample-based-testing.cmake))
Although locally i have an ugly contraption of a shell script to do that.

m-mmm, 2018-10-30:
Hi Roman,

 

I sent this to the dev mailing list:

 

[darktable-dev] Script for checking RAW files rs-check.py

Hi,

I have lots of pictures which are affected by a bug that DT can't process it.
The problem is in the rawspeed module. E.g.:
https://redmine.darktable.org/issues/12152

My workaround is a python script which uses darktable-rs-identify to check all
files in a folder. It creates a report and optionally moves all buggy files to a
subfolder. Including .xmp files if present.

So I can convert all buggy files to DNG, and only those which have a problem.

I'd like to share this script because many have this issue.

How do I share this script?
License should be GPL2?

Requires python 3.5+
Tested on Windows and Linux.

BR, Micha.

 

It is also suitable for end users to check and separate files with dt problems. For me it is a time and frustration saving workaround :-)

 

I never built DT myself. I have only used Windows installer and Gentoo build so far.

 

BR, Micha.

 

Gesendet: Dienstag, 30. Oktober 2018 um 10:42 Uhr

LebedevRI, 2018-10-30:
> I sent this to the dev mailing list:

Ah, that explains why i have not seen it.
I would have guessed you could simply attach it to the mail to the mailing list.

m-mmm, 2018-10-30:
OK, I can send it in the evening.
 

Gesendet: Dienstag, 30. Oktober 2018 um 11:05 Uhr

m-mmm, 2018-10-30:
Ok, I attached my script.
Version 0.0.3 DRAFT
Required Parameter:  path to folder containing raw files
optional Parameter: -m: move files with errors to a sub dir
It generates 3 report files in the picture folder: a log file, a list of ok files 
and a list of buggy files.
Does not dive into sub dirs.

Required: Python 3.5.x+
Tested on Windows, 2.4.4 install has a bug, so please create a folder C:
\darktable\darktable-install\share\darktable\rawspeed and copy C:\Program 
Files\darktable\share\darktable\rawspeed\cameras.xml into this.
Tested on Linux (Gentoo)

BR, Micha.

Am Dienstag, 30. Oktober 2018, 11:05:19 CET schrieb Roman Lebedev:
> > I sent this to the dev mailing list:
> Ah, that explains why i have not seen it.
> I would have guessed you could simply attach it to the mail to the mailing
> list.

LebedevRI, 2018-10-30:
> Ok, I attached my script.

I meant the dt-dev list.
github does not deal with mail attachments, it just silently drops them.

LebedevRI, 2019-05-01:
4c7496309ec6bd4d113ad960de0b94e4332a5d7c

closed 2019-05-01 by LebedevRI