darktable-org/rawspeed #103: EOS M6 Support

drhaynes, 2017-05-22:
Please add support for the new M6. I have one of these so am able to provide any data / sample images as needed. Just let me know.

LebedevRI, 2017-05-22:
Please open new issue on https://redmine.darktable.org/projects/darktable/issues/new
I somewhat suspect that color matrix (== dt support) is wanted too...

drhaynes, 2017-05-22:
Are you an admin there? I have registered and activated an account but am unable to login. Wondering if there are problems...

LebedevRI, 2017-05-22:
Log-in works for me.

LebedevRI, 2017-05-26:
@drhaynes we will hopefully do dt-2.2.5 this weekend, so you may want to open that issue rather sooner than later :)
Additionally, please take one slightly overexposed raw image per each the iso level that  can be set in camera (including all the sub-iso's, and extended iso levels !) , upload that as one archive somewhere (dropbox?), and give me the link.
That would be all that is needed...

LebedevRI, 2017-06-17:
Meh, that did not close this issue, will have to read up on that.
Fixed in https://github.com/darktable-org/rawspeed/commit/392a09c40a9aa60bed599aeb9ccab76ce815e7a8
Fixed in darktable git master.

closed 2017-06-17 by LebedevRI