darktable-org/rawspeed #104: __builtin_unreachable() not properly handled by Apple LLVM version 6.1.0?

aferrero2707, 2017-05-25:
Compilation of 

```    rawspeed/src/fuzzers/HuffmanTable/main.cpp```

stops with an error like

```    ..../rawspeed/src/fuzzers/HuffmanTable/main.cpp:84:3: error: code will never be
      executed [-Werror,-Wunreachable-code]
  __builtin_unreachable();```

while I guess it should simply proceed with the compilation... can this line be safely commented for not-so-recent versions of clang? Or is there another solution that does not involve upgrading the compiler?

Thanks!

LebedevRI, 2017-05-25:
How are you building?
As part of some other build?
If yes, then i'd recommend https://github.com/darktable-org/darktable/blob/f92f3a2577bda0d2faf73f496b9d687e518dac88/src/external/CMakeLists.txt#L16-L17

LebedevRI, 2017-05-25:
Regardless, i'll add a specialcasing into cmakelists.txt to disable `-Wunreachable-code` if it to complain on that code.
Thank you for the report.

aferrero2707, 2017-05-26:
Thanks, much appreciated!

By the way, since I am using RawSpeed as a bundled library for the RAW decoding in my [image editor](https://github.com/aferrero2707/PhotoFlow/tree/stable), do I need all this fuzzers code or can it be safely disabled?

LebedevRI, 2017-05-26:
@aferrero2707 no, you can safely disable all that
Basically just copy [darktable/src/external/CMakeLists.txt](https://github.com/darktable-org/darktable/blob/master/src/external/CMakeLists.txt), and it should be fine.
It'll be kept updated to only build the library but nothing else.

LebedevRI, 2017-05-26:
@aferrero2707 please do check and confirm that this actually fixed the issue you were hitting.

closed 2017-05-26 by LebedevRI