darktable-org/rawspeed #112: Building error

ChihChengYang, 2017-08-07:
Hi,

I use the commands as below to build rawspeed but getting some errors.
Is that correct rule to build? or something wrong with codes?  

thank you... :)

mkdir build; cd build;
cmake .. -DUSE_BUNDLED_PUGIXML=ON -DALLOW_DOWNLOADING_GOOGLETEST=ON
make

[ 20%] Building CXX object src/librawspeed/CMakeFiles/rawspeed.dir/decompressors/FujiDecompressor.cpp.o
/home/jeff/rawspeed/src/librawspeed/decompressors/FujiDecompressor.cpp: In instantiation of ‘rawspeed::FujiDecompressor::xtrans_decode_block(rawspeed::FujiDecompressor::fuji_compressed_block*, rawspeed::BitPumpMSB*, int) const::<lambda(auto:1&&, rawspeed::FujiDecompressor::_xt_lines, rawspeed::FujiDecompressor::_xt_lines, int, rawspeed::FujiDecompressor::xtrans_decode_block(rawspeed::FujiDecompressor::fuji_compressed_block*, rawspeed::BitPumpMSB*, int) const::ColorPos&, rawspeed::FujiDecompressor::xtrans_decode_block(rawspeed::FujiDecompressor::fuji_compressed_block*, rawspeed::BitPumpMSB*, int) const::ColorPos&)> [with auto:1 = rawspeed::FujiDecompressor::xtrans_decode_block(rawspeed::FujiDecompressor::fuji_compressed_block*, rawspeed::BitPumpMSB*, int) const::<lambda(rawspeed::FujiDecompressor::_xt_lines, rawspeed::FujiDecompressor::_xt_lines, int, rawspeed::FujiDecompressor::xtrans_decode_block(rawspeed::FujiDecompressor::fuji_compressed_block*, rawspeed::BitPumpMSB*, int) const::ColorPos&, rawspeed::FujiDecompressor::xtrans_decode_block(rawspeed::FujiDecompressor::fuji_compressed_block*, rawspeed::BitPumpMSB*, int) const::ColorPos&)>]’:
/home/jeff/rawspeed/src/librawspeed/decompressors/FujiDecompressor.cpp:485:24:   required from here
/home/jeff/rawspeed/src/librawspeed/decompressors/FujiDecompressor.cpp:464:41: error: ‘line_width’ is not captured
     while (g.even < line_width || g.odd < line_width) {
                                         ^
/home/jeff/rawspeed/src/librawspeed/decompressors/FujiDecompressor.cpp:464:19: note: ‘const int& line_width’ declared here
     while (g.even < line_width || g.odd < line_width) {

LebedevRI, 2017-08-07:
Hi.
Please duplicate this issue into your's distribution's bugtracker: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=871250
And pass me the bug number.
After i have a few bugid's, i'll add a temporary workaround.

LebedevRI, 2017-08-07:
So far this is as far as i was able to go:
(#gcc on freenode)
```
[14:28:27] <LebedevRI> hello. is there a gcc bug# for this issue already https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=871250 ? i can only find https://gcc.gnu.org/bugzilla/show_bug.cgi?id=78019 but that is different
[15:15:17] <redi> LebedevRI: I don't recall seeing one.
[15:15:26] <redi> " I'm unable to report this upstream, since registration is closed, and password recovery does not seem to function."
[15:15:26] <LebedevRI> now you do :/
...
[15:21:27] <redi> LebedevRI: anyway, if it's a bug in gcc5 and is fixed in gcc6 then there's no point in an upstream bug report
[15:21:35] <redi> it would probably be closed as FIXED
[15:22:19] <LebedevRI> great. can i quote you on that?
[15:22:49] <redi> sure
[15:23:10] <LebedevRI> problem being, i can't just go and bump compiler requirement up to 6
[15:24:03] <LebedevRI> so the fact that gcc5 is the only one with this issue does not help
```
I'm rather tempted to close this as WONTFIX.

ChihChengYang, 2017-08-08:
Hi,
Ok, got it. Thank you for your information.

closed 2017-08-08 by ChihChengYang

LebedevRI, 2017-08-08:
+ https://bugs.launchpad.net/linuxmint/+bug/1709234
I will likely add a workaround still, but the gcc stance is troubling to say the least.

ChihChengYang, 2017-08-09:
I build this branch https://github.com/LebedevRI/rawspeed can pass the prior error with gcc 5.4.0 but getting another error as below.  gcc-7 , clang++-3.9, 4.0  get the fail as well.  :(

[ 43%] Building CXX object src/librawspeed/test/CMakeFiles/CommonTest.dir/__/common/CommonTest.cpp.o
In file included from /home/jeff/rawspeed/build/googletest/googletest-src/googletest/include/gtest/gtest.h:58:0,
                 from /home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:25:
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp: In member function ‘virtual void rawspeed_test::MakeUniqueTest_Test_Test::TestBody()’:
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:285:14: error: ‘make_unique’ is not a member of ‘std’
     auto s = std::make_unique<int>(0);
              ^

LebedevRI, 2017-08-09:
1. that commit is in this repo, https://github.com/darktable-org/rawspeed/commit/4ccee4a699c87062d7c34cfc5931add2d8c35556
2. building works: for gcc-7, clang-4.0: https://travis-ci.org/darktable-org/rawspeed/builds/262338230
so the problem is local to you. maybe you forgot to prune build dir.

LebedevRI, 2017-08-10:
@ChihChengYang were you able to resolve your build issue?

ChihChengYang, 2017-08-10:
No, that got the same fail even pruning build dir or retrying.

Following make steps: (Ubuntu 16.04 32 or 64bit , gcc 5.4.0 )
1. git clone https://github.com/LebedevRI/rawspeed
2. checkout 4ccee4a
3. mkdir build; cd build;
4. cmake .. -DALLOW_DOWNLOADING_PUGIXML=ON -DUSE_BUNDLED_PUGIXML=ON -DALLOW_DOWNLOADING_GOOGLETEST=ON
5. make

/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp: In member function ‘virtual void rawspeed_test::MakeUniqueTest_Test_Test::TestBody()’:
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:285:14: error: ‘make_unique’ is not a member of ‘std’
     auto s = std::make_unique<int>(0);
              ^
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:285:31: error: expected primary-expression before ‘int’
     auto s = std::make_unique<int>(0);
                               ^
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:286:5: error: template argument 1 is invalid
     ASSERT_EQ(*s, 0);
     ^
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:289:14: error: ‘make_unique’ is not a member of ‘std’
     auto s = std::make_unique<int>(314);
              ^
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:289:31: error: expected primary-expression before ‘int’
     auto s = std::make_unique<int>(314);
                               ^
/home/jeff/rawspeed/src/librawspeed/common/CommonTest.cpp:290:5: error: template argument 1 is invalid
     ASSERT_EQ(*s, 314);
     ^
At global scope:
cc1plus: error: unrecognized command line option ‘-Wno-used-but-marked-unused’ [-Werror]
cc1plus: error: unrecognized command line option ‘-Wno-missing-variable-declarations’ [-Werror]

LebedevRI, 2017-08-10:
Finally, reproduction steps :)
Ok, something is wrong when `-DUSE_BUNDLED_PUGIXML=ON` is specified together with `-DBUILD_TESTING=ON`

LebedevRI, 2017-08-10:
@ChihChengYang please try building `git develop` branch of [*this*](https://github.com/darktable-org/rawspeed) repo.

ChihChengYang, 2017-08-11:
@LebedevRI  The building error has been fixed. Thank you. :)

commit 4ccee4a6 (2017-08-08): FujiDecompressor::xtrans_decode_block(): gcc, die already, please.

Fixes #112