darktable-org/rawspeed #113: Support for Sony RX1R-II

alphazo, 2017-09-12:
I see that the original Sony RX1R is supported but not the RX1R-II (DSC-RX1RM2) that has been released in 2015. Is there any chance to add it or use former RX1R as a baseline ?

Thanks.

LebedevRI, 2017-09-12:
Thank you for reporting this iss.e

We do not have any samples for  `Sony DSC-RX1R` camera on https://raw.pixls.us/
Thus, as far i'm concerned it is not really supported.

Similarly with `DSC-RX1RM2`, until there are samples, there is nothing to talk about.

alphazo, 2017-09-12:
Hello,

I don't own any of those cameras but I'm considering purchasing one if I can find a descent Open Source RAW editor. Darktable has been my workhorse for many years on Nikon and Fuji and would love to continue using it.

Is is ok to post RAW pictures on raw.pixls.us found on websites testing those camera and who usually provide a sample gallery with RAW files?

Thanks
Dany

LebedevRI, 2017-09-12:
Uncompressed raw will certainly work. Not sure about compressed, that is why the samples are needed.

> Is is ok to post RAW pictures on raw.pixls.us found on websites testing those camera and who usually provide a sample gallery with RAW files?

No, it is not legal:

> Upload
> - [ ] I declare that I own full rights to this file and I hereby release it under the CC0 licence into the public domain.

zagaberoo, 2018-04-02:
Just dropped off compressed and uncompressed color chart photos!

Probably should have done this when I bought the camera rather than just patching my own local version to dup the A7r2 data.

zagaberoo, 2018-04-06:
Thanks for integrating!