darktable-org/rawspeed #116: IIQ Regression

bieber, 2017-11-27:
I'm not sure if this technically counts as a regression because there aren't any Credo 40 RAW files in the test set (I added one to raw.pixls.us a few hours ago, I'll attach one to this issue as well), but I believe that commit 65cc3c5e0ccce9bc87c3e80703c7c55e8466c587 broke reading of IIQ files from the Leaf Credo 40 digital back, and I'm guessing all other Credo backs as well.  I *think* the problem is that now that MOS and IIQ reading have been split into separate classes, you're using the manufacturer value to decide which decoder to use, "Leaf" gets the MOS decoder and "Phase One A/S" gets the IIQ decoder.

The problem is that Leaf backs *used* to use MOS files, but the new Leaf Credo backs use IIQ files.  So I'm assuming the fix to this would be to look at the model value as well as the camera value when making this determination.  Presumably all future Leaf backs will be using IIQ files as well, so at least the set of backs you'll have to divert to the MOS loader shouldn't ever expand.  Let me know if y'all have time to address this soon-ish, if not I think this might be something I can write a decent patch for

[24-robinson_preserve.zip](https://github.com/darktable-org/rawspeed/files/1504961/24-robinson_preserve.zip)

bieber, 2017-11-27:
(I just made a really quick hack and changed the constant in IiqDecoder to "Leaf," modified MosDecoder to always return `false` from that function, and it did fix the issue, so it looks like that is the root cause)

LebedevRI, 2017-11-27:
Hi!

> I'm not sure if this technically counts as a regression because there aren't any Credo 40 RAW files in the test set

Correct, if there are no appropriate samples on RPU (well, at least one for that camera), then it is not a regression.

> I added one to raw.pixls.us a few hours ago

Thanks, validated!

The reasoning does sound plausible. I'll check...

closed 2017-11-27 by LebedevRI

bieber, 2017-11-27:
Thanks for the fix!

commit f0602c8a (2017-11-27): New Leaf raws (e.g. Credo) are IIQ, not MOS. Fixes #116

It seems that we can simply use the IIQ's magic to
differentiate between the two.

Thanks to @bieber for the report and the sample!