darktable-org/rawspeed #121: Canon CR3 support.

LebedevRI, 2018-03-02:
Though, until the production camera is actually released, and the full set of freely-licensed samples is contributed to the https://raw.pixls.us/, it won't be merged.

LebedevRI, 2018-03-02:
Does not appear to be `TIFF`.
Current best guess: `ISO Media`.

LebedevRI, 2018-03-04:
Hmm, ok, this is strange. Got the `ISO Media` parser working, and hooked-up dummy Cr3Decoder.
The `mdat` chunk is *NOT* lossless jpeg. But it just loads via libjpeg.
Those CR3 appear to be plain demosaiced JPEG's...
(Yes, this is an actual decoded image)
[image]

LibRaw, 2018-03-04:
According to ffprobe, mdat chunk contains four substreams.
First one is, exactly, JPEG preview, so from libjpeg's point of view this is JPEG file + ~30Mb of garbage on tail.

LebedevRI, 2018-03-04:
Aha, thank you, that makes more sense. Let's see...

LibRaw, 2018-03-04:
unfortunately, two data chunks (1624x1080 and 6288x4056, first one probably low-res preview) are not lossless jpegs too :(

LebedevRI, 2018-03-04:
(I'm still hoping they are JPEG w/arithmetic coding)

LibRaw, 2018-03-04:
It does not looks like JPEG (e.g. no FF doubling).

LebedevRI, 2018-03-04:
There are 0xFF0xFF sequences in files 05, 11, 19, (looking at the second half of the file in okteta hex editor)

clanmills, 2018-03-04:
Thank you folks.  I'm going to keep out of a discussion about the pixels and keep to my domain of metadata.  Thanks to your comments, I've now heard of ffprobe.

```
822 rmills@rmillsmbp:~/gnu/github $ exiv2 -pR foo.jpg 
STRUCTURE OF JPEG FILE: foo.jpg
 address | marker       |  length | data
       0 | 0xffd8 SOI  
       2 | 0xffdb DQT   |     132 
     136 | 0xffc0 SOF0  |      17 
     155 | 0xffc4 DHT   |     418 
     575 | 0xffda SOS  
823 rmills@rmillsmbp:~/gnu/github $ ffprobe foo.jpg 
ffprobe version 3.4.1 Copyright (c) 2007-2017 the FFmpeg developers
  built with Apple LLVM version 9.0.0 (clang-900.0.39.2)
  configuration: --prefix=/opt/local --enable-swscale --enable-avfilter --enable-avresample --enable-libmp3lame --enable-libvorbis --enable-libopus --enable-librsvg --enable-libtheora --enable-libopenjpeg --enable-libmodplug --enable-libvpx --enable-libsoxr --enable-libspeex --enable-libass --enable-libbluray --enable-lzma --enable-gnutls --enable-fontconfig --enable-libfreetype --enable-libfribidi --disable-jack --disable-libopencore-amrnb --disable-libopencore-amrwb --disable-indev=jack --enable-opencl --disable-outdev=xv --enable-audiotoolbox --enable-videotoolbox --enable-sdl2 --mandir=/opt/local/share/man --enable-shared --enable-pthreads --cc=/usr/bin/clang --arch=x86_64 --enable-x86asm --enable-libx265 --enable-gpl --enable-postproc --enable-libx264 --enable-libxvid
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Input #0, image2, from 'foo.jpg':
  Duration: 00:00:00.04, start: 0.000000, bitrate: 7605132 kb/s
    Stream #0:0: Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown), 6000x4000, 25 tbr, 25 tbn, 25 tbc
824 rmills@rmillsmbp:~/gnu/github $ 
```

I'll work on the metadata.  We've found the XMP data and I'm confident the Exif (and IPTC) data is in the Canon **_uuid box_** at the top of the file.

Later, I will investigate previews and ICC profiles.

LebedevRI, 2018-03-04:
> I'll work on the metadata. We've found the XMP data and I'm confident the Exif (and IPTC) data is in the Canon uuid box at the top of the file.
> Later, I will investigate previews and ICC profiles.

Thank you, @clanmills! Sounds awesome! :)

lclevy, 2018-03-11:
Phil Harvey exiftool 10.82 can parse Canon CR3

lclevy, 2018-03-11:
for http://img.photographyblog.com/reviews/canon_eos_m50/photos/canon_eos_m50_01.cr3
mdat contains "trak" picture at 0x6d940, 0x37b030, 0x546c70, and TIFF metadata at 0x2565b98

LebedevRI, 2018-03-11:
Yep, that much was understood. Need to continue working on that code.. Hopefully tomorrow?

lclevy, 2018-03-11:
I started a CR3 format description here if you are interested in contributing...
I've done in for CR2 here http://lclevy.free.fr/cr2/ and https://github.com/lclevy/libcraw2

clanmills, 2018-03-12:
Laurent

This is very helpful indeed.  I wrote an email to you to ask if you were considering working on CR3.  Then I made good progress and decided not to send the email.

I've documented my progress:  https://github.com/Exiv2/exiv2/issues/236

Team Exiv2 will have a meeting on May 5 at my home in England.  This project will be discussed.  There are many demands on the team and only a few contributors.  In honestly, I don't think we have the resources to implement support for CR3 in 2018.  https://github.com/Exiv2/exiv2/issues/225

lclevy, 2018-03-12:
Hi,
a lots of updates at https://github.com/lclevy/canon_cr3, you can subscribe to the project to see updates

clanmills, 2018-03-13:
Thanks for doing this.  It's looking good.

I'm puzzled by the claim (also made by @lclevy  and @LibRaw) that there are 4 images in MDAT.  I only see a single JPEG.  I see there are 4 **_trak_** objects which are a hierarchy of items (CRAW, stsz, co64).  Perhaps you could document the meaning of the objects in the **_trak_** hierarchy.
```
532 rmills@rmillsmbp:~/gnu/github/isobmff/pyke369/isobmffdump $ ./isobmffdump --dump stsz ~/Downloads/canon_eos_m50_01.cr3 
@0         | ftyp [24]
@24        | moov [28440]
@32        |   uuid [26216]
@26248     |   mvhd [108]
@26356     |   trak [484]
@26364     |     tkhd [92]
@26456     |     mdia [384]
@26464     |       mdhd [32]
@26496     |       hdlr [33]
@26529     |       minf [311]
@26537     |         vmhd [20]
@26557     |         dinf [36]
@26565     |           dref [28]
@26593     |         stbl [247]
@26601     |           stsd [128]
@26617     |             CRAW [112]
@26729     |           stts [24]
@26753     |           stsc [28]
@26781     |           stsz [20]
                       00000000  00 00 00 00 00 31 ea 2f 00 00 00 01                                                               .....1./....
@26801     |           free [15]
@26816     |           co64 [24]
...
```

lclevy, 2018-03-13:
There are 4 parts in mdat: 3 pictures (JPEG, plus 2 with new compression), then metadata like in tiff

clanmills, 2018-03-13:
@lclevy   I do not agree that there is a tiff in the MDAT.  Here's my evidence:  https://github.com/Exiv2/exiv2/issues/236#issuecomment-372625849

We agree that there is at least one JPEG in the MDAT.   I'm not convinced there are 3 pictures, however I have no proof to assert there is only 1 picture.

clanmills, 2018-03-13:
@lclevy  Apologies.  I am mistaken.  There is a valid tiff in the MDAT and ExifTool reads it.

@LebedevRI Thanks for pointing this out.  By coincidence I discovered and documented this before your email arrived.  Two heads thinking of the same thing!  That's better than no heads thinking!  https://github.com/Exiv2/exiv2/issues/236#issuecomment-372633645

Steady progress.  Great Teamwork. Thanks everybody.

lclevy, 2018-03-13:
updates: THMB and PRVW detailed. confirmed 3 jpeg and their sizes

clanmills, 2018-03-13:
Well done, @lclevy.  Very impressive progress.  It's possible I now have sufficient information to parse all the CR3 metadata and thumbnails in Exiv2.

lclevy, 2018-03-29:
please find parse_cr3.py ...

clanmills, 2018-04-03:
@lclevy Happy Easter.  We were on vacation last week and now I'm home and would like to look at parse_cr3.py.  Where have you put it?

clanmills, 2018-04-03:
Thanks.  That worked first time.  I'll study that.  Thanks very much.

simonszu, 2018-11-27:
Please allow me this one question: Is this still worked on? I have recently bought a M50 and was hoping for linux support since Lightroom already had it. But for now i purely rely on JPEG output, which is a bit sad.

LebedevRI, 2018-11-27:
Presently - no. It **will** _eventually_ be done.
I should probably take a stab at this.

clanmills, 2018-11-27:
I haven't been able to work on this either.  Since July 2017, Exiv2 has been swamped by security issues.  I'm glad to say that every issue has been fixed and we will release Exiv2 0.27 on 28 December 2018.  We intend to offer "dot" releases as necessary during 2019 and 2020 to address security issues.

It has been a major undertaking to deal with security and very little feature development has been possible.  CR3 support is on the "highlight" features for v0.28 and will be discussed at a team meeting in December.  https://github.com/Exiv2/exiv2/projects

The Team may, _or may not_, agree to undertake this work in 2019.

For sure, I will not do this work.  I will be 68 in January and have been contributing daily to Exiv2 for more than 10 years.  During 2019 and 2020, I will be the release engineer (to do the dot releases), and deal with user questions.  However I want to retire and no longer contribute to the C++ code.

I'm delighted to say that the future of Exiv2 is secure.  In 2017 I recruited two outstanding engineers and they have dealt with most of the security issues.  They have also rewritten the build and test infrastructure.

Several observations about open source:

1) Many people request changes
2) Few folks volunteer to help
3) User contributed patches usually result in lots of work concerning platforms and testing

Adobe have 10,000+ people including several thousand C++ Engineers.  I know, I worked for Adobe in San Jose, California for more than 10 years.  Team Exiv2 has been mostly a single volunteer.  Surprise, Lightroom Supports CR3 before Roman, Laurent and Exiv2 can catch up.

If you have any good ideas about how to recruit engineers to join Team Exiv2, I'll be delighted to receive suggestions.

simonszu, 2018-11-27:
@clanmills That's an excellent idea. I'll do it right now.

lclevy, 2018-11-27:
Adobe is certainly using code from Canon

⁣Envoyé par Blue ​

lclevy, 2018-11-27:
> I do have a practical suggestion to make about this.  Canon should be
> contributing to open-source.  When they decide to launch a new format
> on their users, I believe they have engineers in _"Developer
> Relations"_ who will engage with Adobe, Google, Apple, Microsoft and
> others to ensure that PhotoShop and other major applications provide
> support for their format.
>
> I have never received any communication from Canon or any other camera
> manufacturer.  The MakerNote Support in Exiv2 has been reversed
> engineered.  Laurent has done very good work to reverse-engineer CR3.
> Canon sold the camera and contributed nothing to the open-source
> community.
>
> So, @simonszu Would you write to Canon and express your disappointment
> that they have not supported the open-source Community to adopt CR3.

Canon does have a sdk for developpers.
But it is closed source and not public 

⁣Envoyé par Blue ​

simonszu, 2018-11-27:
Sehr geehrte Damen und Herren,

ich möchte gerne meine vor einem Monat neu erworbene Canon M50 unter Linux in Betrieb nehmen. Hier habe ich vor allem das Problem, dass das neue CR3 RAW Format nicht mit den gängigen RAW-Bearbeitungsprogrammen (bspw. RAWtherapee oder DarkTable) kompatibel ist, was sehr schade ist, da ich die M50 momentan ausschließlich über die JPEGs nutzen kann. 
Meines Wissens nach arbeitet Canon bereits mit Adobe und anderen Studios zusammen, um RAW-kompatibilität bei kommerzieller Software zu gewährleisten. Leider wird hier die Linux- und OpenSource-Gemeinschaft völlig außer Acht gelassen. 
Ich würde mich freuen, wenn Canon hier ebenfalls mit den Entwicklern zusammen arbeiten würde, sodass auch Fotografen, die lieber quelloffene Software, bzw. Linux verwenden, das volle Potenzial aus ihrer Kamera schöpfen können. 
Ein mögliches Projekt wäre hierbei z.B. exiv2 (http://www.exiv2.org/), da dies als Grundlage von vielen anderen RAW-bearbeitungstools verwendet wird. 

Bitte sorgen Sie dafür dass ich meine Kamera vollständig benutzen kann. 

Vielen Dank und freundliche Grüße,

<hr>

Dear Sir or Madam,

i'd like to use my Canon M50 which i have bought around a month ago with Linux. I currently have the problem that the new CR3 RAW format isn't compatible with the usual RAW editing tools (like RAWtherapee or DarkTable), which is sad, since i currently only can use the M50 via JPEG output. 
As far as i know, Canon is working with Adobe and other developers to ensure RAW compatibility in commercial software. Unfortunately, the Linux and OpenSource community is completely left behind. 
I would be very happy if Canon could also work with these developers, so that photographers who would rather use open source software or Linux could use the full potential of their cameras. 
A possible project would be exiv2, since that's the underlying library which is used by many other RAW editing tools. 

Please make sure that i can use the full potential of my camera. 

Thank you so much and best regards

<hr>

I have sent the german text to Canon Germany, since i'm from Germany as well. Maybe other users who stumble upon this issue can re-use it, so i have provided an english translation as well.

lclevy, 2018-11-27:
See references:
https://github.com/lclevy/canon_cr3/blob/master/readme.md

⁣Envoyé par Blue ​

LebedevRI, 2018-11-27:
> Ein mögliches Projekt wäre hierbei z.B. exiv2 (http://www.exiv2.org/), da dies als Grundlage von vielen anderen RAW-bearbeitungstools verwendet wird.
>...
> A possible project would be exiv2, since that's the underlying library which is used by many other RAW editing tools

exiv2 *only* has the knowledge to decode the metadata.
It does not, and should not, do any of the **actual** raw image (i.e. the actual pixels) decoding.
Making exiv2 aware of cr3 will not be required for me to make darktable understand cr3.

> Please make sure that i can use the full potential of my camera.

Uh huh.
Regardless of whether this is directed at canon, or any of us, i really don't think it works that way :)

simonszu, 2018-11-27:
It is directed at Canon. In the past i have experienced Canon as a very consumer friendly company which makes sure that their users are happy. 

We will see. I have got an automated response which assured me that i'll get an update by a service guy in 1-2 days. 

The usual problem with Linux support is in most cases that there is a quite small user base, so that it is too expensive to write official tools for it. But maybe once the company is aware that there is already a community which develops quite good applications themselves (and, most importantly, _for free_), and all they need is a little help, some companies are happy to jump in (just look at microsoft). 

If their marketing guys realize that they can "unlock" an even bigger potential user base almost for free...well, i'm just thinking loudly. I'll keep you guys updated once canon has sent me an answer.

clanmills, 2018-11-27:
I only mentioned Canon because we're discussing CR3.  I worked for Canon for 3 years.   I'm a printing software guy and that's what lead me to work on Adobe PostScript and later Acrobat.  Never worked on metadata at Adobe.

It's hard to say how Canon will react.  It's a business matter.  Historically, _Developer Relations_ would have focused on big commercial software vendors such as Adobe and Microsoft.  The contracts and contacts are well defined.  To work with the "open-source" community sounds vague and undefined.  How do they know to contact me, Roman and so on?  There are thousands of projects after all.  They could focus on 5-10 major applications (GIMP, darktable, digiKam, and others).  They should openly publish their format specifications and avoid NDAs.

Thanks for making the effort to bring this to their attention.  Very helpful.  It'll be interesting to hear their response.

hfiguiere, 2018-11-27:
Just a little story about Canon and Linux support:

Over decade ago some of the developers of gphoto2 contacted Canon to try to get some documentation to improve Canon support in gphoto. Canon answer was that they didn't support Linux and that there was that open source project that work on Linux. That project name was... gphoto.

True story.

clanmills, 2018-11-27:
There are very smart people at Canon.  Clearly this reply didn't come from one of their best people!

And as you said, it was 10 years ago.  The world is changing.  Open-source is a **_"tour de force"_**.

rrenomeron, 2018-11-29:
I know it's not ideal, but I was able to import some sample Canon M50 CR3 files into Darktable after converting them to DNG with an appropriate version of Adobe DNG Converter running under wine, and work with the DNGs.  Depending on how demanding your workflow requirements are, it might be a viable workaround until this issue is figured out.

Disclaimer: I don't actually own an M50, I tried this as part of my camera-buying research earlier this year.

LebedevRI, 2018-11-29:
> I also asked Canon's EMail support to release this information.

I really hope all these mails this won't cause an epic (legal) blow-back at me/us/project :)

In principle the patent linked at https://github.com/lclevy/canon_cr3#crx-compression has sufficient into.
At least that is how it looks like, without starting to implement it.

Like i said previously elsewhere(?), i think it will be almost the same
as the GoPro's VC5 decompressor, with the notable differences being:
* different wavelet, 5/3 instead of 2/6 (but the same number of wavelet levels, 3)
* not hardcoded RLV, but Golomb-Rice
* looks like all the wavelet levels are Golomb-Rice (as opposed to the smallest wavelet being uncompressed low-pass)

neo22s, 2019-01-03:
Hello,

Thanks for the job, and joining the conversation, any news on this?

Bought a canon eos R and I can no use the RAW files. I work only on ubuntu :(

Any temporary fix?

LebedevRI, 2019-01-03:
> Hello,
> 
> Thanks for the job, and joining the conversation, any news on this?
> 
> Bought a canon eos R

Aha, finally someone with that camera!
Please help, we need samples for https://raw.pixls.us/ :)

Details:
Pick some static scene (outdoor landscape, no people), use tripod
There are total of 2 * 2 * 4 samples needed (i think?), let me break it down:
* There are two raw modes on that camera (i think?): RAW, C-RAW.
* That camera has dual-pixel mode, which can be disabled and enabled
* That camera can produce raws in 4 different aspect ratios
We need the full coverage of all that, so
* RAW, dual pixel off, 3:2
* RAW, dual pixel off, 4:3
* RAW, dual pixel off, 16:9
* RAW, dual pixel off, 1:1
* RAW, dual pixel on, 3:2
* RAW, dual pixel on, 4:3
* RAW, dual pixel on, 16:9
* RAW, dual pixel on, 1:1
* C-RAW, dual pixel off, 3:2
* C-RAW, dual pixel off, 4:3
* C-RAW, dual pixel off, 16:9
* C-RAW, dual pixel off, 1:1
* C-RAW, dual pixel on, 3:2
* C-RAW, dual pixel on, 4:3
* C-RAW, dual pixel on, 16:9
* C-RAW, dual pixel on, 1:1

Please, do rename the files appropriately!!!
Afterwards, upload each one of these files separately(!!!) to https://raw.pixls.us/

> and I can no use the RAW files. I work only on ubuntu :(
> 
> Any temporary fix?

neo22s, 2019-01-03:
@LebedevRI tomorrow you have it ;)

lclevy, 2019-01-03:
You'll have also information here: https://github.com/lclevy/canon_cr3/blob/master/readme.md#samples

⁣Envoyé par Blue ​

LebedevRI, 2019-01-03:
> @LebedevRI tomorrow you have it ;)

the upload form will say that there are samples already, do upload everything anyway.

LebedevRI, 2019-01-03:
> You'll have also information here: https://github.com/lclevy/canon_cr3/blob/master/readme.md#samples ⁣Envoyé par Blue ​ Le 3 janv. 2019 à 21:53, à 21:53, Chema <notifications@github.com> a écrit:
> […](#)

Yes, i'm aware of those samples.
1. they are not on RPU, therefore i don't/can't/won't use them in the integration testing of the library.
2. They are not under CC0, therefore their usage is somewhat questionable
3. We don't have CC0 sample set anyway, so we need one anyway

neo22s, 2019-01-05:
Hello,

All RAW files here properly named:
https://mega.nz/#!NsQRgCxT!vCh4EaU8FfPsXCl2ge4Ar4XGMhbgtjPMu8qBi6Gp2q0

I am uploading 1 by 1 but It gets stuck and I need to leave in 10 minutes.

LebedevRI, 2019-01-05:
No worries, please do finish uploading them when/if you have time.
Sadly, this isn't really time-critical issue here..

neo22s, 2019-01-05:
Done! so what's next? :D 

I really need this.. I am trying to make a living from photography and doing it only using Linux gets hard... ;)

LebedevRI, 2019-01-05:
Sadly, the best current advice _i_ can get is to build a chroot (https://wiki.debian.org/Debootstrap e.g.), and *temporarily* use adobe dng converter to *temporarily* convert the raws to DNG, while keeping the originals.

neo22s, 2019-01-05:
> https://wiki.debian.org/Debootstrap e.g.), install wine there, and

Ouch. Terrible workflow.

When do you think we can have this done?

seanmavley, 2019-01-20:
@neo22s 
I have the M50, and I must say working with it has been a challenge. Only adobe and canon's dpp support it. And both apps work on Windows only. No linux support.

Currently, my workflow involves running canon dpp in Windows, export all the files, then continue with darktable or gimp in Linux.

@LebedevRI I have the m50, in case samples would be needed from it. Over 5k of CRAW files to pick from in almost all forms of lighting conditions.

neo22s, 2019-01-20:
@seanmavley 

Luckily for me I was barely using the RAW files, I do  fashion photography and only if I messed up badly I needed to use the RAW files (wrong white balance or really bad exposed).

If I was a wedding photographer (which I do sometimes) will be a nightmare. I am ready to use windows those days if this does not go faster.

Such a pity :(

LebedevRI, 2019-02-02:
I have not looked further into that yet.
If anyone wants to, feel free to work on it.

lclevy, 2019-02-05:
it seems Alexey is working on it
https://github.com/lclevy/canon_cr3/issues/7

LebedevRI, 2019-02-05:
Nice!
Good spec will be better than working code (looking at you, fuji decompressor).

neo22s, 2019-02-05:
Thanks guys 👌

Alexey-Danilchenko, 2019-02-05:
@LebedevRI - yet working code speaks for itself and miles better than empty statements.

LebedevRI, 2019-02-05:
A working code (_source_ code under open license, not something else) is sure better than nothing.

That fuji remark was mostly in context of 

> [lclevy/canon_cr3#7](https://github.com/lclevy/canon_cr3/issues/7)

which seemed like an improvement/contribution to the attempt to spec-ify the codec.
I omitted word "even" in "Good spec will be <even> better", thus, having an idea of the state
of fuji decompressor, it _does_ read as some variation on "better no code than that again".

I admit, i'm fond of the effort fuji decompressor took,
and i **will** freely admit that i'm not fully fond of the structure of that code.

In particular, i'm observing that the threading is per-input-tile, and there are 4..12 of them,
which results in load imbalance. so, i'we been wondering whether the algo can be
separated into stages, so _some_ of decoding can be more fine-grainedly parallelized (per-row),
but so far i was unable to fully comb through it. Although i haven't really looked into that too deep.

Oh well, hopefully this didn't degrade things even further :)

Thank you for looking into it!

Alexey-Danilchenko, 2019-02-05:
That format came from reversing just as in Fuji case header format for their data.  Neither will give you the specifications of original format or decoding algorithms. So whatever that is - it is definitely not and attempt to specify the codec, only the file format.

As with anything in reversing, unless you will want to put an effort in understanding the algorithms and rewrite the code, it will not improve from how it was coded by original developers (Fuji, Canon etc.) and from what I seen their coding is left to be desired. I certainly won't do it for you (I have enough spare time spent on it already) so perhaps it's task you will take upon yourself (it will certainly come out better than speeches). In the meantime, I will release CR3 (when ready) to libraw only.

[138 more replies]