darktable-org/rawspeed #122: Stop reinventing the wheel.

LebedevRI, 2018-03-05:
This has been circulating through my head for some time, so it might as well be the time to actually write it down.

Presently RawSpeed is mostly[1] up-to-date with the C++ standard.
It requires C++14-conformant compiler/standard library (clang-3.5+, gcc-5+)

It also has very few external dependencies:
* C++ STL.
* [pugixml](https://github.com/zeux/pugixml), for `data/cameras.xml` reading
* libjpeg (e.g. [libjpeg-turbo](https://github.com/libjpeg-turbo/libjpeg-turbo)), for Lossy JPEG compressed DNG tiles.
* [zlib](https://github.com/madler/zlib), for Deflate-compressed DNG tiles
* [googletest](https://github.com/google/googletest) and [googlebenchmark](https://github.com/google/benchmark) for obvious development reasons.

There are a few things that are the matter of fact as of now, and they should remain valid going forward:
1. **All** the current dependencies (except STL, as expected) are optional, in the sense that RS can be built without `(#)` them
2. and retain *practically* all functionality `(#)`.
3. All dependencies except two (C++ STL, and libjpeg (should be doable, too)) can be natively built in-tree `(#)`
by the current CMake build system,
4. and they are not too large, heavy `(#)`.

This is super important.

This is great.
However, there are still, and always will be, things that are not present in the used standard.
Some of them are domain-specific, and won't ever[2] be in the standard:
* Huffman table (multiple implementation variants: binary tree, vector with strict ordering, LUT+vector, ...)
* JPEG
* All the various manufacturer-specific decompressors 
* TIFF
* CIFF
* ISO Base Media File Format (CR3)
* ...

So those just have to[3] be implemented.

But, for sufficiently good, maintainable code, which i'm trying to write here, there are always some smaller patterns,
data structures that can be noticed, factored-out and re-used, thus adding abstraction and increasing readability.
I can summarize this as a table:

Name                                                                   | status in RS             | status in C++
-----------------------------------------------------------------------|--------------------------|------------------
`Endianness`                                                           | Already used             | C++20              
`Mutex` w/[TSA](https://clang.llvm.org/docs/ThreadSafetyAnalysis.html) | Already used             | Never
`Span`-like thingy                                                     | Already used (`Buffer`)  | Nope
Actual `Span`                                                          | Want to use              | Nope (GSL)    
`optional`                                                             | Want to use              | C++17      
`string_view`                                                          | Want to use              | C++17      
`smallvector` (w/small size optimization)                              | Want to use              | Never
`ArrayRef` (const reference to `array`/`vector`)                       | Would like to use        | Not that i know of?
??? I'm sure there are some more

All these things need to be
* Implemented sanely
* Play nicely with ASAN/UBSAN/MSAN/TSAN (!)
* Ideally, use sanitizer annotations (!!)
* Don't have overhead / have minimal overhead

Now, it is of course possible to either
* Don't bother, not use any abstractions, keep taking shit on the codebase.
* Use `std::experimental`/newer standards.
  But not really, not everything is even in the future C++ standard's drafts/proposals.
  And it will introduce additional restrictions on supported standard libraries versions,
  E.g. libcxx trunk just dropped  most of the `std::experimental`.
* Copy single needed headers from somewhere (`LLVM` ADT / `libcxx` trunk / ...) to `src/external/`
  * Snowball effect, headers are usually not self-contained
  * It's the dirtiest version bundling, they wont ever be updated, and will rot.
  * I'm not a lawyer, i don't know what licensing reprecussions that will have on the LGPLv2-licensed RS.
* Same as the previous step, but trim the headers to a minimum
* Reinvent the wheel, iteratively implement things from scratch, adding functionality when the need arises.
  Basically what is being done currently. Not too horrible. But this is the wheel reinvention.
  I'd much rather actually use the appropriate atomic (in the sense of them being the minimal tool to solve the task)
  data types to solve the actual raw decoding problems at hand.
* Or, bite the bullet, and pick some library which already implements these things, and depend on it.

I'd like to explore the last option.
To recap, see the things at the beginning marked with `(#)`.
It is obvious that points 1. and 2. will *NOT* be applicable here.
But the points 3. and 4. **MUST** remain true.

There are several possibilities:
* LLVM Adaptive Data Types.
  Not really a standalone library.
  So would require copying/bundling the selected headers. Not cool.
  But should be pretty small.
* I'm sure everyone is familiar with Boost :)
  * Rather old library, has a lot of mileage.
  * 23 MLOC as per https://www.openhub.net/p/boost.
    So this **undoubtedly** does **NOT** fit the 3. and 4.
* [folly](https://github.com/facebook/folly)
  Has a long list of deps, including boost. Skipping.
* Guidelines Support Library
  Super minimalisting,d oes not contain most of the things, only some.
* ???
* Google's [abseil](https://github.com/abseil/abseil-cpp).
  * Well, yeah, it's google :/
  * No releases. Copylib ('live at head', as it's now called). Fedora will be delighted!
    * So 3. *will* be strongly satisfied with this.
  * Rather small, at least right now:
    ```bash
    abseil-cpp$ find -iname '*.h' -or -iname '*.cc' | xargs wc -l | tail -n 1
    66605 total
    abseil-cpp$ find -iname '*.h' -or -iname '*.cc' -not -iname '*_test*' | xargs wc -l | tail -n 1
    40502 total
    ```
    * So at least right now, 4. *will* satisfied.
  * As a bonus, it does have sanitizers annotations!

Is there some other alternative i missed?

Please just don't even suggest RIIR :) Rust does have a *lot* of potential, and will hopefully murder C, but it is just not ready to fill the shoes it is trying to wear.

I'm in the corner here, and not sure where to push. At least this somewhat summarizes things...

1. mainly, pthreads vs. C++ native threading concepts (thread, lock)
2. see hilarious proposal to standardize cairo-based 2D graphic for C++. :(
3. Of course, there are always some libraries out there which already implement them, but as a general principle, which i feel quite strong about, only the very very basic or very very edge-case/complicated stuff should be used from libraries.
   AND it is highly ideal that all three things marked with `(#)` remain possible.

JonathanDCohen, 2018-03-05:
Hey there!  Just dropping in from the Abseil team.  Just some thoughts I had reading your post -- 

Abseil has everything specified in your table except for Endian support:

| Data Structure  | Abseil Type  | Abseil file  |
|---|---|---|
| Mutex  | absl::Mutex  | absl/synchronization/mutex.h  |
| Span  | absl::Span*  | absl/types/span.h  | 
|  optional | absl::optional**  | absl/types/optional.h  |
| string_view  | absl::string_view  | absl/strings/string_view.h  |
| smallvector  | absl::InlinedVector***  | absl/container/inlined_vector.h  |
| ArrayRef  | absl::Span<const T>  | absl/types/span.h  |

* Note that the interface of absl::Span is more like that of std::string_view than anything in the std::span proposal

** before C++17, absl::optional is our own implementation.  In C++17 and beyond, it becomes a simple typedef for std::optional

*** We also have FixedArray, a runtime-sized small-object-optimized array with fixed size

--

Currently our CMake support is a bit wonky, but I'm currently in the process of improving it as a high-priority task -- we should have better support for using us via ExternalProject, for example.

In terms of external dependencies, we do currently rely on cctz for Time support, but that is also being worked on as a high priority task.

My question for you, then, as we work to turn Abseil into a more mature open source project, is what are the pain points currently which would keep you from depending on Abseil for your low level abstractions?

LebedevRI, 2018-03-05:
Thanks for dropping by!

> Abseil has everything specified in your table

Nice :)

> except for Endian support:

That is mostly non-problem, that one is [trivial](https://github.com/darktable-org/rawspeed/blob/fb157cc89ff5097c41f5eb1f5eb8b1b9164299bc/src/librawspeed/io/Endianness.h#L30-L56).

>  ArrayRef | absl::Span<const T> | absl/types/span.h

Oh, nice, i did not thought of that!

> Currently our CMake support is a bit wonky, but I'm currently in the process of improving it as a high-priority task -- we should have better support for using us via ExternalProject, for example.

**Ideally** i'd expect it to be similar to [that of googletest](https://github.com/google/googletest/tree/master/googletest#incorporating-into-an-existing-cmake-project), with the same **flexible** set of variants.

> what are the pain points currently which would keep you from depending on Abseil for your low level abstractions?

That's **The** question here :)

The versioning scheme (live at head / copylib / just use the trunk) is likely the biggest sorrow point.
Don't get me wrong, i totally get the reasoning, after all, the current RawSpeed *is* using that very same model... It kinda works so far (except with Fedora, see e.g. [[1]](https://pagure.io/packaging-committee/issue/550), google `fedora + darktable + rawspeed` can show you more gore), because there are rather good reasons to do it this way.

But if a copylib starts using a copylib, i'm not so sure how things will work out...
*If* abseil is to have a proper 'release' tags (like it already promises to, [see](https://github.com/abseil/abseil-cpp/issues/87#issuecomment-370438295)),
then there would be some room to breathe.

At minimum, one could simply manually track said 'release tag' via submodule.
At maximum, distros could package new abseil release tarball (*source* tarball, like debian does for googletest) every half a year.
Though i doubt they will like it. But i'm also not sure they will like all those bundled copies :)

TLDR: can't check the water temperature without somehow measuring the water temperature.

Did that answer the question, or was the question different?

JonathanDCohen, 2018-03-05:
I'm not sure I understand why building Abseil from trunk impedes your own ability to build from trunk unless we accidentally break you -- I assume that is the issue?  Anyways, as was pointed out in our repo, we are very soon having a serious discussion about stable release branches / tags.  So stay tuned there. This also makes me start to think about some way to specify "try to build / test with trunk abseil.  If it passes, replace my copy of abseil with HEAD, else keep my current version"  

As for CMake, it seems like the only real issue there is that googletest has example code for you to download and build them at configure time, and then asks you to `add_subdirectory` just as we currently do. We can definitely look into supporting that -- if you are willing to use bleeding-edge cmake 3.11 then you may want to look into FetchContent(), which does the get-my-dependency-at-configure time for you but allows you to then build and link it at the same time as the rest of your build.

LebedevRI, 2018-03-05:
> I'm not sure I understand why building Abseil from trunk impedes your own ability to build from trunk unless we accidentally break you -- I assume that is the issue?

Not really. I'm not sure how to explain it to the people who don't already understand it, sorry :/
Not even looking at the distro side of things, this 'just use current git master' is a serious change from the usual approach. Plus, i don't _actually_ know the opinion of the actual users of *this* library on that matter.
(CC @hanatos / @houz, @LibRaw, @aferrero2707, ..., possibly @Beep6581 in the future, ....; all the interested parties are likely watching the repo anyway)

> Anyways, as was pointed out in our repo, we are very soon having a serious discussion about stable release branches / tags. So stay tuned there. 

Good. Let's see what happens there :)

> FetchContent(), which does the get-my-dependency-at-configure time for you but allows you to then build and link it at the same time as the rest of your build.

I have this figured out with [GoogleTest.cmake](https://github.com/darktable-org/rawspeed/blob/0896f1b253d3b2cc7f7a060da243fbe69104e936/cmake/Modules/GoogleTest.cmak)e + [GoogleTest.cmake.in](https://github.com/darktable-org/rawspeed/blob/0896f1b253d3b2cc7f7a060da243fbe69104e936/cmake/Modules/GoogleTest.cmake.in)
*If* abseil supports that approach (if it does not, then this is all moot point and i definitively can't pick abseil...), it will just work with `s/GoogleTest/abseil/i` so to speak.

But let's take the step back, shall we. I did not pick abseil. I did not _not_ pick abseil.
It's still an open question as of right now.
(Though, yeah, unless something rather drastic happens, i guess abseil will be the answer here..)

JonathanDCohen, 2018-03-05:
> I have this figured out with GoogleTest.cmakee + GoogleTest.cmake.in

The directions at https://github.com/google/googletest/tree/master/googletest#incorporating-into-an-existing-cmake-project should work just fine for Abseil already.  If you decide to go this route and have issues please let me know by filing a bug against abseil.

>But let's take the step back, shall we. I did not pick abseil. I did not not pick abseil.
It's still an open question as of right now.

Oh of course.  But picking up this issue seemed like a great way for us to reach out and get more feelers about how different people are feeling about Abseil :)  I hope this didn't feel like pressuring you to use us, obviously you should do whatever works best for you.

houz, 2018-03-06:
As a user of rawspeed I'd like to point out that I wouldn't be happy to see the build system download stuff from the Internet. At all. For google test that's a non-issue as users don't have to build the tests. But a library that is a hard dependency has to be either shipped by the OS or bundled in the source tree.

Apart from that I am happy with anything that isn't Boost.

darix, 2018-03-06:
my 2 cents ... as a packager.

No releases just sucks. For many distros it means we will have at least 2 copies of abseil on the distro (rawtherapee and darktable). And I really don't get what would be so hard with tagging releases and allow proper packaging that way. If you expect your users to be able to use HEAD all the time then all your QA should be done already and it is all about making a tag.

In the long term I hope we can also get rawspeed releases at some point to avoid the duplication between darktable and rawtherapee. But that is a discussion for another day.

JonathanDCohen, 2018-03-06:
@houz The current abseil CMake directions suggest using a local fork and add_subdirectory() already.  A way to do this while staying current to Abseil HEAD is to set up a local mirror of abseil which is pulled from during the build process, for example.

@darix the 5 second version is that Abseil is so low-level that if we have tagged releases, we are almost guaranteeing ODR violations -- suppose your project uses Abseil v1, but you also use protobuf which uses Abseil v2 and gRPC which uses Abseil v3.  But we are likely to have something like a 6 month stable release timeline which is a simple tag on our Github repo.  We also want to avoid binary distribution for the same reason.  See [our TL Titus Winters' cppCon keynote for more detail](https://www.youtube.com/watch?v=tISy7EJQPzI)

LebedevRI, 2018-03-06:
> using a local fork and add_subdirectory() already. A way to do this while staying current to Abseil HEAD is to set up a local mirror of abseil which is pulled from during the build process, for example.

The 'how' part is quite clear to us. We won't invent a new wheel there (LOL), but just add an 'abseil' submodule into https://github.com/darktable-org/rawspeed/tree/develop/src/external
Since right now darktable is using rawspeed as a submodule already, this won't add any new problems. It's a charted territory.

I do understand the gravity of the ODR issue, and the reluctance to do releases.
I absolutely do *NOT* want binary releases. Only a tag/source releases.

I'm hazy on the part about abseil being used by the library, and then by the software using that library.
I do understand that it will cause problems. Right now i'm not aware of any RS users already [[in]directly] using abseil, so i'd prefer to leave this out until that actually happens.
But when it does happen, there are some easy solutions i see:
1. An cmake switch telling rawspeed not to build it's own version of abseil, but just use the one provided (via cmake target) by the parent build system.
2. An option to specify the path to the abseil sources to be used
3. ???

More than one option can exist at the same time. The second option will be available anyway from the start, i think.

> staying current to Abseil HEAD

Let me be blunt: i don't see this happening.
Some specific commit/tag will be tracked, and updated manually. Ideally, that 6-month tag.
**BUT** the packagers will be free to replace the copy with some other version to address the ODR concerns/etc. That's the Key.

JonathanDCohen, 2018-03-06:
@LebedevRI @darix This is a great conversation from the perspective of packagers -- would you all mind moving this over to the abseil github so the rest of the team can be in on the discussion?

LebedevRI, 2018-03-06:
Absolutely do feel free to @LebedevRI highlight me in the relevant discussions! (packaging, and the cmake stuff to)

LebedevRI, 2018-03-08:
@JonathanDCohen where do you want me do output all these thoughts?
(i guess, it would be best to do that before https://github.com/abseil/abseil-cpp/issues/87#issuecomment-370438295 is done..)

ilyapopov, 2018-06-20:
By the way, Abseil team released an LTS branch: https://abseil.io/about/releases

LebedevRI, 2018-06-20:
Still no release though https://github.com/abseil/abseil-cpp/releases

LebedevRI, 2018-08-07:
https://github.com/abseil/abseil-cpp/releases/tag/20180600

darix, 2018-08-07:
https://github.com/abseil/abseil-cpp/issues/155

LebedevRI, 2021-11-26:
A lot of water has flown since.
The world isn't what it used to me.
I _think_ this isn't going to be happening.

closed 2021-11-26 by LebedevRI