darktable-org/rawspeed #125: FUJIFILM DBP for GX680 RAF support

wtanmdt, 2018-05-10:
The developer of LibRaw (Alex Tutubalin) is so nice that he tailored a program for this special raw file. I have uploaded it to:
http://www.weldpass.com/upload/LibRaw-snapshot-20180509-Win64.zip

The zip file includes the source code, which is internal\dcraw_common.cpp, function void unpacked_load_raw_FujiDBP( ) Metadata read/parse are in parse_fuji( ) and in identify( )

The raw converted resolution is 5430 x 3845.
If I shoot JPG format, the resolution is up to 7648 x 5408.
And the color profile needs improvement since the color target shot was not utilized in the program.

These are shots on the color target of X-Rite ColorChecker Passport:
http://www.weldpass.com/upload/Fujifilm-DBP-for-GX680-daylight.RAF
http://www.weldpass.com/upload/Fujifilm-DBP-for-GX680-tungsten.RAF

RawDigger is not going to support this sensor. I wonder if rawspeed will be able to support it so that I can process it with Darktable.

Thanks.

LibRaw, 2018-05-10:
Minor correction: RawDigger (beta version you already have) indeed supports this format.  The only thing is not supported  is RGB render  (this is not supported for all cameras where sensor resolution does not match RGB render resolution)

LebedevRI, 2018-05-10:
That looks like uncompressed raw with some tiling.
But it does not look like just tiling, since all the 'tiles' on the same line appear to have the same content, so only the rows are different.
I'm guessing there is some kind of sub-samplin going on.

LebedevRI, 2018-05-10:
I'm having trouble finding pdf usermanual for that back.

> The only thing is not supported is RGB render (this is not supported for all cameras where sensor resolution does not match RGB render resolution)

So there is more than one raw format?

LibRaw, 2018-05-10:
RawDigger does not shows RGP rendering for 'non-square pixels' (pixel 'aspect ratio' is not 1.0), e.g. Nikon D1X, most Fuju SuperCCD cameras, some very old Kodaks. Only RAW display (composite and per-channel) is possible for such cameras.

The only cause for that is interface problem: RAW/RGB rendering are switchable via single button, there is no way to do that if RGB/RAW render size or orientation not match each other.

wtanmdt, 2018-05-10:
Sorry that I delete the snapshot since I do not have the permission from Alex to publish it before testing.

LebedevRI, 2018-05-10:
> Sorry that I delete the snapshot since I do not have the permission from Alex to publish it before testing.

I did not download it anyway.

LibRaw, 2018-05-10:
Here is decoder:
`  nTiles = 8;
  tile_width = raw_width / nTiles;

  ushort *tile;
  tile = (ushort *)calloc(raw_height, tile_width * 2);

  for (tile_n = 0; tile_n < nTiles; tile_n++)
  {
    read_shorts(tile, tile_width * raw_height);
    for (scan_line = 0; scan_line < raw_height; scan_line++)
    {
      memcpy (&raw_image[scan_line * raw_width + tile_n * tile_width], &tile[scan_line * tile_width], tile_width * 2);
    }
  }
  free (tile);
`

LebedevRI, 2018-05-10:
@wtanmdt before i proceed with this issue, do you own that back?
I need an answer to https://github.com/darktable-org/rawspeed/issues/125#issuecomment-388118669

wtanmdt, 2018-05-10:
@LebedevRI 
Yes, I own that back. 
Your help is really very appreciated.

LebedevRI, 2018-05-10:
Note that was only the first part of the question.
Is there more than one raw format selectable on that back?

wtanmdt, 2018-05-10:
@LebedevRI 
There is only one raw format available on that back.

LebedevRI, 2018-05-10:
OK, thank you.

wtanmdt, 2018-05-10:
@LebedevRI 
Thank you for your time and efforts.

LebedevRI, 2018-05-11:
@wtanmdt if you could contribute some better (than we currently have) sample to https://raw.pixls.us, that would be great, i don't really like the current one https://raw.pixls.us/getfile.php?type=preview&id=2069
Something like outside nature landscape would be best.

wtanmdt, 2018-05-11:
@LebedevRI 
I will. Please allow me 2 days as it is overcast today and tomorrow.

wtanmdt, 2018-05-13:
Is this photo better? @LebedevRI 

[image]

LebedevRI, 2018-05-13:
Looks underexposed by one stop, but yes, better.
(RPU seems down at the moment :/)

LebedevRI, 2018-05-13:
Thanks, deduplicated down to https://raw.pixls.us/getfile.php?type=preview&id=2574

LebedevRI, 2018-05-13:
After looking at the exif (e.g. [diff](https://raw.pixls.us/diff.php/2574/1769)), i'm not seeing any obvious tag that would identify this raw variant from the others :/
Maybe other than `RawImageFullSize` vs `RawImageSize` mismatch..

wtanmdt, 2018-05-15:
This is beyond my knowledge. Many thanks for your time. @LebedevRI

wheropl, 2019-10-22:
@wtanmdt could i get a pic of that back? i'm really interested in it

wtanmdt, 2019-10-23:
@wheropl here you are:
[image]