darktable-org/rawspeed #145: Does anyone actually depend on updating cameras.xml only?

LebedevRI, 2018-09-18:
https://github.com/darktable-org/rawspeed/tree/615af31a22f0212bb7b646f246a253a513ba9ff6/data
```
This enables us to add support for cameras which were not yet released when the code was written.
```
To me this reads like there is implied support for the **any** version of the `cameras.xml`
to work with **any** version of the library.
I somewhat suspect this is already silently broken here and there.

Is anyone actually actively using this, and depends on this?

CC @LibRaw, @aferrero2707, @rawstudio (@abrander, @anderskvist), @axxel 
Not sure who else should be cc'd here.

abrander, 2018-09-19:
I have only a vague memory of how we use RawSpeed in Rawstudio-land, but I don't remember we ever updated `cameras.xml` without updating RawSpeed.

LibRaw, 2018-09-19:
Current LibRaw+RawSpeed (very old one) combo uses compiled-in cameras.xml and no interface provided for user-specified cameras.xml.

axxel, 2018-09-19:
I considered the xml and the library to be a unity, too. In fact I would very much prefer if the information inside the xml would be compiled into the binary. The whole burden it puts on the client side build and distribution system of putting this file at the right place and dealing with search path issues on all platforms is just a nuisance and brings no value (to me).

I'd also like to mention our (involved and for the most part fruitful) discussion in #26. If they are considered a unity, your arguments against any changes to the xml layout would just vanish.

LebedevRI, 2018-09-21:
Thanks everyone for the feedback!
It is utmost relieving that no one *actually* used that.