darktable-org/rawspeed #149: Several Nikon RAW files fail to open due to error in HuffmanTable

ilyapopov, 2018-10-20:
I have three raw files from today's shoot that fail to open in darktable 2.4.4 with the following errors:

```
[rawspeed] (DSC_6132.NEF) void rawspeed::HuffmanTable::setCodeValues(const rawspeed::Buffer&), line 176: Corrupt Huffman. Code value 92 is bigger than 16
[temperature] `NIKON CORPORATION NIKON D7100' color matrix not found for image
[temperature] failed to read camera white balance information from `DSC_6132.NEF'!
[rawspeed] (DSC_6133.NEF) void rawspeed::HuffmanTable::setCodeValues(const rawspeed::Buffer&), line 176: Corrupt Huffman. Code value 92 is bigger than 16
[rawspeed] (DSC_6134.NEF) void rawspeed::HuffmanTable::setCodeValues(const rawspeed::Buffer&), line 176: Corrupt Huffman. Code value 92 is bigger than 16
[temperature] `NIKON CORPORATION NIKON D7100' color matrix not found for image
[temperature] failed to read camera white balance information from `DSC_6134.NEF'!
```

Raw file is available at https://yadi.sk/i/4KCDBWBj0t_84A

ilyapopov, 2018-10-20:
Probably duplicate of #100

LebedevRI, 2018-10-29:
Of #129.
(Almost 2 years from introduction of an regression is indeed not ok.)

closed 2018-10-29 by LebedevRI