darktable-org/rawspeed #159: CFA pattern in CR2 files

photopea, 2019-01-01:
Hi, I am making my own parser and renderer for CR2, and I ran into a strange problem.

I notice, that sometimes, CR2 files use the Bayer pattern [0,1,1,2], but sometimes, they use [1,2,0,1] . How does your software find out, which pattern should be used?

I looked into this: https://sno.phy.queensu.ca/~phil/exiftool/TagNames/Canon.html , but can not find anything about CFA patterns :(

LebedevRI, 2019-01-01:
> I notice, that sometimes, CR2 files use the Bayer pattern [0,1,1,2], but sometimes, they use [1,2,0,1]

Yep.

> How does your software find out, which pattern should be used?

It doesn't right now, everything is hardcoded in https://github.com/darktable-org/rawspeed/tree/develop/data
That is a bug, not a feature though.

> I looked into this: https://sno.phy.queensu.ca/~phil/exiftool/TagNames/Canon.html , but can not find anything about CFA patterns :(

I think,
```
raw.pixls.us-unique/Canon$ EXIFTOOL */*  | grep -i cfa
[EXIF]          0xc5e0 CR2CFAPattern                   : [Red,Green][Green,Blue]
[EXIF]          0xc5e0 CR2CFAPattern                   : [Green,Blue][Red,Green]
[EXIF]          0xc5e0 CR2CFAPattern                   : [Green,Blue][Red,Green]
[EXIF]          0xc5e0 CR2CFAPattern                   : [Red,Green][Green,Blue]
[EXIF]          0xc5e0 CR2CFAPattern                   : [Red,Green][Green,Blue]
[EXIF]          0xc5e0 CR2CFAPattern                   : [Red,Green][Green,Blue]
[EXIF]          0xc5e0 CR2CFAPattern                   : [Red,Green][Green,Blue]
```

lclevy, 2019-01-01:
https://github.com/lclevy/libcraw2
See craw2.c

LebedevRI, 2019-01-01:
> https://github.com/lclevy/libcraw2
> See craw2.c

No license stated (no LICENSE file, no per-file license headers).
(and, GPLv2 is an //interesting//, rather crippling, choice for a library.)

photopea, 2019-01-01:
@LebedevRI Yes, you are right, it is the tag 0xc5e0 in the fourth IFD, thanks!

The value 1 is [0,1,1,2] , the value 3 is [1,2,0,1]. I have not seen a CR2 file with other values.

closed 2019-01-01 by photopea

LebedevRI, 2019-01-02:
> @LebedevRI Yes, you are right, it is the tag 0xc5e0 in the fourth IFD (not EXIF), thanks!
> 
> The value 1 is [0,1,1,2] , the value 3 is [1,2,0,1]. I have not seen a CR2 file with other values.

Roughly, the actual CFA is always the same: RGGB (1223)
```
-----
RG|RG
GB|GB
-----
RG RG
GB GB
```

There can be 3 extra variants: one line cropped off top (GBRG, 1201)
```
RG RG
-----
GB|GB
RG|RG
-----
GB GB
```

one line cropped from left (GRBG, 1021)
```
-----
R|GR|G
G|BG|B
-----
R GR G
G BG B
```
one line cropped from top and left (BGGR, 1220)
```
RGRG
-----
G|BG|B
R|GR|G
-----
G BG B
```

photopea, 2019-01-02:
@LebedevRI Yes, I know, that there can be four "permutations" (with greens on different rows), I just don't know if they occur in CR2 and what values of 0xc5e0 they have (probably 0 and 2, but not sure which is which).