darktable-org/rawspeed #160: CR2 files, that are not TIFF

photopea, 2019-01-02:
Hi, I am making my own parser and renderer of CR2 files.

I came across [a CR2 file from Canon - PowerShot A480](https://raw.pixls.us/getfile.php/2813/nice/Canon%20-%20PowerShot%20A480.CR2), which does not have a TIFF structure (from the first sight - it starts with bytes 32 8 8 130).

However, Darktable can open this file. Do you know where can I get more info about this format?

LebedevRI, 2019-01-02:
Since you are writing your own, i'd think you'd want to be able to solve such things on your own.
There is a working code. Have you tried applying debugger?

photopea, 2019-01-02:
I don't do it for fun, it would be great to use an existing solution, but there are no CR2 parsers in Javascript :( I just thought you are aware of some human-readable description of this kind of CR2 format.

I haven't worked with C or any compiled language in years (and I am scared by them :D as there are usually hundreds of megabytes of data I don't understand), so I just read the source code here on GitHub myself.

The first bytes of this CR2 are not mentioned in [Darktable Magic Bytes](https://github.com/darktable-org/darktable/blob/43b4442d1b9084e7eb0a7b77a8d44407773c845b/src/common/imageio.c). Also, there is no occurance of "Canon" ASCII string in the file, so I guess it is encoded somehow.

LebedevRI, 2019-01-02:
> I don't do it for fun, it would be great to use an existing solution, but there are no CR2 parsers in Javascript :(

That sounds like a fun job then :)
Have you considered/evaluated using WebAssembly, to simply compile/use some existing library?

> I just thought you are aware of some human-readable description of this kind of CR2 format.

*Public* - not really.
https://www.sno.phy.queensu.ca/~phil/exiftool/canon_raw.html
and https://github.com/darktable-org/rawspeed/find/develop "crw"
might be of help.

But given that it is not "for fun", (i'm _guessing_ not under free license, either),
not sure i can be of much help, and that these issues aren't misplaced,
though i can't suggest a better place either..

> I haven't worked with C or any compiled language in years (and I am scared by them :D as there are usually hundreds of megabytes of data I don't understand), so I just read the source code here on [...]

Ctrl+F for "Canon" (without quotes) [does](https://github.com/darktable-org/darktable/blob/43b4442d1b9084e7eb0a7b77a8d44407773c845b/src/common/imageio.c#L399-L406) find it, i think?

lclevy, 2019-01-02:
https://github.com/lclevy/libcraw2/blob/master/docs/cr2_poster.pdf

⁣Envoyé par Blue ​

photopea, 2019-01-02:
I published my parsers as open-source under MIT licence, e.g. [UTIF.js](https://github.com/photopea/UTIF.js) (there are many other parsers in my repository).

@LebedevRI I am aware of the CRW / CIFF format, but I think it is not the case, as CRW has to start with "II" characters, while my CR2 file starts with 0x20, 0x8, 0x8, 0x82. The "Canon" does not occur in my CR2 file, while it occurs in every TIFF CR2 / CIFF CRW file.

LebedevRI, 2019-01-02:
> I published my parsers as open-source under MIT licence, e.g. [UTIF.js](https://github.com/photopea/UTIF.js) (there are many other parsers in my repository).

Fun, whatever works for you i guess :)

> @LebedevRI I am aware of the CRW / CIFF format, but I think it is not the case, as CRW has to start with "II" characters, while my CR2 file starts with 0x20, 0x8, 0x8, 0x82. The "Canon" does not occur in my CR2 file, while it occurs in every TIFF CR2 / CIFF CRW file.

Yep, i see my previous comment.

photopea, 2019-01-02:
@LebedevRI Oh I see, so you "guess" everything just from the file size (if it corresponds to the resolution of some camera, multiplied by some bit depth). 

One more question: I see, that you don't have the color transformation matrices in your XML camera database. Do you always extract them from RAW files? I found such database with matrices in "dcraw", but many cameras are missing there.

LebedevRI, 2019-01-02:
> @LebedevRI Oh I see, so you "guess" everything just from the file size (if it corresponds to the resolution of a camera, multiplied by a bit depth).

Yep.
https://github.com/darktable-org/rawspeed/blob/92e0c3cd7be44ec441844967825e512d51a28cfa/src/librawspeed/decoders/NakedDecoder.cpp
It's super fragile, since it *only* matches based on filesize.
As soon as you have two different raws (two different cameras, different sensors, diferent CFA pattern)
with the same filesize, you have a problem since you won't know which one it is. (Such collisions has already happened with some two cameras)
 
> One more question: I see, that you don't have the color transformation matrices in your XML camera database.

That is correct.

> Do you always extract them from RAW files? I found such database with matrices in "dcraw", but many cameras are missing there.

*This* library does not do anything regarding color matrixes.
These matrixes aren't stored in most of the normal raw files, they are only required for DNG,
so each program that uses this library has it's own set of color matrixes (in case of darktable, it's https://github.com/darktable-org/darktable/blob/c9a4741debb7441503522de1c43cdc8110888a80/src/external/adobe_coeff.c)
(Maybe those should be moved into `cameras.xml`, maybe not, not quite sure.)

So yeah, all this raw 'business' is a little bit messy, with constant wheel reinvention by each program ^^

lclevy, 2019-01-02:
Color Matrix for a given camera can be extracted from a dng which original image was the cr2 from that camera.
This is what I made on this script.
https://github.com/lclevy/libcraw2/blob/master/dng_info.sh
Are you using CHDK to have raw output on A480?

⁣Envoyé par Blue ​

photopea, 2019-01-02:
@LebedevRI BTW. what is the actual pixel data format for these CHDK files? My file seems to have 12 bits per pixel. It starts with bytes (32,8,8, 130,130,32), which should be four pixel values (red,green,red,green). But if I read them "a regular way", I get (512, 2056, 2088, 544), which are pretty strange values.

@lclevy For DGN files, I read the matrix out of the DNG file. I only need it for non-DNG files. I still don't quite understand what CHDK (or using it) catually means. There is no Wikipedia article. Is it this software? http://chdk.wikia.com/wiki/CHDK

BTW. I never rook a RAW photo in my life, and I don't know the terminology :(

LebedevRI, 2019-01-03:
> I came across [a CR2 file from Canon - PowerShot A480](https://raw.pixls.us/getfile.php/2813/nice/Canon%20-%20PowerShot%20A480.CR2), which does not have a TIFF structure (from the first sight - it [...]

Hmm, fun, and that is also a collision, `PowerShot SX110 IS` entry is being used.
Though here it looks like a good match.

> @LebedevRI BTW. what is the actual pixel data format for these CHDK files? My file (at the top) seems to have 12 bits per pixel. It starts with bytes (32,8,8, 130,130,32), which should be four pixel values (red,green,red,green). But if I read them "a regular way", I get (512, 2056, 2088, 544), which are pretty strange values.

It does appear to be 12-bit uncompressed packed raw, with `BitOrder_MSB16`.
 
> @lclevy For DNG files, I read the matrix out of the DNG file. I only need it for non-DNG files. I still don't quite understand what CHDK (or using it) catually means. There is no Wikipedia article.

> Is it this software? http://chdk.wikia.com/wiki/CHDK

Yes.

> BTW. I never took a RAW photo in my life, or owned a DSLR camera, and I don't know the terminology :(