darktable-org/rawspeed #170: Probably high resolution RAW decoding issue with Panasonic DC-G9 (Firmware 1.2)

hoedur, 2019-02-18:
When opening a High Resolution RAW (the large variant with ~125 MB file size) with darktable 2.6.0, artifacts can be seen on the right border of the picture (panorama) or on the top border (portrait).

This issue was approved by @LebedevRI. Mre details are to follow if needed.

hoedur, 2019-02-18:
Workaround: use the crop editing feature of the "raw black/white point" module:

1. Edit "~/.config/darktable/darktablerc"
2. Find the line "plugins/darkroom/rawprepare/allow_editing_crop=false"
3. Change "false" to "true"
4. Open darktable and go to the "RAW black/white point" module
5. There should be a slider "crop width"; increase this value until the artifacts are gone
6. If you don't see this slider, then you might first delete "~/.config/darktable/" (create a backup first, as this fries your settings!)

kmilos, 2023-04-11:
Yep, both the single 20MP shot out of the hi-res sequence, _and_ the hi-res output as well seem to need extra 16px cropped.

The single shot has width of 5280 instead of the 5264 found in the normal mode.

More bizarrely, there is also this burst mode (sample also in RPU), where the width stayed the same as normal, but the height went up by 8 pixels (look like both top and bottom)!

jsmucr, 2023-04-11:
Would it help if I provided raws in various different modes?

kmilos, 2023-04-11:
> Would it help if I provided raws in various different modes?

They're all on RPU already AFAICT.

kofa73, 2023-04-11:
Is this related to High-Res mode? My LX7, which has no such feature, is affected by a similar issue. @kmilos is aware of the bug report I've opened for darktable, but maybe it's worth linking here:
https://github.com/darktable-org/darktable/issues/14082

jdneumeyer77, 2023-04-21:
This appears to be affecting the Panasonic S5 high-res raw files as well, but not the normal raw files. It looks like a regression because it was working fine. I haven't kept up since my PR was merged for v7 Panasonic raw format a couple months ago. Now I'm getting pink garbage on the right side of hi-res as well. I haven't had time to investigate more, but it looks similar to https://github.com/darktable-org/darktable/issues/14082.

jsmucr, 2023-06-27:
I think this approach with custom hardcoded crops is wrong in general. It's not just the hi-res mode. Bursts have different crop too. Also, the [0, 0, -58, 0] default crop is wrong because there are still some tiny artifacts visible.

It will never work well the way it is.

jsmucr, 2023-06-28:
@kofa73 @jdneumeyer77 Yes, it's all the same thing. I'm working on a fix for static cases such as hi-res or burst modes, but that won't help with all the issues.

jsmucr, 2023-06-28:
I've analyzed a collection of my G9 photos from May, and there's apparently no point in trying to fix it by adding more modes.
https://discuss.pixls.us/t/panasonic-g9-rawspeed-and-cropping/38236/5?u=newerth

The entire approach will need to change.

commit 302a284b (2023-08-11): Crop Panasonic G9 images using vendor metadata

This is supposed to fix #170 and help prepare other related fixes.

Now the decoders can decide whether to supply their own crop data when the Crop
tag in a camera.xml entry is empty or absent. Rw2Decoder has been updated to
leverage this new ability, and the Crop tags for Panasonic G9 entries have been
removed.
PR #490 (2023-08-11): Crop Panasonic G9 images using vendor metadata
This is supposed to fix #170 and help prepare other related fixes.

Now the decoders can decide whether to supply their own crop data when the `Crop` tag in a `camera.xml` entry is empty or absent. `Rw2Decoder` has been updated to leverage this new ability, and the `Crop` tags for Panasonic G9 entries have been removed.
closed 2023-08-11 by LebedevRI