darktable-org/rawspeed #178: Panasonic ZS200 files don't open in darktable

junkyardsparkle, 2019-06-03:
Using current git master, Panasonic ZS200 files fail to open. I'm not sure I understand exactly how the `cameras.xml` file is supposed to work, but it *looks* like the ZS200 and TZ202 are conflated within a single definition, with only TZ200 listed as an alias. Maybe one of the other two should also be an alias, rather than mixed together?

LebedevRI, 2019-06-03:
Please start with contributing full sample set to RPU :)

LebedevRI, 2019-06-03:
I agree that the fix looks correct (i didn't google that those cameras are aliases),
but it is a general policy of not adding things without test coverage.
And since there is no sample for ZS200, there is no test coverage.

junkyardsparkle, 2019-06-03:
All of those models were *already in the definition*, though; I didn't add any models, just fixed the broken formatting. This is known to be a standard (stupid) thing Panasonic does with regional model names... see the TZ100 definition, for example. Anyway, I have it working for what I needed, so you can close this and wait for somebody to submit samples if you want. In any case, it's worth noting that the current definition doesn't make sense as it is.

LebedevRI, 2019-06-03:
Hmm, looks like i messed up in https://github.com/darktable-org/rawspeed/commit/1a1e8317f3acfceff736297e6bc637804274e4a6 / https://redmine.darktable.org/issues/12295
Which kind-of stresses the point about tests :/

junkyardsparkle, 2019-06-03:
I've had plenty of reasonable assumptions backfire on my lately, so I'm not going to argue against that policy. ;)

closed 2019-06-06 by LebedevRI