darktable-org/rawspeed #184: WebAssembly version of RawSpeed

photopea, 2019-06-25:
Hi, I am the creator of www.Photopea.com , which can open many RAW formats. I implemented my own decoding of TIFF, DNG, NEF, CR2 and ARW in Javascript (the whole pipeline, from a binary file to a sRGB image).

Many users are asking me about the support of less common proprietary RAW formats, such as Fuji RAF. I was thinking about replacing my current work with a WebAssebly version of your library.

Would you be interested in releasing a WebAssembly version?

Also, I was thinking about the right interface. The input could still be a binary file (a sequence of bytes), while the output would be a structure corresponding to a DNG file (it can be an actual binary DNG, or some key-value map, corresponding to the content of DNG). So any software in the world, which supports a DNG format, could switch to supporting all raw formats by including your library. 

It can also simplify your library (e.g. now, you apply a linearization map for each format independently, while you could just add it to the DNG structure without applying it).

LebedevRI, 2019-06-25:
Hi.

> Would you be interested in releasing a WebAssembly version?

I'm sorry, i don't know what that means.
The sources are all [here](https://github.com/darktable-org/rawspeed), presumably you'd need to only compile it into WebAssembly.

photopea, 2019-06-25:
Are you providing any ready-to-use compiled releases of this library, other than the complete Darktable?

It would be great, if you could maintain WebAssembly releases of the library. I think changes in the source code might be necessary, to make the code more robust (that is why I don't feel confident to do it myself). I would be happy to support you financially in this task.

LebedevRI, 2019-06-25:
> Are you providing any ready-to-use compiled releases of this library

The library is a copylib, it can not be built as a shared library; there is no "public API".
It can only be built as a static library, as part of the project that is using it.

photopea, 2019-06-25:
I think you could easily create ANY-RAW-to-DNG converter out of this library. It could be an alternative to Adobe's converter, which is 400 MB. Many people would prefer your converter (as it would probably be smaller and would run on all OSs). 

Doy you think any of developers would be interested in moving in such direction? It would still be usable by Darktable, but the interface (inputs-outputs) might change. I could make donations to you, and ask others about donations.

LebedevRI, 2019-06-25:
This project (https://github.com/darktable-org/rawspeed) is only about the library.
If someone wants to implement open-source raw-to-dng converter with it, they are absolutely free to do so.
(Spoiler: DNG conversion is lossy, it's not a good idea to convert to DNG in general)

photopea, 2019-06-25:
DNG conversion can be both lossy and lossless. In general, for any RAW format, a corresponding data can be stored in DNG. 

I don't know how many people are contributing to this library. I wanted to know, if anybody of them would be interested into making RAW-to-DNG library out of it, as they know Rawspeed the best.

LebedevRI, 2019-06-25:
I'm sorry that i won't be of much help with the WebAssembly part of the question,
i've never encountered that before. If there are specific questions, that aren't
WebAssembly-specific, i may be able to help.

As for the DNG question, conversion to DNG is frowned upon,
so i'm not sure anyone will want to work on that.
Note that there isn't a single *compressor*, encoder here, only decoders, decompressors;
(that may or may not change some day, it might be good for end-to-end testing)

I'd personally recommend using some library directly, not trying to first use some library
to convert format X to format Y, and then using some other library to then read Y.
Speaking of, maybe LibRaw will be more suited for your needs, don't know.

LebedevRI, 2019-06-25:
Also, maybe try asking in https://discuss.pixls.us/

closed 2019-06-25 by LebedevRI

ybouane, 2025-01-28:
Were you able to find a solution to this?
I'm building [libraw-wasm](https://github.com/ybouane/libraw-wasm) and want to add support for rawspeed as the current solution seems to be very slow...

So far I couldn't compile rawspeed to wasm and use it in conjunction to libraw.