darktable-org/rawspeed #185: Sony ARQ support

heckflosse, 2019-06-25:


LebedevRI, 2019-06-25:
There is not a single `*.ARQ` raw on RPU.
Which camera is that?

heckflosse, 2019-06-25:
It's not a native camera format. It's the format the sony software creates (not in camera) when you combine 4 pixelshift ARW files.

Camera is Sony 7R III

heckflosse, 2019-06-25:
@LebedevRI I forgot to mention: I don't mean that kind of support, that gives me a combined image from the ARQ file as PixelShift2DNG does. To use RawSpeed in RT we need to get the 4 frames out of the ARQ file as in this code:
https://github.com/Beep6581/RawTherapee/blob/dev/rtengine/dcraw.cc#L2527

`shot_select` is the number of the frame [0;3]

LebedevRI, 2019-06-25:
That is how i understood it, yes.

LebedevRI, 2019-07-12:
Up, any clue who has appropriate camera and can help contribute nessesary sample?
I think @cryptomilk had some similar camera?

cryptomilk, 2019-07-12:
I don't have that camera but I posted in a Sony A7R3 forum and asked for a shot. If we don't get it I could check a store it and ask if I could take a picture with the camera.

LebedevRI, 2019-07-12:
Samples recieved by RPU, reviewed&accepted, thanks to whoever contributed them
and to @cryptomilk for leading the search!