darktable-org/rawspeed #186: FujiDecompressor algorithm

photopea, 2019-06-25:
Hi, this is a little OT.

I would like to implement the decompression of Fuji RAF files, the method implemented in https://github.com/darktable-org/rawspeed/blob/develop/src/librawspeed/decompressors/FujiDecompressor.cpp

Could any of authors write down a description of the method in English (e.g. in Markdown)? It should not take more than an hour. I can pay you e.g. 150 USD.

I will implement it according to your description, and improve the description (add images, etc) and will publish it on GitHub for the whole world for free :)

LebedevRI, 2019-06-25:
I think you want to post that in https://github.com/LibRaw/LibRaw repo.

closed 2019-06-25 by LebedevRI

photopea, 2019-06-25:
I could not find any code related to RAF compression in LibRaw. Are you sure it can do it?