darktable-org/rawspeed #195: Sigma '.X3F' raw format support

LebedevRI, 2019-06-30:
The X3F was nominally supported, but the support was removed in 1cda622444fb13abe35ff42d1b3a7d940f728100

madwyn, 2020-05-22:
I noticed LibRaw integrated the source code from [x3f tools ](https://github.com/Kalpanika/x3f). I'm wondering if this is a viable approach for rawspeed too?

I've used rawspeed for a while, but I couldn't figure out how the Sigma Quattro X3 sensor would map to the raw image data structure, i.e. blue channel has 2 times width and height compared to other channels. There might be interpolation process needed during the unpack stage.

LebedevRI, 2020-05-22:
> I noticed LibRaw integrated the source code from [x3f tools ](https://github.com/Kalpanika/x3f). > I'm wondering if this is a viable approach for rawspeed too?

Not directly, no.

As a recent example, see GoPro's VC5 decompressor - they did [release source code](https://github.com/gopro/gpr#license-terms),
which in principle could be just bundled into the library, but that is all sorts of ugly.
Instead, in #146 a clean-plate decompressor was written based on that source code.

So something similar can happen for X3F too.

madwyn, 2020-05-22:
Thanks, I see what you mean and that makes sense to me.

I guess it will be a matter of time when "someone" dedicates time to write a proper decompressor based on x3f tools.

LebedevRI, 2020-05-22:
Yep, you are absolutely correct.
Also as per https://github.com/darktable-org/rawspeed#background-of-rawspeed
> Background of RawSpeed
The main objectives were to make a very fast loader that worked for 75% of the cameras out there, and was able to decode a RAW file at close to the optimal speed. The last 25% of the cameras out there could be serviced by a more generic loader, or convert their images to DNG – which as a sidenote usually compresses better than your camera.