darktable-org/rawspeed #196: Some Leica camera definitions missing (Q,Q2,SL,CL)

jenshannoschwalm, 2019-07-01:
Some trivial camera definitions taken from exif data.

[addleicas.txt](https://github.com/darktable-org/rawspeed/files/3344013/addleicas.txt)

LebedevRI, 2019-07-01:
Yes, indeed, DNG cameras are _generally_ not listed in `cameras.xml`;
If you submit *this* as PR i'd merge bug it's not really a bug as it shouldn't really affect anything.

jenshannoschwalm, 2019-07-01:
I understand what you mean but:
From what i understand this is the code section for dt where camera manufacturer, Model ... are prepared in such a way we can use the 'short form'.
When i tried the denoiser profile generator it wrote "Leica" and "Q2" to match.
Without the above patch the profile was not found and had to be manually fixed to "LEICA CAMERA AG" and "LEICA Q2", after including the patch "Leica" and "Q2" were just fine.

Same seems to be the case with matrixes search in adobe_coeff.

Do i oversee something or did i mess things up?

Hanno

LebedevRI, 2019-07-01:
Aha.
Well yeah, all that `<ID>` business is extremely messy :S
Feel free to submit a PR with that diff then.

jenshannoschwalm, 2019-07-01:
Closed because PR #197 solves this

commit 2f7d3ee9 (2019-07-01): Merge pull request #197 from jenshannoschwalm/develop

Add Leica camera descriptions (Q,Q2,SL,CL) for proper <ID>. Fixes #196
closed 2019-07-01 by LebedevRI