darktable-org/rawspeed #200: The bigger picture of rawspeed: raw data versus metadata?

mcourteaux, 2019-09-01:
I know that RawSpeed intentionally does not extract metadata (like thumbnails, EXIF data, and alike) out of the raw files. LibRaw on the other hand does this. Now I'm wondering how one should go about using RawSpeed if you want the fast decoding performance of RawSpeed, but still want to extract a thumbnail for example; and have access to all the other metadata?

It feels bloated to run both LibRaw and RawSpeed on the same file. Reading the file twice, and parsing the file twice. DarkTable probably can extract the metadata. How is it done; and is it considered fast?

I was hoping that instead of providing you with the metadata itself, RawSpeed could tell us the byte-offsets in the file of several metadata sections. For example: the JPEG-thumbnail sits at offset 518 and is 420230 bytes in length. Same for a potential EXIF section and so forth. If this strategy makes sense at least.

So in general: How does one use RawSpeed **and** get the metadata (like photographer name, time the photo was taken, GPS coordinates, etc etc)? I think the answer to this question should probably be somewhere in the README?

LebedevRI, 2019-09-01:
> So in general: How does one use RawSpeed **and** get the metadata (like photographer name, time the photo was taken, GPS coordinates, etc etc)?

By using proper metadata handling library.
darktable does not use libraw at all, uses rawspeed, and has the access to thumbnails, GPS, etc.
https://github.com/Exiv2/exiv2

mcourteaux, 2019-09-01:
Oh perfect! Now I feel a bit silly. Thanks for sharing this information. You could consider adding this hint to the README?

LebedevRI, 2019-09-01:
> Oh perfect! Now I feel a bit silly. Thanks for sharing this information. You could consider adding this hint to the README?

Oh, i was afraid you knew that already.

> You could consider adding this hint to the README?

Yeah, it should be documented better.

To be noted, there were some discussions previously,
it _might_ be the case that a _bit_ more metadata handling will be added,
but it's rather not decided yet. (namely, adobe color matrix and crop-as-jpeg)

madwyn, 2020-05-24:
Was there any discussion to abstract the colour handling part into a separate library? E.g. black level subtraction, colour space conversion and etc.

I read a little through darktable, LibRaw and RawTherapee's source code, they share a lot common in handling colour, maybe it's because they are all originated from dcraw?