darktable-org/rawspeed #201: Canon EOS 250D support

pr0m1th3as, 2019-09-13:
Hi, if it would be of any help I can buy a Grey White Balance Colour Mid-Size Card and upload the cr3 raw files in pixls.us for this particular model. Thanks

LebedevRI, 2019-09-13:
No need for any color targets or anything.
A simple full raw sample set for that camera is missing on https://raw.pixls.us/,
so it you feel like contributing with that, it'd be great.

Roughly: there should be 8 samples: for each aspect ratio, one with each compression mode
That is 3:2 CRAW, 3:2 RAW, 4:3 CRAW, ...
I don't think there is dual-pixel on that camera? (if there is, then one extra dimension, so 16 samples)

Please rename each file accordingly before upload!

lclevy, 2019-09-13:
Also continuous mode (12 bits), please ?

LebedevRI, 2019-09-13:
Right

pr0m1th3as, 2019-09-13:
Hi, thanks for the feedback. I just uploaded a set of 8 images (cRaw + Raw @ 1:1, 3:2, 4:3 + 16:9 aspect ratios) at raw.pixls.us. I am afraid I am not really sure what Iclevy's comment mean with continuous mode (12-bit). If I can be of further assistance, please let me know. Thank you for all the great work in Darktable.

LebedevRI, 2019-09-13:
> Hi, thanks for the feedback. I just uploaded a set of 8 images (cRaw + Raw @ 1:1, 3:2, 4:3 + 16:9 aspect ratios) at raw.pixls.us.

Thank you very much, validated.

> I am afraid I am not really sure what Iclevy's comment mean with continuous mode (12-bit).

From other discussions elsewhere i believe there's suspicion that the raws
that are produced in continuous mode are different from the normal raws.
So it'd be good to have RAW+CRAW raw taken in continuous shooting mode
(raws from the end of the burst, if there are multiple continuous modes - the fastest one i guess).

pr0m1th3as, 2019-09-13:
Hi, thanks for the advice. I just uploaded another 4 images, taken in continuous shooting mode and silent continuous shooting mode (that's the only 2 available options in the 250D). The first batch of 8 photos I uploaded were taken in single shooting mode, so I guess this is it. 2 files are in RAW and 2 are in compressed RAW. I used 3:2 aspect ratio for all 4 new images. If you consider that more images in continuous shooting modes with different aspect ratios could be of any assistance, I would be happy to provide them. I just didn't want to upload lots of files that might be useless for the cause. Thanks

LebedevRI, 2019-09-13:
@lclevy 
https://raw.pixls.us/getfile.php/3395/raw/Canon250D_RAW-silent_cont._shooting-3:2.CR3
https://raw.pixls.us/getfile.php/3394/raw/Canon250D_RAW-cont.shooting_mode-3:2.CR3
https://raw.pixls.us/getfile.php/3392/raw/Canon250D_cRAW-silent_cont._shooting-3:2.CR3
https://raw.pixls.us/getfile.php/3391/raw/Canon250D_cRAW-cont.shooting_mode-3:2.CR3

let me know which ones of these (if any) are useful to that 12-bit question.

lclevy, 2019-09-13:
Thanks you Andreas, i'll check if there are 12bits using my custom tool parse_cr3.py this evening.
Kind regards,
Laurent

pr0m1th3as, 2019-09-13:
I am glad I was able to help somehow.
Best regards,
Andreas

lclevy, 2019-09-13:
Hi,
These CR3 are all 14bits. Maybe 12bits is only for powershot and Digic8 models. 
I heard also recent models (e.g., G7X-III & G5X-II) support high-speed raw bursts which are recorded in a single cr3 file.
https://www.cameralabs.com/canon-powershot-g7x-iii-review/

```
I:\dev\canon_cr3>python parse_cr3.py d:\cr3_samples\250d\Canon250D_cRAW-cont.shooting_mode-3_2.CR3 

sensorInfo(w=6288, h=4056, lb=276, tb=48, rb=6275, bb=4047)
craw
b'Canon EOS 250D'
cmp1(iw=1624, ih=1080, tw=1624, th=1080, d=14, p=4, cfa=1, extra=0, wl=3, b35=0, hsize=544)
cmp1(iw=6288, ih=4056, tw=3144, th=4056, d=14, p=4, cfa=0, extra=0, wl=3, b35=128, hsize=1080)

I:\dev\canon_cr3>python parse_cr3.py d:\cr3_samples\250d\Canon250D_cRAW-silent_cont._shooting-3_2.CR3 

sensorInfo(w=6288, h=4056, lb=276, tb=48, rb=6275, bb=4047)
craw
b'Canon EOS 250D'
cmp1(iw=1624, ih=1080, tw=1624, th=1080, d=14, p=4, cfa=1, extra=0, wl=3, b35=0, hsize=544)
cmp1(iw=6288, ih=4056, tw=3144, th=4056, d=14, p=4, cfa=0, extra=0, wl=3, b35=128, hsize=1080)

I:\dev\canon_cr3>python parse_cr3.py d:\cr3_samples\250d\Canon250D_RAW-cont.shooting_mode-3_2.CR3 

sensorInfo(w=6288, h=4056, lb=276, tb=48, rb=6275, bb=4047)
raw
b'Canon EOS 250D'
cmp1(iw=1624, ih=1080, tw=1624, th=1080, d=14, p=4, cfa=1, extra=0, wl=0, b35=0, hsize=112)
cmp1(iw=6288, ih=4056, tw=3144, th=4056, d=14, p=4, cfa=0, extra=0, wl=0, b35=0, hsize=216)

I:\dev\canon_cr3>python parse_cr3.py d:\cr3_samples\250d\Canon250D_RAW-silent_cont._shooting-3_2.CR3 

sensorInfo(w=6288, h=4056, lb=276, tb=48, rb=6275, bb=4047)
raw
b'Canon EOS 250D'
cmp1(iw=1624, ih=1080, tw=1624, th=1080, d=14, p=4, cfa=1, extra=0, wl=0, b35=0, hsize=112)
cmp1(iw=6288, ih=4056, tw=3144, th=4056, d=14, p=4, cfa=0, extra=0, wl=0, b35=0, hsize=216)
```

LebedevRI, 2019-09-13:
> These CR3 are all 14bits.

:( i'll drop them then.

lclevy, 2019-09-13:
about raw burst mode: 
https://helpx.adobe.com/camera-raw/kb/raw-burst-mode.html
"Canon CR3 Raw Burst Mode image rolls"