darktable-org/rawspeed #203: line 441 not inside image

oib, 2019-10-19:
After creating a HDR image from some DNG images, i get a notification that is not available
the file exists but is not viewable
 
```
[rawspeed] (IMG_20191018_200915.1-hdr.dng) void rawspeed::DngDecoder::handleMetadata(const rawspeed::TiffIFD*), line 441: Rectangle (0, 0, 2592, 1944) not inside image (0, 0, 2576, 1928).
allocation failed???
```

Debian/testing
darktable Version: 2.6.2-3
Depends: libc6 (>= 2.27), libcairo2 (>= 1.14.0), libcolord-gtk1 (>= 0.1.20), libcolord2 (>= 1.4.3), libcups2 (>= 1.7.0), libcurl3-gnutls (>= 7.56.1), libexiv2-14 (>= 0.25), libflickcurl0, libgcc1 (>= 1:4.0), libgdk-pixbuf2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.39.4), libgomp1 (>= 4.9), libgphoto2-6 (>= 2.5.10), libgphoto2-port12 (>= 2.5.10), libgraphicsmagick-q16-3 (>= 1.3.5), libgtk-3-0 (>= 3.21.5), libilmbase24 (>= 2.3.0), libjpeg62-turbo (>= 1.3.1), libjson-glib-1.0-0 (>= 0.13.2), liblcms2-2 (>= 2.2+git20110628), liblensfun1 (>= 0.3.2), liblua5.3-0, libopenexr24 (>= 2.3.0), libopenjp2-7 (>= 2.0.0), libosmgpsmap-1.0-1 (>= 1.1.0), libpango-1.0-0 (>= 1.14.0), libpangocairo-1.0-0 (>= 1.14.0), libpng16-16 (>= 1.6.2-1), libpugixml1v5 (>= 1.6), librsvg2-2 (>= 2.14.4), libsecret-1-0 (>= 0.7), libsoup2.4-1 (>= 2.47.4), libsqlite3-0 (>= 3.6.0), libstdc++6 (>= 5.2), libtiff5 (>= 4.0.3), libwebp6 (>= 0.5.1), libx11-6, libxml2 (>= 2.7.4), libxrandr2 (>= 2:1.2.99.3), zlib1g (>= 1:1.2.0), libjs-prototype, libjs-scriptaculous

LebedevRI, 2019-10-19:
Sounds like broken DNG to me.
Can you upload it somewhere?

oib, 2019-10-19:
[IMG_20191018_200915.1-hdr.dng.zip](https://github.com/darktable-org/rawspeed/files/3746741/IMG_20191018_200915.1-hdr.dng.zip)

LebedevRI, 2019-10-19:
And it was produced by? darktable hdr merge feature?
Can you also post one original raw that was used to assemble this hdr?

oib, 2019-10-19:
[IMG_20191018_200915.zip](https://github.com/darktable-org/rawspeed/files/3746770/IMG_20191018_200915.zip)
produced by darktable "create HDR" (from selected images)

LebedevRI, 2019-10-19:
Yep, as expected.
It should filter out `ACTIVEAREA` exif tag.
Please report that to https://github.com/darktable-org/darktable/issues/new/choose

closed 2019-10-19 by LebedevRI