darktable-org/rawspeed #211: Don't set frame or stack-size via compiler commandline

cryptomilk, 2019-12-30:
Hi Roman,

I've talked to musl developers on IRC in #musl @ freenode about setting stack and frame size.

https://github.com/darktable-org/rawspeed/blob/develop/cmake/compiler-warnings.cmake#L16

This should be done using `pthread_attr_setstacksize()` if really needed for musl libc.

LebedevRI, 2019-12-30:
`-W<>` are *warning* flags, they do not actually set the limits.

LebedevRI, 2019-12-30:
To allow forward progress here, can you specify not *what* (current subj), but *why*?

cryptomilk, 2019-12-31:
You probably get warnings with this if you enable AddressSanitzer. It does some nasty things.

LebedevRI, 2019-12-31:
Please feel free to post coherent bugreport, if one exists and it's not a duplicate of #212.

closed 2019-12-31 by LebedevRI