darktable-org/rawspeed #214: DNG from CR3 not opening - AbstractDngDecompressor error

alexvanderberkel, 2020-01-02:
I converted a CR3 into a DNG file with Adobe DNG converter. Currently darktable is not able to open the DNG file. Terminal shows following error message: 

[rawspeed] (file.dng) void rawspeed::AbstractDngDecompressor::decompress() const, line 208: Too many errors encountered. Giving up. First Error:
rawspeed::LJpegDecompressor::LJpegDecompressor(const rawspeed::ByteStream&, const rawspeed::RawImage&), line 42: Unexpected component count (2)
allocation failed???

In addition the following two error are shown consecutively

(darktable:7246): GLib-GObject-WARNING **: 10:45:48.696: invalid cast from 'GtkBox' to 'GtkGrid'

(darktable:7246): Gtk-CRITICAL **: 10:45:48.696: gtk_grid_attach_next_to: assertion 'GTK_IS_GRID (grid)' failed

LebedevRI, 2020-01-02:
Please attach the DNG in question.

alexvanderberkel, 2020-01-02:
DNG files are not allowed as an attachmend and are to big. Link is valid temporarily https://send.firefox.com/download/e60737a68263e269/#c2O81MxWHaJ0R5aZ381qPw

LebedevRI, 2020-01-02:
> DNG files are not allowed as an attachmend and are to big.

Yep. Putting it in a .zip and attaching works.

alexvanderberkel, 2020-01-02:
> > DNG files are not allowed as an attachmend and are to big.
> 
> Yep. Putting it in a .zip and attaching works.

Zip was unfortunately still over the 10MB limit...

LebedevRI, 2020-01-02:
Hmm, that's a new restriction then, i don't recall seeing it before.
That being said i failed to download, link expired.

alexvanderberkel, 2020-01-02:
Will send you the link via email to avoid that the link will expire to some reasons

LebedevRI, 2020-01-02:
Is that DualISO?
While it is easy to convince rawspeed to load it
```diff
diff --git a/src/librawspeed/decompressors/LJpegDecompressor.cpp b/src/librawspeed/decompressors/LJpegDecompressor.cpp
index f5d6d86c..64b2fb2b 100644
--- a/src/librawspeed/decompressors/LJpegDecompressor.cpp
+++ b/src/librawspeed/decompressors/LJpegDecompressor.cpp
@@ -39,6 +39,7 @@ LJpegDecompressor::LJpegDecompressor(const ByteStream& bs, const RawImage& img)
     ThrowRDE("Unexpected data type (%u)", mRaw->getDataType());
 
   if (!((mRaw->getCpp() == 1 && mRaw->getBpp() == 2) ||
+        (mRaw->getCpp() == 2 && mRaw->getBpp() == 4) ||
         (mRaw->getCpp() == 3 && mRaw->getBpp() == 6)))
     ThrowRDE("Unexpected component count (%u)", mRaw->getCpp());
```
it is then discarded in darktable.

alexvanderberkel, 2020-01-02:
It is Dual Pixel Raw

LebedevRI, 2020-01-02:
Guessed as much.
I'll relax rawspeed-side check, but i'm no going to do about darktable-side thing here.
<rant>
Partly because now darktable simply does not load for me, hangs for a minute and segfaults.
It's amazing to see my multi-year efforts of stabilizing that code gone this fast.
I guess i will be forced to move the shop elsewhere after all.
</rant>
Feel free to open issue there.

closed 2020-01-03 by LebedevRI

jenshannoschwalm, 2020-01-05:
@LebedevRI @alexvanderberkel  I got darktable decoding dual pixel dng files.

There is one more thing in rawspeed that "requires relaxing". `bool DngDecoder::decodeBlackLevels(const TiffIFD* raw) {` in DngDecoder.cpp should accept cpp==2 too.

So i suggest it to be
``` 
 if ((mRaw->getCpp() < 1) || (mRaw->getCpp() > 2))
    return false;

```
Can you agree here? And could you modify this or do you want me to do a pr?
BTW it would be nice to push current rawspeed to dt :-)