darktable-org/rawspeed #215: DNG BlackLevel handling when cpp != 1

LebedevRI, 2020-01-05:
@LebedevRI @alexvanderberkel  I got darktable decoding dual pixel dng files.

There is one more thing in rawspeed that "requires relaxing". `bool DngDecoder::decodeBlackLevels(const TiffIFD* raw) {` in DngDecoder.cpp should accept cpp==2 too.

So i suggest it to be
``` 
 if ((mRaw->getCpp() < 1) || (mRaw->getCpp() > 2))
    return false;

```
Can you agree here? And could you modify this or do you want me to do a pr?
BTW it would be nice to push current rawspeed to dt :-)

_Originally posted by @jenshannoschwalm in https://github.com/darktable-org/rawspeed/issues/214#issuecomment-570887227_

More specifically, as per https://www.adobe.com/content/dam/acom/en/products/photoshop/pdfs/dng_spec_1.4.0.0.pdf, all that blacklevel info is not at all specific to cpp=1/cpp=2.
See `BLACKLEVELREPEATDIM`, `BLACKLEVEL`, `BLACKLEVELDELTAV`, `BLACKLEVELDELTAH`.
Which only highlights how broken the current 4-channel `mRaw->blackLevelSeparate` is.

jenshannoschwalm, 2020-01-05:
Ok, i agree on missing proper blacklevel support.

You know http://wwwimages.adobe.com/www.adobe.com/content/dam/acom/en/products/photoshop/pdfs/dng_spec_1.5.0.0.pdf ?
There is even more to be worked on.

Do you have priority list of more "things-to-be-fixed" regarding dng?

LebedevRI, 2020-01-05:
> Ok, i agree on missing proper blacklevel support.
> 
> You know http://wwwimages.adobe.com/www.adobe.com/content/dam/acom/en/products/photoshop/pdfs/dng_spec_1.5.0.0.pdf ?

Oh, nice, i should update my bookmark then.

> There is even more to be worked on.
> 
> Do you have priority list of more "things-to-be-fixed" regarding dng?

I don't think i have any list regarding DNG. (well, other than the general issue of 'multi-image raw support')

madwyn, 2020-05-20:
Thanks for pointing me to the right direction. I'm wondering will this be fixed in the `develop` branch?

RGB DNG files have `cpp=3` and they also can have black levels label `50714 Black Level`.

It's confusing that black level handling needs to consider `cpp`, I can understand `<1` or `>3` looks "bad".

LebedevRI, 2020-05-20:
Just so we have frame of reference, can you upload here (you may need to zip it first) some such DNG?

parafin, 2020-05-20:
By RGB DNG do we mean LinearRaw DNG with SamplesPerPixel equal to 3?

LebedevRI, 2020-05-20:
> By RGB DNG do we mean LinearRaw DNG with SamplesPerPixel equal to 3?

Yes. I'd be particularly interested in some sane sample, with different black levels
in all 3 channels (as opposed to the dual-pixel dng's..)

madwyn, 2020-05-20:
The file size is well over 10MB, here is the link:
https://we.tl/t-vRmIDWqgyY

madwyn, 2020-05-20:
The file was originally a X3F image converted by x3f tools to DNG, then I used Adobe DNG converter "converted" it again to get some compression. I would assume the tags should be intact.

@parafin, you are right, it's LinearRaw with `cpp=3`.

LebedevRI, 2020-05-20:
[_SDI0247.X3F.dng-partaa.zip](https://github.com/darktable-org/rawspeed/files/4658384/_SDI0247.X3F.dng-partaa.zip)
[_SDI0247.X3F.dng-partab.zip](https://github.com/darktable-org/rawspeed/files/4658385/_SDI0247.X3F.dng-partab.zip)
[_SDI0247.X3F.dng-partac.zip](https://github.com/darktable-org/rawspeed/files/4658386/_SDI0247.X3F.dng-partac.zip)
[_SDI0247.X3F.dng-partad.zip](https://github.com/darktable-org/rawspeed/files/4658387/_SDI0247.X3F.dng-partad.zip)

madwyn, 2020-05-20:
> > By RGB DNG do we mean LinearRaw DNG with SamplesPerPixel equal to 3?
> 
> Yes. I'd be particularly interested in some sane sample, with different black levels
> in all 3 channels (as opposed to the dual-pixel dng's..)

I extracted the values using exiftool, they look identical:

```
50714 Black Level                     : 206.875 206.875 206.875
```

madwyn, 2020-05-29:
When reading the code, I also noticed that the White Level tag is always treated as an integer filed, but according to the DNG spec, it's a filed contains as many values as the "plane" number.

Will this behaviour get changed, just like black levels?