darktable-org/rawspeed #224: Panasonic DC-LX100M2 support

lake-effect, 2020-02-01:
Filing this to track support for this camera. I'm providing raws to raw.pixls.us in the meantime.

Log on startup with camera connected through USB:

```
RawSpeed:Unable to find camera in database: 'Panasonic' 'DC-LX100M2' ''
Please consider providing samples on <https://raw.pixls.us/>, thanks!
[rawspeed] (20200201_0046.RW2) bool rawspeed::RawDecoder::checkCameraSupported(const rawspeed::CameraMetaData*, const string&, const string&, const string&), line 166: Camera 'Panasonic' 'DC-LX100M2', mode '' not supported, and not allowed to guess. Sorry.
```

lake-effect, 2020-02-01:
It looks like someone beat me to uploading by about a year: https://raw.pixls.us/data/Panasonic/DC-LX100M2/

LebedevRI, 2020-02-01:
Support for that camera was added in https://redmine.darktable.org/issues/12431 / 970c7e666fb585a5498d2a0de6bcdee8f594735e, which was propagated into darktable in 090a1e8fcdfe9c51fd9dfca10b867a9004395185, and was first released in darktable release-2.6.0rc1
IOW that camera is most certainly not supported in dt-2.4

closed 2020-02-01 by LebedevRI

lake-effect, 2020-02-01:
Thanks for letting me know! Easy enough to update.