darktable-org/rawspeed #225: Darktable cannot open the hdr dng image created with Lightroom classic.

KristijanZic, 2020-02-20:
**Describe the bug**
Darktable cannot open the hdr dng image created with Lightroom classic in darkroom.
Also the thumbnails for such images are not shown after importing them.

When opening such image Darktable throws this error:
[image]

**To Reproduce**
1. Download this hdr dng image that I've created in Lightroom classic (created from multiple CR2 images I shot on my Canon 77D):
https://drive.google.com/open?id=1SouINXVW3I2jiWxb-77sQS44x0sjvTkR
2. Import in in Darktable
3. Double click the imported image to open it in darkroom.
You should now see the error.

**Expected behavior**
Dartktable lightable should show thumbnails for such images after importing
Darktable darkroom should load the image and enable the editing of such an image.

**Platform (please complete the following information):**
 - Darktable Version: 3.0.0, 3.0.1 master
OS: Ubuntu 19.10 x86_64 
Host: Inspiron 13-7359 
Kernel: 5.3.0-40-generic 
Uptime: 1 day, 17 hours, 32 mins 
Packages: 1697 (dpkg), 36 (flatpak), 28 (snap)
Shell: bash 5.0.3 
Resolution: 1920x1080 
DE: GNOME 3.34.1 
Theme: Adwaita-dark [GTK2/3] 
Icons: Yaru [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i7-6500U (4) @ 3.100GHz 
GPU: Intel Skylake GT2 [HD Graphics 520] 
Memory: 6843MiB / 7818MiB

jenshannoschwalm, 2020-02-20:
btw, with exiv2 -pa there seem to be no exif decoding problems, this is clearly another example of a dual pixel file.
```
Exif.SubImage1.SamplesPerPixel               Short       1  2
Exif.SubImage1.BlackLevel                    Rational    8  524288/256 524288/256 524032/256 524032/256 524544/256 524544/256 524032/256 524032/256
```
@LebedevRI here is one example for the need to analyse more than 4 black level values.

LebedevRI, 2020-02-20:
Aha.
So that cryptic error most likely was meant to tell us that the `uncompress`
has run out of space in output (destination) buffer.

LebedevRI, 2020-02-20:
That will still require darktable-side fix, which i guess is tracked in https://github.com/darktable-org/darktable/issues/4006

jenshannoschwalm, 2020-02-22:
ooh, one more problem, this file is a sraw with 16bit floats. Simple not supported atm...

LebedevRI, 2020-02-22:
It seems i got bamboozled, that can't possibly be dual-pixel raw, let me see..

LebedevRI, 2020-02-22:
Welp, that doesn't look like it decoded correctly :)
[image]

jenshannoschwalm, 2020-02-22:
Whow.
You saw that this is not 32bit but 16bit floats?

LebedevRI, 2020-02-22:
This is better, i suppose.
[image]

closed 2020-02-22 by LebedevRI

LebedevRI, 2020-02-22:
@KristijanZic cheers.

LebedevRI, 2020-02-23:
@jenshannoschwalm 
> You saw that this is not 32bit but 16bit floats?

Note that non-32-bit floatint-point Deflate-compressed images are expanded into 32-bit floating point
https://github.com/darktable-org/rawspeed/blob/cd380bb9a209bd2e7a0e7022b0cab04528d151e7/src/librawspeed/decompressors/DeflateDecompressor.cpp#L236-L244

jenshannoschwalm, 2020-02-23:
Thanks for pointing to this :-)!