darktable-org/rawspeed #226: OSX Build error

MStraeten, 2020-03-25:
found following error on building darktable on OSX after updating to XCode 11.4

```
[  6%] Building CXX object src/external/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/common/RawImage.cpp.o
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:408:23: error:
      variable 'threads' must have explicitly specified data sharing attributes
  for (int i = 0; i < threads; i++) {
                      ^~~~~~~
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:404:34: note:
      explicit data sharing attribute requested here
#pragma omp parallel for default(none)                                         \
                                 ^
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:409:47: error:
      variable 'height' must have explicitly specified data sharing attributes
    int y_offset = std::min(i * y_per_thread, height);
                                              ^~~~~~
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:404:34: note:
      explicit data sharing attribute requested here
#pragma omp parallel for default(none)                                         \
                                 ^
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:412:33: error:
      variable 'task' must have explicitly specified data sharing attributes
    RawImageWorker worker(this, task, y_offset, y_end);
                                ^~~~
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:404:34: note:
      explicit data sharing attribute requested here
#pragma omp parallel for default(none)                                         \
                                 ^
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:409:33: error:
      variable 'y_per_thread' must have explicitly specified data sharing
      attributes
    int y_offset = std::min(i * y_per_thread, height);
                                ^~~~~~~~~~~~
/Users/martinstraeten/src/darktable/src/external/rawspeed/src/librawspeed/common/RawImage.cpp:404:34: note:
      explicit data sharing attribute requested here
#pragma omp parallel for default(none)                                         \
                                 ^
4 errors generated.
make[2]: *** [src/external/rawspeed/CMakeFiles/rawspeed.dir/src/librawspeed/common/RawImage.cpp.o] Error 1
make[1]: *** [src/external/rawspeed/CMakeFiles/rawspeed.dir/all] Error 2
make: *** [all] Error 2
martinstraeten@Martins-MacBook-Pro build %
```

LebedevRI, 2020-03-25:
I'm guessing that is darktable <2.6.3?

MStraeten, 2020-03-25:
no, current master
`git pull --recurse-submodules https://github.com/darktable-org/darktable.git master`

LebedevRI, 2020-03-25:
Hm, i guess we are hitting this
https://github.com/darktable-org/rawspeed/blob/c25c748a5c5127d2380f92d53e4eb3f1931df9b6/cmake/src-dependencies.cmake#L70-L72
Can you try changing that to
```diff
  if((CMAKE_CXX_COMPILER_ID STREQUAL "Clang" AND
      CMAKE_CXX_COMPILER_VERSION VERSION_LESS 7.0) OR
-     (CMAKE_CXX_COMPILER_ID STREQUAL "AppleClang"
-      # XCode 10 is broken. Maybe XCode 11 will be ok?
+     (CMAKE_CXX_COMPILER_ID STREQUAL "AppleClang" AND
+      CMAKE_CXX_COMPILER_VERSION VERSION_LESS 11.4
     ))
```
or something like that. Does that help?

MStraeten, 2020-03-25:
cmake gives
```
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- The C compiler identification is AppleClang 11.0.3.11030032
```
so 
changing to
```
(CMAKE_CXX_COMPILER_ID STREQUAL "AppleClang" AND
CMAKE_CXX_COMPILER_VERSION VERSION_LESS 11.0.3
```
helps

MStraeten, 2020-03-25:
letzte funktionierende Compilerversion war:
(CMAKE_C_COMPILER_VERSION "11.0.0.11000033")

LebedevRI, 2020-03-25:
Well then, thank you for trying, please submit that as a PR :)

MStraeten, 2020-03-25:
i have no idea how to do a pull request for this

MStraeten, 2020-03-31:
ok, after reading about github porcess, i just tried - hope that's ok ;)

closed 2020-03-31 by LebedevRI

LebedevRI, 2021-08-20:
@MStraeten @parafin sorry for necro-posting, but i'm revisiting this area of cmake, and i'm somewhat confused.
This workarounds https://bugs.llvm.org/show_bug.cgi?id=35873, which was fixed in LLVM/Clang 7: https://godbolt.org/z/v9dcT9rK5
But as per https://en.wikipedia.org/wiki/Xcode#Xcode_7.0_-_10.x_(since_Free_On-Device_Development)_2, the first XCode version based on LLVM 7 was `XCode 10.2` (aka `10.0.1 (clang-1001.0.46.3)`).

So how come
> (CMAKE_CXX_COMPILER_VERSION "11.0.0.11000033")

which i believe belongs to XCode 11.0, is still broken?

I guess my main question is: which is the first XCode version for which https://godbolt.org/z/v9dcT9rK5 is fixed?

PR #227 (2020-03-31): Update src-dependencies.cmake
fixes #226 OSX Build error like mentioned there