darktable-org/rawspeed #229: Image cropping for DNGs

jenshannoschwalm, 2020-04-13:
As the rawspeed README.rst says

> Rawspeed
> - optionally crops off  “junk” areas of images, containing no valid image information.

> RawSpeed does **NOT**…
> - crop the image to the same sizes as manufactures, but supplies biggest possible images.

At least while using dt for DNGs rawspeed always crops to the area defined by `DefaultCropOrigin` and `DefaultCropSize`. (I know that dt offers hidden options to change these crops.)

This helps preventing artefacts on the borders and is a safe solution but it does not in general offer the largest image size.

I think this contradicts the README. 

Questions arising:
1. Are you going to keep the behavior? --> Mentioning this in the README?
2. For some cameras this default cropping is pretty strong, the sensor offers more data seen in other applications. If i want to sort of add an exception list this would have to be done on the application side? Or would this needs to be done somehow in `DngDecoder::handleMetadata`

jenshannoschwalm, 2020-04-13:
My bad, sorry abut the noise. Seems to be handled already in dt now. I missed the fixing commit.

LebedevRI, 2020-04-15:
I think making docs better is always good. As for dt, i'm not following the changes closely,
so i'm not really sure what you are talking about, so might be helpful to add some references.

jenshannoschwalm, 2020-04-15:
dt of course uses the sensor size for image->width/height.

The used area is restricted in rawprepare module to the cropped area deteted by rawspeed. Hidden options allow to modify this cropping to anything. So my point is, why do you crop further than ActiveArea?

In dt we can copy module's content to other images, fine - but this also effects the whitepoint. So - if i want the complete sensor data i have either 'tune' every image or find another way.

Exif data for an example image:
```
Exif.SubImage1.ImageWidth                    Long        1  8424
Exif.SubImage1.ImageLength                   Long        1  5632
Exif.SubImage1.ActiveArea                    Short       4  0 0 5632 8392
Exif.SubImage1.DefaultCropOrigin             Short       2  12 24
Exif.SubImage1.DefaultCropSize               Short       2  8368 5584
```

LebedevRI, 2020-04-15:
> So my point is, why do you crop further than ActiveArea?

Crop *where*? Can you perhaps show the particular DNG in question?
I assume it is cropped as it should be cropped as per the info in the metadata.

jenshannoschwalm, 2020-04-16:
All images are atm cropped to the area defined by DefaultCropOrigin & DefaultCropSize.
But if rawspeed would follow it's docs it would return ActiveArea as the cropped region.

>    RawSpeed does NOT…
        crop the image to the same sizes as manufactures, but supplies biggest possible images.

LebedevRI, 2020-04-16:
I think we're talking past each other here.
Firstly, can you upload the DNG in question? (you need to archive it first)
Secondly, reading https://github.com/darktable-org/rawspeed/blob/708f3cdc7bf1e1a7cbe52c479a625bda6d08455c/src/librawspeed/common/RawImage.cpp#L258-L279 i think the cropping doesn't actually delete the data, it again only adjusts the crop.
Thirdly, this isn't only about DefaultCropOrigin & DefaultCropSize, there is also `TrimBounds` DNG opcode.

jenshannoschwalm, 2020-04-16:
> I think we're talking past each other here.

It seems so :-) Here is one example i uploaded a time ago.

https://raw.pixls.us/getfile.php/3204/nice/Leica%20-%20Q2%20-%2014bit%20uncompressed%20(3:2).DNG

Yes - the cropping doesn't delete the data. I think it adjusts the crop "too far". 

Yes there might be 2 opcodes involved, BadPixelsList should accept pixels in the whole ActiveArea and TrimBounds that might "force" us to crop. I confess i didn't test this (unfortunately i don't even know how to. I could add some fprintf of couse in rawspeed  code)

https://github.com/darktable-org/rawspeed/blob/b9eca12d91ac96e864f40e3c6c093f7957f22227/src/librawspeed/decoders/DngDecoder.cpp#L450-L470

LebedevRI, 2020-04-17:
> Yes - the cropping doesn't delete the data. I think it adjusts the crop "too far".

DNG is kinda special. It adjusts the crop as specified in the metadata.
In that case i indeed don't see bad pixels in the cropped pixels,
but i presume the crop is used to align the centre of the image and the sensor centre,
while maintaining image aspect ratio.

So i'm all for documentation change :)

jenshannoschwalm, 2020-05-21:
> but i presume the crop is used to align the centre of the image ...

I guess you might be right here. I looked in the specs and searched the net for any other docs about this topic, i could find nothing! So - even if other soft uses this in a different way - no other safe way to go maybe.

closed 2020-05-21 by jenshannoschwalm