darktable-org/rawspeed #232: Fujifilm X-T4 "compressed lossless" raw support

LebedevRI, 2020-05-09:
I have now uploaded three .RAF files (on a X-rite colorchecker classic):
1) uncompressed
2) Compressed lossless
3) Compressed

_Originally posted by @frli8848 in https://github.com/darktable-org/darktable/issues/4943#issuecomment-625797717_

kmilos, 2021-12-15:
Apart from X-S10 and X-E4, the lossy mode is also present for GFX 100 (no lossy samples on RPU though), GFX100S and GFX50S II, perhaps it is all the same?

kmilos, 2024-01-22:
Cf. https://github.com/LibRaw/LibRaw/compare/0.20.2...0.21.2#diff-433cfe5a298e40678f55be557c31e36344f501e5587fd9da4a49769c6e706c61

The header "version" byte was renamed to "lossless" and values 0 and 1 are accepted, with lossy implementation added.

jdneumeyer77, 2024-03-22:
@kmilos I might not have time right away, but I can look into it. Is there a specific commit? That's a pretty wide-range you provided.

kmilos, 2024-03-23:
> Is there a specific commit?

Unfortunately not, they usually just push the whole milestone at once. The link should take you just to the relevant file though.