darktable-org/rawspeed #239: How to link with RawSpeed dynamically?

mcourteaux, 2020-08-25:
I was investigating the licence for RawSpeed, and it requires my application to dynamically link with RawSpeed, in such a way that someone can build their own RawSpeed shared library and replace the packaged library with their own in my application, as long as they make sure their version is interface compatible.

However, the CMakeLists only builds a static library, judging by this line:
https://github.com/darktable-org/rawspeed/blob/5743628b9d4649aa027df62b91f4882e78de7063/CMakeLists.txt#L224

This makes it impossible to use the library without modifications, I think. The minimal modification is changing this to SHARED, I understand. Next, due to the fact that this now becomes a shared library, instead of a static one, the linker is now looking for `_rawspeed_get_number_of_processor_cores`. This function is implemented in my application, and when RawSpeed was statically linked in, it worked fine, but now it's dynamically linked, the linker is looking for that symbol when producing the `librawspeed.so`/`librawspeed.dylib`, which it obviously can't find.

What would be the general guidelines here, in order to comply with the license?

LebedevRI, 2020-08-25:
Before looking any further, sanity check

> it requires my application to dynamically link 

I'm guessing said application is not under FLOSS license, something closed-source/proprietary?

mcourteaux, 2020-08-25:
Yes, indeed. It's my attempt at building the fastest and most elegant raw processor / photo editor, as alternative to Lightroom. If interested: https://silvernode.io/

LebedevRI, 2020-08-25:
One more thing to note, is that no symbols are being exported from such `librawspeed.so`,
because no symbols are annotated as API, and `-fvisibility=hidden` is specified.
It's more or less intentional that way.

So i would say, the common ground is to have some "shim" `libsilvernode-rawspeed.so`,
for which the sources **are** available, that statically links to RawSpeed,
contains `rawspeed_get_number_of_processor_cores()` function,
and provides a wrapper whatever functionality you actually need.

mcourteaux, 2020-08-25:
> So i would say, the common ground is to have some "shim" libsilvernode-rawspeed.so,

Aha, yes this seems like an option, but a very clumsy approach, IMO. Because in the end, literally no-one will be interested in replacing the working version of RawSpeed with a custom-compiled version of RawSpeed in such an application. I think in general, people would be more interested in my next proposal below.

> no symbols are annotated as API, and -fvisibility=hidden is specified

I'd be more than happy to work on RawSpeed itself and make PR which makes it usable as a shared library as well. Although then you'd depend on the compiler's C++ ABI, but as far as I understand, it would suffice to just specify which compiler to use to produce the `librawspeed.so`. Any thoughts on the idea of making RawSpeed shared-library compatible?

LebedevRI, 2020-08-25:
> Because in the end, literally no-one will be interested in replacing the working version of RawSpeed with a custom-compiled version of RawSpeed in such an application

Yep. I believe that part of the license is about having to actually publish all the changes made to LGPL-licensed code.
Having the ability to link to link to rebuilt library is a tool to ensure that all the changes were actually provided.

> > no symbols are annotated as API, and -fvisibility=hidden is specified
> 
> I'd be more than happy to work on RawSpeed itself and make PR which makes it usable as a shared library as well.
> Any thoughts on the idea of making RawSpeed shared-library compatible?

There are two things to consider:
1. The moment there is a shared library, "every" project using RawSpeed will be forced/suggested/etc
    by distro package maintainers to switch away from bundled/copylib version of the library, to `.so` version.
    This is going to be a major PITA, because e.g. different set of cameras is supported by different versions,
    so you simply can't blindly strip-out bundled rawspeed, and use system version.
    This is pretty much a show-stopper here, because i know it will be a showstopper for darktable.
2. Having such a bare librawspeed.so will mean giving up on proper LTO+`-fvisibility=hidden`.
    I don't have measurements to prove it, but i do expect it to matter.
    OTOH yes, that would make PGO easier..

Both of these issues, especially the second one, will not be there for a "shim".

> Although then you'd depend on the compiler's C++ ABI, but as far as I understand, it would suffice to just specify which compiler to use to produce the `librawspeed.so`.

I'm not very sure how that is relevant here? That's up to those who build and distribute it.

mcourteaux, 2020-08-25:
> Yep. I believe that part of the license is about having to actually publish all the changes made to LGPL-licensed code.
> Having the ability to link to link to rebuilt library is a tool to ensure that all the changes were actually provided.

Yes, and that's where the license doesn't trust users of the library. I'm literally using the master branch of RawSpeed, and every change I made is merged back in by Pull Requests. It's kind of a bummer that we now have to go through this hassle, just for legal purposes, while the actual intent of the license is fulfilled: contributing to the project in an open-source manner.

> so you simply can't blindly strip-out bundled rawspeed, and use system version.

I understand, but I don't see why that is a problem. If that is a problem for some application, then one should just not use the system version. Additionally the C++ ABI will be an issue if a package manager would provide shared libraries for RawSpeed (see below), unless a C-wrapper is provided.

I think it would be useful to have a build option in the CMakeLists that allows you to build a shared library from source, for the projects who would make good use of it, like mine. Having this option in CMakeLists doesn't really imply that therefore shared libraries should be available through a package manager?

> Having such a bare librawspeed.so will mean giving up on proper LTO. I don't have measurements to prove it, but i do expect it to matter.

I wouldn't know either, but I expect this to have minimal impact as most of the time spent in RawSpeed code is in the decoding/decompression loops etc.

> mean giving up on `-fvisibility=hidden`

Why is this important? Honest question here: I don't know, and I'm not too familiar with this.

> OTOH yes, that would make PGO easier..

I don't know what PGO is.

> I'm not very sure how that is relevant here? That's up to those who build and distribute it.

The name mangling, struct packing, class layouts, vtables, etc depends on the compiler and the settings. One cannot simply distribute a shared `librawspeed.so` with the closed-source application, and expect a hacker who wishes to replace `librawspeed.so` to know which ABI is used. So I guess one would need to specify the compiler and the settings, to make sure you comply with the LGPL2.1 license.

LebedevRI, 2020-08-25:
> Yes, and that's where the license doesn't trust users of the library. I'm literally using the master branch of RawSpeed,

Note that `master` branch is _very_ outdated.
Currently `stable` is what i'd recommend to be used, or `develop` if feeling lucky.

> and every change I made is merged back in by Pull Requests. It's kind of a bummer that we now have to go through this hassle, just for legal purposes, while the actual intent of the license is fulfilled: contributing to the project in an open-source manner.

Note that it's not about actually upstreaming every change, but simply having them in the open.
For example, it wouldn't be incorrect to just have all these changes for building shared library published but not upstream.

> > I'm not very sure how that is relevant here? That's up to those who build and distribute it.
> Additionally the C++ ABI will be an issue if a package manager would provide shared libraries for RawSpeed (see below), unless a C-wrapper is provided.
> 
> The name mangling, struct packing, class layouts, vtables, etc depends on the compiler and the settings. One cannot simply distribute a shared `librawspeed.so` with the closed-source application, and expect a hacker who wishes to replace `librawspeed.so` to know which ABI is used. So I guess one would need to specify the compiler and the settings, to make sure you comply with the LGPL2.1 license.

That is a traditional problem for libraries with C++ public API, yes.
IMO discussing it only creates confusion.
It's downstream/packaging/consumer issue.

> > so you simply can't blindly strip-out bundled rawspeed, and use system version.
> 
> I understand, but I don't see why that is a problem. If that is a problem for some application, then one should just not use the system version.
> I think it would be useful to have a build option in the CMakeLists that allows you to build a shared library from source, for the projects who would make good use of it, like mine.
> Having this option in CMakeLists doesn't really imply that therefore shared libraries should be available through a package manager?

Well, that's the thing, i'd like to agree with you, but that isn't the wording some radical distros use(d?).
See e.g. https://lwn.net/Articles/660429/
 
> > Having such a bare librawspeed.so will mean giving up on proper LTO. I don't have measurements to prove it, but i do expect it to matter.
> 
> I wouldn't know either, but I expect this to have minimal impact as most of the time spent in RawSpeed code is in the decoding/decompression loops etc.
 
> > mean giving up on `-fvisibility=hidden`
> 
> Why is this important? Honest question here: I don't know, and I'm not too familiar with this.

https://gcc.gnu.org/wiki/Visibility
https://clang.llvm.org/docs/LTOVisibility.html

Basically, when there is 
```
void foo() { <whatever> }
void bar() { foo(); <whatever> }
```
`-fvisibility=hidden`, tells the compiler that the `foo()` is and only is that `void foo() { <whatever> }` it sees,
and that the `foo()` is only needed for the callees that are seen.
Which means, `foo()` is not exported from `.so`, and `foo()` is known to be that very `foo()`,
not some other `foo()` that might happen to be picked up from `LD_PRELOAD` from some other library.

This is very fundamental.

> > OTOH yes, that would make PGO easier..
> 
> I don't know what PGO is.

https://en.wikipedia.org/wiki/Profile-guided_optimization
https://clang.llvm.org/docs/UsersManual.html#profile-guided-optimization

Basically, it provides compiler with better insight into the code being compiled,
based on the observations made when actually executing the code being compiled,
which should result in better optimized code.

mcourteaux, 2020-08-25:
Thank you very much for these resources and explanation. I still need to read them though. 

> Which means, foo() is not exported from .so, and foo() is known to be that very foo(),
> not some other foo() that might happen to be picked up from LD_PRELOAD from some other library.

I think LTOVisibility would only be a problem if you don't use a namespace? Here, everything is wrapped in the `rawspeed` namespace. If then, the dynamic loader finds a symbol with the exact same name mangling scheme in a `rawspeed` namespace that sits in a library loaded by `LD_PRELOAD`, then I would assume this is 100% on purpose.

> Note that master branch is very outdated.
> Currently stable is what i'd recommend to be used, or develop if feeling lucky.

Yes, thanks for the warning. 😄  I was mistaken, I'm using `develop`.

---

Conclusion here is that I'll likely go for the "shim" wrapper and make that public. I guess it will be way less effort than what I proposed initially (making RawSpeed shared-library-compatible). It might even be a nice opportunity for enthousiast to start building more "backend plugins" for the photo editing app I'm building. For example, somebody might implement a LibRaw backend, a JPEG-turbo backend (although I'll do this myself likely).

Thank you very much for taking the time to have this discussion with me.

parafin, 2020-08-25:
> Well, that's the thing, i'd like to agree with you, but that isn't the wording some radical distros use(d?).
> See e.g. https://lwn.net/Articles/660429/

I'm afraid that will indeed be a problem. And giving that bad decisions are made occasionally by packagers (e.g. distributing broken incompatible alpha version of lensfun) it will surely bring pain to everyone involved when wrong rawspeed version will be coupled with darktable or some other user of the library. As an alternative to shim library, a patch can be maintained implementing said shared library support in rawspeed, but without upstreaming it (though I suspect some packagers might still take it and use for building the package). But creating a shim library sounds much more easier.

LebedevRI, 2020-09-05:
So i've checked, and indeed, dowadays for normal `Release` build the perf difference between the current `static` build,
and a `shared` one w/o `-fvisibility` things is mostly within the noise: [rawspeed-so.rsbench.txt](https://github.com/darktable-org/rawspeed/files/5177829/rawspeed-so.rsbench.txt)

So in principle just from the perf viewpoint i would be okay with making it a possibility.
But not from the correctness standpoint, it makes it too easy to mismatch versions. :/

mcourteaux, 2020-10-14:
I created a fork which allows for dynamic linking: https://github.com/mcourteaux/rawspeed

LebedevRI, 2020-10-14:
Cool!
What works for you works for you.