darktable-org/rawspeed #248: Decoding Hasselblad Lossless JPEG

photopea, 2020-11-04:
Hi, this is not a bug, I am just really desperate, as I have been trying to implement my own Hasselblad decoder for three days, but it still has problems. Could you help me?

I am testing it with this file: http://www.rawsamples.ch/raws/hasselblad/h3d-39ii/RAW_HASSELBLAD_H3D39II.3FR

At Byte 54, there is 0xffda marker - Start of Scan

At Byte 64, the Huffman stream begins. From this spot, I "reverse" each 4 bytes of the file. A new order is ... 67, 66, 65, 64; 71, 70, 69, 68; 75, 74, ....  When reading a byte 0xff in a stream, the next byte is not skipped, as in regular JPEG.

Now I have these bits:
11111111;10111111;11100000;00000000;00100000;00000000;01111101;11101011;11101000;10111110;

I start to read pairs of values: 2 lengths, 2 codes (differences), and so on.  The "word" 1111111110 corresponds to the value 15 in a Huffman tree. Then, there is 1111111110 again. The following 15 bits are 000000000000001 (number one) and 000000000000001 (number one again).

At the beginning, I read these lengths and numbers:

     15 15     1       1
      2  11     1 1954
    12 10 2238 740
    10 10   884 696

Are my steps correct so far? How exactly should I combine these values into sensor values?

I am following https://github.com/darktable-org/rawspeed/blob/develop/src/librawspeed/decompressors/HasselbladDecompressor.cpp

So the first two sensor values are 0x8000 + 1  and  0x8000 + 1 ?

LebedevRI, 2020-11-05:
Have you tried stepping through the working code with debugger?

photopea, 2020-11-05:
I haven't worked with C or C++ or the command line in the last 10 years and I don't have an experience with any development environment. It would probably take me several more days to learn how to compile Darktable.

Could I ask someone to do debug it and tell me, what are the first values?

photopea, 2020-11-05:
Guys, I figured it out! thanks!

closed 2020-11-05 by photopea