darktable-org/rawspeed #251: Support uncompressed float DNGs

kmilos, 2020-11-13:
Unfortunately it doesn't seem the OP will contribute the report:

https://discuss.pixls.us/t/raw-samples-wanted/5420/110

LebedevRI, 2020-11-13:
Thanks. The support is trivial, i already have it working.

Can someone help with a test coverage here?
The appropriate, CC0-licensed sample is needed to be present on [RPU](https://raw.pixls.us/).
Looks like original poster used "Ricoh Theta Z1 using the Dual Fisheye Plugin
(which creates a stacked DNG, in my sample these are 7 exposures)",

kmilos, 2020-11-13:
Can't help there I'm afraid, I don't have the Theta Z1 camera.

[HDRMerge](https://github.com/jcelaya/hdrmerge/blob/master/src/DngFloatWriter.cpp) has the horizontal diff + compression hard-coded, but I suppose that could be hacked around to produce a sample if one doesn't turn up...

kmilos, 2020-11-22:
I have ended up patching HDRMerge to output uncompressed, and have just uploaded the trio of 16/24/32b fp samples to RPU (same images as already present in the database, just uncompressed).

The only difference to the Ricoh Theta Z1 using the Dual Fisheye Plugin seems to be that HDRMerge uses tiles instead of strips (a single one at that), don't know if that is significant...

LebedevRI, 2020-11-22:
> I have ended up patching HDRMerge to output uncompressed, and have just uploaded the trio of 16/24/32b fp samples to RPU (same images as already present in the database, just uncompressed).

Thank you!

> The only difference to the Ricoh Theta Z1 using the Dual Fisheye Plugin seems to be that

> HDRMerge uses tiles instead of strips (a single one at that), don't know if that is significant...

No, just that by itself is not significant.

LibRaw, 2020-11-22:
Adobe DNG SDK (source code) assumes that uncompressed is always striped (with single stripe, because offsets are easy to calculate).

kmilos, 2020-11-22:
> Adobe DNG SDK (source code) assumes that uncompressed is always striped (with single stripe, because offsets are easy to calculate).

Makes sense, but I just went for the minimal changes to HDRMerge to write out uncompressed...

LibRaw, 2020-11-22:
Also, for 24-bit sample, byte order looks wrong. Here is Adobe Camera Raw screenshot with this file: https://www.dropbox.com/s/uw9ufsw890isc2e/screenshot%202020-11-22%2021.14.49.png?dl=0

LebedevRI, 2020-11-22:
And 16-bit image decodes differently from the original motivational sample from

> https://discuss.pixls.us/t/raw-samples-wanted/5420/110

So i'll remove those 2 samples from RPU

kmilos, 2020-11-22:
I can read the 16b one just fine using tifffile in Python, so please keep it. I have no way of verifying 24b ATM.

LebedevRI, 2020-11-22:
> Also, in 24-bit sample byte order looks wrong. Here is Adobe Camera Raw screenshot with this file: https://www.dropbox.com/s/uw9ufsw890isc2e/screenshot%202020-11-22%2021.14.49.png?dl=0

FYI you can just paste images into comment editor here

> I can read the 16b one just fine using tifffile in Python, so please keep it, I believe it is valid. The difference could be due to strip vs tiles... I have no way of verifying 24b ATM.

It is not tiling vs striping, all that is nicely abstracted away.
It decodes as-if it's either predicted, or the byte packing is also wrong.

kmilos, 2020-11-22:
The following snippet works exactly the same on both the 16b and 32b uncompressed samples I produced, so believe those are valid test cases:

```python
import numpy as np
from matplotlib import pyplot as plt
import tifffile

tif = tifffile.TiffFile('hdrmerge-bayer-fp16-uncompressed.dng')
cfa = tif.series[1].asarray()
tif.close()

# RGGB
r = cfa[0::2,0::2].astype(np.float32)
g = (cfa[1::2,0::2].astype(np.float32) + cfa[0::2,1::2].astype(np.float32))/2
b = cfa[1::2,1::2].astype(np.float32)
rgb = np.dstack([r, g, b])

plt.imshow((rgb-2047)/16383)
```

(Numpy/tifffile unfortunately don't support 24b floats, so I can't test the same way...)

LebedevRI, 2020-11-22:
That doesn't look right, you are forgetting to perform byteswapping on 4 bytes.

LebedevRI, 2020-11-22:
(unless of course you are on big-endian CPU)

LibRaw, 2020-11-23:
This is II* (intel/LE byte order) DNG/TIFF files, so byteswapping is not needed on Intel/LE fof 16 and 32 bits.

24 bit is another story (and the only reliable reference for it is Adobe DNG SDK source, because there is no such thing as 'IEEE 24bit fp')

LebedevRI, 2020-11-23:
> This is II* (intel/LE byte order) DNG/TIFF files, so byteswapping is not needed on Intel/LE for 16 and 32 bits.

Ah yes, that's a good point, i only thought that it was the problem, but didn't have a moment to check code locally.

LebedevRI, 2020-11-23:
@kmilos ok, i take it back, these samples are perfect. thank you.

closed 2020-11-23 by LebedevRI

kmilos, 2020-11-23:
> these samples are perfect

You mean the 16/32b ones?

Do I still need to byte swap to construct the 24b case if that one doesn't work w/ Adobe SW?

LebedevRI, 2020-11-23:
> > these samples are perfect
> 
> You mean the 16/32b ones?

All three.

> Do I still need to byte swap to construct the 24b case if that one doesn't work w/ Adobe SW?

24bit case also just works here.
Unless someone can point out why it's wrong, i'm gonna err on the side of "that's just adobe's bug".

kmilos, 2020-11-23:
Btw, I found slightly faster code for fp16->fp32 conversion that avoids the looping, already used it in darktable's TIFF reader in https://github.com/darktable-org/darktable/commit/589cdcc8d397fcaa1706d9058b4f8d1ce81c155d

LebedevRI, 2020-11-23:
I'm not merging that. Type punning is UB.
There should be no loop there at all, and i will fix that in the compiler: https://reviews.llvm.org/D91038

LibRaw, 2020-11-23:
> "that's just adobe's bug".

AFAIK, Adobe DNG SDK is the only reference implementation for 24-bit FP.TIFF(/DNG)

kmilos, 2020-11-23:
> 24bit case also just works here.
> Unless someone can point out why it's wrong, i'm gonna err on the side of "that's just adobe's bug".

There is a possibility that both HDRMerge and you write/read in the same (wrong?) byte order and Adobe SW is correct? ;)

[This does seem suspect and big-endian to me](https://github.com/jcelaya/hdrmerge/blob/f5a2538cffe3e27bd9bea5d6a199fa211d05e6da/src/DngFloatWriter.cpp#L458).

LebedevRI, 2020-11-23:
> Sure, this type punning is not ideal.

That's an understatement :)

> If it ends up being fixed in the compiler and vectorizable, then great.

To be noted, other than type punning, that solution looks very different, i didn't digest it fully yet.

> I guess you also thought about using F16C intrinsics (HDRMerge uses them)? Don't know what's the aarch64 equivalent though...

Yeah. 
/me doesn't really like intrinsics.

> There is a possibility that both HDRMerge and you write/read in the same (wrong?) byte order and Adobe SW is correct? ;)
>
> [This does seem suspect and big-endian to [...]

I think the relevant reference here is:
https://www.adobe.com/content/dam/acom/en/products/photoshop/pdfs/dng_spec_1.4.0.0.pdf
```
BitsPerSample
Supported values are from 8 to 32 bits/sample. The depth must be the same for each sample if
SamplesPerPixel is not equal to 1. If BitsPerSample is not equal to 8 or 16 or 32, then the bits
must be packed into bytes using the TIFF default FillOrder of 1 (big-endian), even if the TIFF
file itself uses little-endian byte order.
```

24 is not 8/16/32, so it *should* be big-endian regardless, even though the DNG itself is little-endian.
So right now i'm not seeing the problem yet.

kmilos, 2020-11-23:
Ah, I missed that in the spec, sounds ok then.

If need be, I can always generate the 24b test w/ the order reversed...

LibRaw, 2020-11-23:
The problem:  24-bit DNGs created by Adobe (tools and software that uses DNG SDK) will not read correctly by RawSpeed.

LebedevRI, 2020-11-23:
> The problem: 24-bit DNGs created by Adobe (tools and software that uses DNG SDK) will not read correctly by RawSpeed.

I'm not disagreeing with that.
The question is, should the spec be read as:
  BitsPerSample
  Supported values are from 8 to 32 bits/sample. The depth must be the same for each sample if
  SamplesPerPixel is not equal to 1. If **SampleFormat is 1 (unsigned integer) and** BitsPerSample
  is not equal to 8 or 16 or 32, then the bits must be packed into bytes using the TIFF default FillOrder
  of 1 (big-endian), even if the TIFF file itself uses little-endian byte order.
?

LibRaw, 2020-11-23:
I'm not sure what is 'Big Endian' for 24-bit floats.

kmilos, 2020-11-23:
> I'm not sure what is 'Big Endian' for 24-bit floats.

Same as 16-bit or 32-bit, no? Which byte of the 2/3/4 byte word you store first...

kmilos, 2020-11-23:
FillOrder talks about packing **bits**, and yet mentions big-endian, which is a **byte** ordering concept... Must say both the TIFF and DNG specs are somewhat confusing here...

LebedevRI, 2020-11-23:
> I'm not sure what is 'Big Endian' for 24-bit floats.

Alright, this is not productive. If we are to be compatible with DNG SDK,
i'm looking forward to seeing someone contributing the proper samples to RPU,
namely, https://raw.pixls.us/getfile.php/771/raw/5G4A9394.CR2
converted into 16/24/32 bit uncompressed floating point,
each one of those 3 samples should come in 2 variations:
little-endian DNG and big-endian DNG.

kmilos, 2020-11-23:
When you think about it, I don't see why would the 24-bit float would be treated differently to 16-bit or 32-bit ones, you let the machine take care of the endianness since you're dealing with integer number of bytes... It is possible that Adobe forgot to update the wording of that section once they introduced floats in 1.4. Indeed in the 1.3 version this section is the same.

kmilos, 2020-11-23:
Currently I think there might indeed exist a problem with the HDRMerge and Rawspeed understanding/and implementation of 24-bit floats: bytes always written as big-endian w/ FillOrder=1 tag, but bits within a byte are LSB first; FillOrder seems to mean "MSB first" though?

Agreed, canonical samples are indeed needed.

LebedevRI, 2020-12-23:
> > Sure, this type punning is not ideal.
> 
> That's an understatement :)
> 
> > If it ends up being fixed in the compiler and vectorizable, then great.

And done, https://github.com/llvm/llvm-project/compare/e124844709228a076483d8b6101bbb362caf625f~1...2b61e7c68cda16159b186fea1a1b1252b94b7e16

> To be noted, other than type punning, that solution looks very different, i didn't digest it fully yet.
> 
> > I guess you also thought about using F16C intrinsics (HDRMerge uses them)? Don't know what's the aarch64 equivalent though...
> 
> Yeah.
> /me doesn't really like intrinsics.

commit 6d618a6c (2020-11-23): Merge branch 'dng-uncompressed-f16' into develop

Fixes #251.

* dng-uncompressed-f16:
  DNGDecoder: support 24-bit uncompressed floating point
  DNGDecoder: support 16-bit uncompressed floating point
  Extract 16/24-bit floating-point expanders into their own header file